zigpy/zha-device-handlers #23: Object has no attribute 'cluster_command'

prairiesnpr, 2018-12-31:
I am attempting to get the Iris Smart Plug V1 sensors working.  So far, I have the switch portion working, but still have work to do on the usage reporting.  However, the logs are full of the following:

`Error calling 'listener.cluster_command': Sensor object has no attribute 'cluster_command' 
‎18‎:‎17 __main__.py (WARNING)`

`Error calling 'listener.cluster_command': Switch object has no attribute 'cluster_command'
‎18‎:‎17 __main__.py (WARNING)`

Any ideas what I am doing wrong here?

```
<SimpleDescriptor endpoint=2 profile=49686 device_type=7 device_version=0 input_clusters=[240, 239, 238] output_clusters=[]>                                            
<SimpleDescriptor endpoint=240 profile=49686 device_type=7 device_version=0 input_clusters=[2941] output_clusters=[]>
```

```
class StatusBool(bytes):
    def serialize(self):
        _LOGGER.debug("Serialize not implemented.")

    @classmethod
    def deserialize(cls, data):
        bytes = int.from_bytes(data[:1], 'little')
        s = data[1:bytes + 1]
        r = int.from_bytes(s, 'little') & 1
        return bool(r), b''

class IrisPlug(CustomDevice):
    class IrisSmartPlugUsageCluster(LocalDataCluster, ElectricalMeasurement):
        cluster_id = 0x00ef
        name = "Iris Smart Plug Usage"
        ep_attribute = 'analog_value'

        attributes = {
            0x0081: ('active_power', t.uint32_t),
            0x0082: ('cumlative_usage', (t.uint32_t, t.uint32_t, t.uint24_t, t.uint8_t,)),
        }
        client_commands = {
            0x0081: ('active_power', (t.uint32_t,), False),
            0x0082: ('cumlative_usage', (t.uint32_t, t.uint32_t, t.uint24_t, t.uint8_t,), False),
        }
        server_commands = {}

    class IrisSmartPlugUknCluster(CustomCluster):
        cluster_id = 0x00f0
        name = "Iris Smart Plug Ukn"
        ep_attribute = 'smartplug'
        void_input_commands = [0x00fb]
        attributes = {
            0x00fb: ('current_temp', (t.uint64_t, t.uint32_t, t.uint16_t,)),
        }
        client_commands = {
            0x00fb: ('current_temp', (t.uint64_t, t.uint32_t, t.uint16_t,), False),
        }
        server_commands = {}

    class IrisOnOff(EventableCluster, OnOff):
        """Attributes and commands for switching devices between
        ‘On’ and ‘Off’ states. """
        cluster_id = 0x00ee
        name = 'On/Off'
        ep_attribute = 'on_off'
        def __init__(self, *args, **kwargs):
            super().__init__(*args, **kwargs)

        def handle_cluster_request(self, tsn, command_id, args):
            self._update_attribute(self._attridx.get('on_off'), args[0])
            def __init__(self, *arg, **kwargs):
                super().__init__(*args, **kwargs)

        @util.retryable_request
        def request(self, general, command_id, schema, *args, manufacturer=None, expect_reply=True):
            def __init__(self, *args, **kwargs):
                super().__init__(*args, **kwargs)

            if command_id == 0:
                sequence = self._endpoint._device.application.get_sequence()
                asyncio.ensure_future(self._endpoint._device.request(0xc216, 0x00ee, 0, 2, sequence, b'\x11\x00\x01\x01', expect_reply=False))
                sequence = self._endpoint._device.application.get_sequence()
                return asyncio.ensure_future(self._endpoint._device.request(0xc216, 0x00ee, 0, 2, sequence, b'\x11\x00\x02\x01\x01', expect_reply=False))
            elif command_id == 1:
                sequence = self._endpoint._device.application.get_sequence()
                asyncio.ensure_future(self._endpoint._device.request(0xc216, 0x00ee, 0, 2, sequence, b'\x11\x00\x01\x01', expect_reply=False))
                sequence = self._endpoint._device.application.get_sequence()
                return asyncio.ensure_future(self._endpoint._device.request(0xc216, 0x00ee, 0, 2, sequence, b'\x11\x00\x02\x00\x01', expect_reply=False))
            return None

        attributes = {
            0x0000: ('on_off', t.Bool),
        }
        server_commands = {
            0x00: ('on',(), False),
            0x01: ('off',(), False),
        }
        client_commands = {
            0x0080: ('on_off', (StatusBool,), False),
        }

    signature = {
        2: {
            'profile_id': 0xc216,
            'device_type': 0x0007,
            'input_clusters': [0x00ee, 0x00ef, 0x00f0],
            'output_clusters': [],
        },
        240: {
            'profile_id': 0xc216,
            'device_type': 0x0007,
            'input_clusters': [2941],
            'output_clusters': [],
        },
    }
    replacement = {
        'endpoints': {
            2: {
                'manufacturer': 'Iris',
                'model': 'AlertMe',
                'input_clusters': [IrisOnOff, IrisSmartPlugUsageCluster, IrisSmartPlugUknCluster],
                'output_clusters': [IrisOnOff],
            }
        },

    }```

dmulcahey, 2018-12-31:
Are you sure the SimpleDescriptor is correct?

Also, stick to The base custom cluster for your base class. The local data and eventable bases aren’t necessary for what you’re implementing.

Try extending just custom cluster and the cluster that implements the functionality without providing any custom types or code.. that may be enough.

The error is because you don’t have a listener for cluster_command

You need a listener for each event that is possible even if it does nothing

prairiesnpr, 2018-12-31:
The SimpleDescriptor matches the endpoints as I know them, I've had these devices running for a few years on a custom controller that I'm now trying to transition to Home Assistant.

I am able to read the sensor in Home Assistant now, had to change the id for active power to 0.  I was under the impression that refencing the ElectricalMeasurement type would set the sensor type in Home Assistant, but looking at the zha sensor component, it is coded to the cluster id and is thus set as a generic sensor.  I guess I will need to add a custom sensor type to home assistant to get the functionality I want.

I went ahead and changed the base class to custom cluster and it did not impact the cluster_command error.  Would you mind pointing me towards an example of how to register a listener for cluster_command?  I wasn't able to find an example in the existing quirks showing this.

dmulcahey, 2018-12-31:
It would be in the entity class or a listener in HA. Are you using a custom component here? Or a custom entity class in HA? 

Also, you shouldn’t need the on off cluster in output clusters. It generally only ends up there for remote like devices.

Can you put the full log on something like hastebin too? Or at least a decent chunk of it? 

For examples:

https://github.com/home-assistant/home-assistant/blob/ff80fc347bd37bc9264cb9a19093052f85b268bf/homeassistant/components/zha/entities/listeners.py#L35

My guess is that you’re getting those logs because you have on/off in output clusters in the replacement signature. 

The base class changes are just so everything functions as you would expect... localdatacluster is for emulating a cluster and eventable is really for remotes and buttons that are stateless. You have a plug so neither of those should be necessary. 

HA has a sensor for both electrical measurement and metering for HA. They are both pretty raw atm. What are you expecting to see in HA?

dmulcahey, 2018-12-31:
Also, if you have discord I can help in realtime...

prairiesnpr, 2019-01-01:
I'm not using a custom component or entity class currently, although I may need to from the looks of it.

Log is here:  Sensor error is repeated, since I have two plugs on the network and they repeatedly fire messages with current and cumulative usage as well as another message that may be temp, but not certain.  Near the end I switched the plug on and off so you can see the same error for the switch component.

https://hastebin.com/veyozibade.coffeescript

The goal was to have be able to view the current state of the plug as well as turn it on and off. (this works now)

Then I would like to have Home Assistant track the current usage of the plug, it does this now, looks like this.

[image]

In the log view though it looks like this:
[image]

Would prefer to have a graph of watts over time for the instant reading and a then the cumulative reading would be a value like the instant usage is now.

Current quirk code: https://hastebin.com/otayajepec.py

I'll try removing the output cluster when I get a chance and follow up with the results, I may be off on my terminology here, but thought the output cluster was what allowed the state of the switch to be set.  

Thanks for the help, I have discord, but will be in and out tonight, not going to have a chance to do much more troubleshooting.  I hate to post on these things, usually can figure them out myself, but this was supposed to be the easiest device to get working and it's taken me awhile to get this far.

dmulcahey, 2019-01-01:
If you have 5 minutes humor me and try something... remove all of the code from the on/off cluster and the electrical cluster... keep the declarations the way they are and just set the cluster id’s. Get rid of everything else. Also remove on/off from the output clusters in the replacement signature.

prairiesnpr, 2019-01-01:
Just so I'm clear, remove the attributes, client_commands, and server_commands also?

dmulcahey, 2019-01-01:
The hastebin links just went empty on me... send me the code again really quick and I'll hack it up. Will take 2 mins.

prairiesnpr, 2019-01-01:
Hastebin seems to be down, can't add anything.

Here is the code:

```
import zigpy.types as t
from zigpy.quirks import CustomDevice, CustomCluster
from zigpy.zcl import foundation
from zigpy import util
from zhaquirks import EventableCluster, LocalDataCluster
from zigpy.zcl.clusters.general import Basic, OnOff, Identify, Ota, LevelControl, PowerConfiguration, Scenes, BinaryInput, Groups

from zigpy.zcl.clusters.homeautomation import ElectricalMeasurement
from zigpy.zcl.clusters.smartenergy import Metering
from zigpy.zcl.clusters.measurement import TemperatureMeasurement
from zigpy.profiles import PROFILES, zha
import asyncio
import logging

_LOGGER = logging.getLogger('zigpy.zcl')

class StatusBool(bytes):
    def serialize(self):
        _LOGGER.debug("Serialize not implemented.")

    @classmethod
    def deserialize(cls, data):
        bytes = int.from_bytes(data[:1], 'little')
        s = data[1:bytes + 1]
        r = int.from_bytes(s, 'little') & 1
        return bool(r), b''

class IrisPlug(CustomDevice):
    class IrisSmartPlugUsageCluster(CustomCluster, ElectricalMeasurement):
        cluster_id = 0x00ef
        ep_attribute = 'electrical_measurement'
        def handle_cluster_request(self, tsn, command_id, args):
            cmd_name = self.client_commands[command_id][0]
            self.debug('Command Id %s (%s) - %s' % (command_id,cmd_name, args[0]))
            self._update_attribute(self._attridx.get(cmd_name), args[0])
            def __init__(self, *arg, **kwargs):
                super().__init__(*args, **kwargs)
        attributes = {
            0x0: ('active_power', t.uint32_t),
            0x0082: ('cumlative_usage', (t.uint32_t, t.uint32_t, t.uint24_t, t.uint8_t,)),
        }
        client_commands = {
            0x0081: ('active_power', (t.uint32_t,), False),
            0x0082: ('cumlative_usage', (t.uint32_t, t.uint32_t, t.uint24_t, t.uint8_t,), False),
        }
        server_commands = {}

    class IrisSmartPlugUknCluster(CustomCluster):
        cluster_id = 0x00f0
        name = "Iris Smart Plug Ukn"
        ep_attribute = 'smartplug'
        void_input_commands = [0x00fb]
        attributes = {
            0x00fb: ('current_temp', (t.uint64_t, t.uint32_t, t.uint16_t,)),
        }
        client_commands = {
            0x00fb: ('current_temp', (t.uint64_t, t.uint32_t, t.uint16_t,), False),
        }
        server_commands = {}

    class IrisOnOff(CustomCluster, OnOff):
        """Attributes and commands for switching devices between
        ‘On’ and ‘Off’ states. """
        cluster_id = 0x00ee
        #cluster_id = OnOff.cluster_id
        name = 'On/Off'
        ep_attribute = 'on_off'
        def __init__(self, *args, **kwargs):
            super().__init__(*args, **kwargs)

        def handle_cluster_request(self, tsn, command_id, args):
            self.debug('Command Id %s - %s' % (command_id, args[0]))
            self._update_attribute(self._attridx.get('on_off'), args[0])
            def __init__(self, *arg, **kwargs):
                super().__init__(*args, **kwargs)

        @util.retryable_request
        def request(self, general, command_id, schema, *args, manufacturer=None, expect_reply=True):
            def __init__(self, *args, **kwargs):
                super().__init__(*args, **kwargs)

            if command_id == 0:
                sequence = self._endpoint._device.application.get_sequence()
                asyncio.ensure_future(self._endpoint._device.request(0xc216, 0x00ee, 0, 2, sequence, b'\x11\x00\x01\x01', expect_reply=False))
                sequence = self._endpoint._device.application.get_sequence()
                return asyncio.ensure_future(self._endpoint._device.request(0xc216, 0x00ee, 0, 2, sequence, b'\x11\x00\x02\x01\x01', expect_reply=False))
            elif command_id == 1:
                sequence = self._endpoint._device.application.get_sequence()
                asyncio.ensure_future(self._endpoint._device.request(0xc216, 0x00ee, 0, 2, sequence, b'\x11\x00\x01\x01', expect_reply=False))
                sequence = self._endpoint._device.application.get_sequence()
                return asyncio.ensure_future(self._endpoint._device.request(0xc216, 0x00ee, 0, 2, sequence, b'\x11\x00\x02\x00\x01', expect_reply=False))
            return None

        attributes = {
            0x0000: ('on_off', t.Bool),
        }
        server_commands = {
            0x00: ('on',(), False),
            0x01: ('off',(), False),
        }
        client_commands = {
            0x0080: ('on_off', (StatusBool,), False),
        }

    signature = {
        2: {
            'profile_id': 0xc216,
            'device_type': 0x0007,
            'input_clusters': [0x00ee, 0x00ef, 0x00f0],
            'output_clusters': [],
        },
        240: {
            'profile_id': 0xc216,
            'device_type': 0x0007,
            'input_clusters': [2941],
            'output_clusters': [],
        },
    }
    replacement = {
        'endpoints': {
            2: {
                'manufacturer': 'Iris',
                'model': 'AlertMe',
                'input_clusters': [IrisOnOff, IrisSmartPlugUsageCluster, IrisSmartPlugUknCluster],
                'output_clusters': [IrisOnOff],
            },
            240: {
                'input_clusters': [2941],
                 }
        },

    }    
```

dmulcahey, 2019-01-01:
Try this (just make sure the formatting is right):

```
import zigpy.types as t
from zigpy.quirks import CustomDevice, CustomCluster
from zigpy.zcl import foundation
from zigpy import util
from zhaquirks import EventableCluster, LocalDataCluster
from zigpy.zcl.clusters.general import Basic, OnOff, Identify, Ota, LevelControl, PowerConfiguration, Scenes, BinaryInput, Groups

from zigpy.zcl.clusters.homeautomation import ElectricalMeasurement
from zigpy.zcl.clusters.smartenergy import Metering
from zigpy.zcl.clusters.measurement import TemperatureMeasurement
from zigpy.profiles import PROFILES, zha
import asyncio
import logging

_LOGGER = logging.getLogger('zigpy.zcl')

class IrisPlug(CustomDevice):
    class IrisSmartPlugUsageCluster(CustomCluster, ElectricalMeasurement):
        cluster_id = 0x00ef

    class IrisSmartPlugUknCluster(CustomCluster):
        cluster_id = 0x00f0
        name = "Iris Smart Plug Ukn"
        ep_attribute = 'smartplug'
        void_input_commands = [0x00fb]
        attributes = {
            0x00fb: ('current_temp', (t.uint64_t, t.uint32_t, t.uint16_t,)),
        }
        client_commands = {
            0x00fb: ('current_temp', (t.uint64_t, t.uint32_t, t.uint16_t,), False),
        }
        server_commands = {}

    class IrisOnOff(CustomCluster, OnOff):
        """Attributes and commands for switching devices between
        ‘On’ and ‘Off’ states. """
        cluster_id = 0x00ee

    signature = {
        2: {
            'profile_id': 0xc216,
            'device_type': 0x0007,
            'input_clusters': [0x00ee, 0x00ef, 0x00f0],
            'output_clusters': [],
        },
        240: {
            'profile_id': 0xc216,
            'device_type': 0x0007,
            'input_clusters': [2941],
            'output_clusters': [],
        },
    }
    replacement = {
        'endpoints': {
            2: {
                'manufacturer': 'Iris',
                'model': 'AlertMe',
                'input_clusters': [IrisOnOff, IrisSmartPlugUsageCluster, IrisSmartPlugUknCluster],
                'output_clusters': [],
            },
            240: {
                'input_clusters': [2941],
                 }
        },

    }    
```

prairiesnpr, 2019-01-01:
uploaded and ran.

Logs here: https://pastebin.com/KJT1py1m

State of switch no longer reported in Home Assistant
No longer able to turn switch on or off from Home Assistant
Current Watts not updated, no other stats are shown other than online.

One note, these switches don't follow the standard. I takes two messages to get them to switch states, one to unlock and another to change the state, both have to be sent to the same endpoint.

dmulcahey, 2019-01-01:
Try this: https://pastebin.com/raw/sAnBrNYc

prairiesnpr, 2019-01-01:
Logs from latest code:
https://pastebin.com/TcB7adBe

Results, no change, unable to change state from home assistant, state not update when changed at switch.

I'm guessing the state isn't changed from home assistant, since the command has to be sent to cluster 0x0000,

prairiesnpr, 2019-01-01:
And the state change is reported from the plug on 0x0080, and the package is

dmulcahey, 2019-01-01:
Do you mean attribute and not cluster? You don’t even have cluster 0 in the signature. If so, it should be easy enough to correct that. We can discuss on discord at some point tomorrow if you’re around.

dmulcahey, 2019-01-08:
how did you make out with this?

prairiesnpr, 2019-01-10:
Does this still require your custom home assistant branch?

dmulcahey, 2019-01-10:
No. Support launched in .85 today.

prairiesnpr, 2019-01-13:
I finally solved this.  Had to override handle_message to not fire the listener event.

closed 2019-01-13 by prairiesnpr

dmdzero, 2022-03-10:
> I finally solved this. Had to override handle_message to not fire the listener event.

Hello! I'm sorry for bringing up this old issue, but could you explain how did you get the smartplug to pair? I can't even get it to pair. Smartplug is in pairing mode and discovered as unknown device, but unable to pair. If I understand correctly, quirks are applied after device pairing?