zigpy/zha-device-handlers #29: Xiaomi Button state is inverted

frog32, 2019-01-23:
I am using the xiaomi round button and its state reported is inverted. When not pushing the button it is showing as on and when pushing it it shows as off.

dmulcahey, 2019-01-23:
can you post the network traffic for the interaction? I’m pretty sure that’s how this device works. The button acts as a toggle button.

drjared88, 2019-01-25:
Not sure exactly what you are looking for as the signature but in my home-assistant.log when it is doing it's quirk stuff it shows up as 'lumi.sensor_switch'

From what I've seen is that Mija devices have the same name as the Aqaras except without the .aq2 at the end.

frog32, 2019-01-27:
Removing and pairing the button again results in a few stacktraces.
https://pastebin.com/Pi7TzZf7

dmulcahey, 2019-01-27:
That’s all bellows and Zigpy. Usually you only see those if the device didn’t remove or join completely. 

I’d try this: 

get the ieee address for the device (this is on the zha panel in HA and looks like “FG:YU:...”

Try to remove it again

Shut down ha and check the zigbee.db tables for any reference to the ieee from step 1. If found, delete the rows

Start HA and join the device again. For ALL Xiaomi devices you need to press the button ~5 seconds after pairing starts until it completes

frog32, 2019-01-27:
Had to do that since after a restart I was stuck with just an attribute in the database which makes zigpy throw up. I am pressing that button more often than every 5 seconds already but have no idea for how long that is needed. Most of the time I am doing that for a minute.
https://pastebin.com/6XqUYJFe
Pairing did not work but this time it did not screw the database entirely.

frog32, 2019-01-27:
I tried again and after a few tries I ended up with an almost valid result:
https://pastebin.com/hkV5U88J

dmulcahey, 2019-01-27:
@frog32 if it’s a Mija device my handlers won’t even kick in.

frog32, 2019-01-27:
I know, I thought that it is still worth posting the debug output for later usage.

dmulcahey, 2019-01-27:
You can leave the issues open that way you get notified when I get them supported :)

dlashua, 2019-02-06:
The binary_sensor for the round button does indeed show as "ON" normally, and switches to "OFF" when held down. I agree that this should be inverted. 

Double, Triple, and Quadruple clicks results in the following in logs:
````
Feb 06 09:35:50 maverick hass[30965]: 2019-02-06 09:35:50 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event zha_event[R]: unique_id=zha_event.lumi_lumi_sensor_switch_019d581c_1, command=attribute_updated, args=attribute_id=32768, attribute_name=Unknown, value=2>
````

The value is 2 for a double click, 3 for triple, and 4 for quadruple. As it is it's enough to trigger on in Home Assistant, however, it would be better to have the attribute named. Additionally, the event generated is different for a single click. It looks like this:
````
Feb 06 09:43:54 maverick hass[30965]: 2019-02-06 09:43:54 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event zha_event[R]: unique_id=zha_event.lumi_lumi_sensor_switch_019d581c_1, command=attribute_updated, args=attribute_id=0, attribute_name=on_off, value=Bool.false>
````

Followed by this when the button is released:
````
Feb 06 09:43:54 maverick hass[30965]: 2019-02-06 09:43:54 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: domain=notify, service=daniel_house, service_data=message=zha_event seen {'unique_id': 'zha_event.lumi_lumi_sensor_switch_019d581c_1', 'command': 'attribute_updated', 'args': {'attribute_id': 0, 'attribute_name': 'on_off', 'value': <Bool.true: 1>}}>
````

Bellows Device Information:
````
Device:
  NWK: 0x60a6
  IEEE: 00:15:8d:00:01:9d:58:1c
  Endpoints:
    1: profile=0x104, device_type=DeviceType.DIMMER_SWITCH
      Input Clusters:
        Basic (0)
        Identify (3)
        Ota (25)
        Manufacturer Specific (65535)
      Output Clusters:
        Basic (0)
        Identify (3)
        Groups (4)
        Scenes (5)
        On/Off (6)
        Level control (8)
        Ota (25)
````

glassbase, 2019-02-23:
IMO this should be a `sensor.` with states of single, double, triple, quad. More user friendly for the people who don't want to code but would like to utilize these types of devices with conditions in automations of single/double/etc. 

I recently switched from deconz integration to ZHA integration and this was the one thing I had working via appdaemon and deconz_event. (I will rewrite my appdaemon app to utilize the zha_event)

dmulcahey, 2019-02-23:
They emit events for this purpose already to enable that. You can see the event data by using the new event tool in HA and subscribing to zha_event. All of the actions you mentioned are already available.

glassbase, 2019-02-23:
I forgot about that new event watching tool.

dmulcahey, 2019-02-26:
@glassbase I also like the idea of them being sensors that show the name / data from the last event that was fired. I think it may be noisy for many users though. Can you do me a favor and add a poll to the community forum for HA and get feedback from the community? I see 3 options:

- leave existing binary sensors - pretty flaky
- switch to sensor that shows data from last action
- remove the sensor all together and rely on events and bindings

dmulcahey, 2019-03-03:
will be in next release

closed 2019-03-03 by dmulcahey