zigpy/zha-device-handlers #32: Request for support for Lutron connected bulb remote

w1ll1am23, 2019-02-08:
Issue was originally opened here https://github.com/zigpy/bellows/issues/141 but was requested to be moved to the device handlers I believe the relevent information required is here. 

```
2019-02-07 18:37:44 INFO (MainThread) [zigpy.device] [0x6e04] Discovered endpoints: [1]
2019-02-07 18:37:45 INFO (MainThread) [zigpy.endpoint] [0x6e04:1] Discovered endpoint information: <SimpleDescriptor endpoint=1 profile=49246 device_type=2080 device_version=2 input_clusters=[0, 4096, 65280, 64580] output_clusters=[4096, 3, 6, 8, 4, 5, 0, 65280]>
2019-02-07 18:37:45 DEBUG (MainThread) [zigpy.endpoint] [0x6e04:1] Manufacturer: Lutron
2019-02-07 18:37:45 DEBUG (MainThread) [zigpy.endpoint] [0x6e04:1] Model: LZL4BWHL01 Remote
```

dmulcahey, 2019-02-21:
Events should be emitted for this naturally in ZHA now, and binding support is on the way!

dmulcahey, 2019-02-26:
@w1ll1am23 if you have time I can show you how to do this on discord and you can PR it if you'd like?

roblandry, 2019-03-09:
I have something for you to give a try.

1. Create a Lutron folder in the quirks directory. (ie. /usr/local//lib/python3.7/site-packages/zhaquirks/Lutron/) This location will depend on what os you are running and where python packages are stored.
```mkdir /usr/local//lib/python3.7/site-packages/zhaquirks/Lutron/```

2. Create a __init__.py file in the newly created folder.
``` touch __init__.py ```

3. Paste the following code into a newly created file 'LZL4BWHL01Remote.py' 
``` nano LZL4BWHL01Remote.py```

```python
"""Device handler for Lutron LZL4BWHL01 Remote."""
from zigpy.profiles import zll
from zigpy.zcl.clusters.general import Basic, Identify, Groups, Scenes, OnOff,\
    LevelControl
from zigpy.zcl.clusters.lightlink import LightLink
from zigpy.quirks import CustomDevice

MANUFACTURER_SPECIFIC_CLUSTER_ID_1 = 0xFF00 # decimal = 65280
MANUFACTURER_SPECIFIC_CLUSTER_ID_2 = 0xFC44 # decimal = 64580

class LutronLZL4BWHL01Remote(CustomDevice):
    """Custom device representing Lutron LZL4BWHL01 Remote."""

    signature = {
        #  <SimpleDescriptor endpoint=1 profile=49246 device_type=2080 
        #  device_version=2
        #  input_clusters=[0, 4096, 65280, 64580]
        #  output_clusters=[4096, 3, 6, 8, 4, 5, 0, 65280]>
        1: {
            'manufacturer': 'Lutron',
            'model': 'LZL4BWHL01 Remote',
            'profile_id': zll.PROFILE_ID,
            'device_type': zll.DeviceType.CONTROLLER,
            'input_clusters': [
                Basic.cluster_id,
                LightLink.cluster_id,
                MANUFACTURER_SPECIFIC_CLUSTER_ID_1,
                MANUFACTURER_SPECIFIC_CLUSTER_ID_2
            ],
            'output_clusters': [
                LightLink.cluster_id,
                Identify.cluster_id,
                OnOff.cluster_id,
                LevelControl.cluster_id,
                Groups.cluster_id,
                Scenes.cluster_id,
                Basic.cluster_id,
                MANUFACTURER_SPECIFIC_CLUSTER_ID_1
            ],
        },
    }

    replacement = {
        'endpoints': {
            1: {
                'manufacturer': 'Lutron',
                'model': 'LZL4BWHL01 Remote',
                'profile_id': zll.PROFILE_ID,
                'device_type': zll.DeviceType.CONTROLLER,
                'input_clusters': [
                    Basic.cluster_id,
                    LightLink.cluster_id,
                    MANUFACTURER_SPECIFIC_CLUSTER_ID_1,
                    MANUFACTURER_SPECIFIC_CLUSTER_ID_2
                ],
                'output_clusters': [
                    LightLink.cluster_id,
                    Identify.cluster_id,
                    OnOff.cluster_id,
                    LevelControl.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    Basic.cluster_id,
                    MANUFACTURER_SPECIFIC_CLUSTER_ID_1
                ],
            }
        },
    }
```

4. Restart home assistant to reload the quirks. 

5. Go into the zha configuration panel in home assistant, select your device, and click reconfigure.

Let me know how it goes. :-)

w1ll1am23, 2019-03-10:
@roblandry no luck. Followed your directions with my dev install of HA and I see that the ZHA entity shows that the quirk is loaded, but the binary sensor that is associated with the remote doesn't change states at all with any button presses.

roblandry, 2019-03-16:
Ok, so I have looked into these devices a little, and there is little documentation available online. ST is generally a good go to as the device handlers are freely available to look at. If I'm not mistaken, most if not all of the other HA products do not support this device "pico" remote. The only workaround i have seen is this: https://community.smartthings.com/t/how-to-control-anything-with-inexpensive-lutron-pico-remotes/117761 for ST. Honestly I am not very fluent with the backend zha stuff, but if i were you, i would play around with the bellows cli and see what is reported on the different clusters to see if you can find the commands to make it work. Maybe @dmulcahey has a better suggestion?

w1ll1am23, 2019-03-16:
@roblandry just to clarify this isn't a pico remote, that is a different product offered by Lutron. 

Pico - https://www.homedepot.com/p/Lutron-Pico-Remote-Control-for-Caseta-Wireless-Dimmer-White-PJ2-3BRL-WH-L01R/204718223

roblandry, 2019-03-16:
ok, then i apologize for going down that rabbit hole. lol

jdub1810, 2019-04-30:
Hi, I was curious if there was progress on the Connected Bulb Remote or if the work was being done on the Pico Remote.  Thanks.

sreknob, 2019-05-22:
Found this via search. The LZL4BWHL01 is the Connected Bulb Remote that @w1ll1am23 is looking for and that's the part number that is referenced in the quirks file that @roblandry posted... so it appears that you're both talking about the same device.

There is a lot of stupid terminology regarding these remotes... the PICO are Lutron ClearConnect and can only be used with the Lutron switches and bridge (and Wink that has a ClearConnect radio) but the Connected Bulb Remotes are zigbee devices (but some people mistakenly refer to them as PICO remotes).

I also have a Connected Bulb Remote and would like to get it working with HA as I move from Wink. I will try out this as well.

Looking at the smartthings device handler [here](https://gist.github.com/workingmonk/47c99fb01d4a3caa355a) from this [thread](https://community.smartthings.com/t/alpha-lutron-connected-bulb-remote/27918) it appears that the fingerprints and clusters used are different. Seem that ST never got it fully integrated because the lack of "grouping support" in ST. Not sure if this is an obstacle for ZHA or not... See [here](https://community.smartthings.com/t/lutron-zigbee-remote-lcbr-as-button-controller/73951/3)

@w1ll1am23 have you found any usable way to make this work with HA?

w1ll1am23, 2019-05-26:
@sreknob I haven't found a way to use my remote. Kinda just waiting for someone to get it working. I don't really have a big need for it, just don't want to get rid of it if it will work eventually.

bballwiz5, 2019-05-30:
I would also like to move off of Wink and these remotes are the only thing keeping me there. I finally had time to hook up a zigbee packet sniffer to see how Wink was configuring the remote and why the zha component wasn't seeing any events from them.

The connected bulb remote seems to only be able to send broadcast packets to a group. Wink configures the remote by binding the level control and on/off clusters to a **group** id, not a direct bind to the controller. When a button is pressed the remote sends a level control broadcast to the group specified in that binding.

The current zha implementation attempts to bind the level control cluster to the controller (hub) directly. The remote reports this binding as successful, but when a button is pressed, it still sends a broadcast to its group (in this case a seemingly random group). Since the hub is not listening for that random group, it does not see the message. This mechanism makes sense for the original purpose of these remotes, but does not work as well for controlling the hub.

I have not yet had time to look into how to solve this issue in a device handler.

Adminiuga, 2019-05-30:
Sounds like a ZLL remote, similar to IKEA 5 button remote. How many of those remotes are you planning to have?

bballwiz5, 2019-05-30:
I have 13 hooked up to my Wink. Is the ikea remote supported by HA? Does it do the same bind to a group, not a device thing? I didn't see a quirk for it.

bballwiz5, 2019-06-01:
I'd like to work on developing a quirk for this, but its unclear on where to put the quirk file in hass.io. Any pointers?

roblandry, 2019-06-02:
@bballwiz5, the quirk stub is already located at the top of the post, however, more work needs to be done to get it to work properly. I'm not too familiar with hasso, but the quirks would go in the zhaquirks folder hidden somewhere in your root. It would be the same place where all of your python modules are located.

bballwiz5, 2019-06-02:
@roblandry thanks for that stub. Unfortunately it doesn't work due to the issue I raised above (the remote does not support direct binding to the controller).

I did figure out how to load custom quirks in hass.io. You can install the custom_deps addon from the [development hassio addons](https://github.com/home-assistant/hassio-addons-development) repository. Then add zhaquirks and start the addon. This will load the zhaquirks python package in the /config/deps folder. This overrides homeassistant's current version. You can then add the quirk to the right site_packages folder.

More directly pertaining to this issue: I managed to create a quirk that appears to work. At least now I receive events in zha_event. This quirk builds on the excellent recent work by @Adminiuga [adding group management to zigpy](https://github.com/zigpy/zigpy/pull/156). Therefore, it requires the latest zigpy and bellows versions (can add them as well to custom_deps addon). It works by creating a single group that will only have the controller (hub) as a member. Then all the remotes are bound to broadcast to that group. This groupid should likely live somewhere outside this quirk. Let me know if this quirk works for anyone else and I'll submit a PR.

```python
"""Device handler for Lutron LZL4BWHL01 Remote."""
from zigpy.profiles import zll
from zigpy.zcl.clusters.general import Basic, Identify, Groups, Scenes, OnOff,\
    LevelControl
from zigpy.zcl.clusters.lightlink import LightLink
from zigpy.quirks import CustomDevice, CustomCluster
from zigpy.zdo import types

MANUFACTURER_SPECIFIC_CLUSTER_ID_1 = 0xFF00 # decimal = 65280
MANUFACTURER_SPECIFIC_CLUSTER_ID_2 = 0xFC44 # decimal = 64580

CONTROLLER_GROUP_ID = 0x30

class GroupBoundClusterMixin(object):
    """Group bound cluster."""

    async def bind(self):
        controllerid = self._endpoint.device.application.ieee
        await self._endpoint.device.application.get_device(controllerid).add_to_group(CONTROLLER_GROUP_ID)
        dstaddr = types.MultiAddress()
        dstaddr.addrmode = 1
        dstaddr.nwk = CONTROLLER_GROUP_ID
        dstaddr.endpoint = self._endpoint.endpoint_id
        return await self._endpoint.device.zdo.Bind_req(self._endpoint.device.ieee, self._endpoint.endpoint_id, self.cluster_id, dstaddr)

class OnOffGroupCluster(GroupBoundClusterMixin, CustomCluster, OnOff):
    pass
    
class LevelControlGroupCluster(GroupBoundClusterMixin, CustomCluster, LevelControl):
    pass

class LutronLZL4BWHL01Remote(CustomDevice):
    """Custom device representing Lutron LZL4BWHL01 Remote."""

    signature = {
        #  <SimpleDescriptor endpoint=1 profile=49246 device_type=2080 
        #  device_version=2
        #  input_clusters=[0, 4096, 65280, 64580]
        #  output_clusters=[4096, 3, 6, 8, 4, 5, 0, 65280]>
        1: {
            'manufacturer': 'Lutron',
            'model': 'LZL4BWHL01 Remote',
            'profile_id': zll.PROFILE_ID,
            'device_type': zll.DeviceType.CONTROLLER,
            'input_clusters': [
                Basic.cluster_id,
                LightLink.cluster_id,
                MANUFACTURER_SPECIFIC_CLUSTER_ID_1,
                MANUFACTURER_SPECIFIC_CLUSTER_ID_2
            ],
            'output_clusters': [
                LightLink.cluster_id,
                Identify.cluster_id,
                OnOff.cluster_id,
                LevelControl.cluster_id,
                Groups.cluster_id,
                Scenes.cluster_id,
                Basic.cluster_id,
                MANUFACTURER_SPECIFIC_CLUSTER_ID_1
            ],
        },
    }

    replacement = {
        'endpoints': {
            1: {
                'manufacturer': 'Lutron',
                'model': 'LZL4BWHL01 Remote',
                'profile_id': zll.PROFILE_ID,
                'device_type': zll.DeviceType.CONTROLLER,
                'input_clusters': [
                    Basic.cluster_id,
                    LightLink.cluster_id,
                    MANUFACTURER_SPECIFIC_CLUSTER_ID_1,
                    MANUFACTURER_SPECIFIC_CLUSTER_ID_2
                ],
                'output_clusters': [
                    LightLink.cluster_id,
                    Identify.cluster_id,
                    OnOffGroupCluster,
                    LevelControlGroupCluster,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    Basic.cluster_id,
                    MANUFACTURER_SPECIFIC_CLUSTER_ID_1
                ],
            }
        },
    }
```

Adminiuga, 2019-06-02:
Does it have  a `touchlink` cluster? I wonder if it is similar to Ikea 5
button remote.

sreknob, 2019-06-02:
@bballwiz5 - Thanks for sharing. I updated bellows (0.8.0) and zigpy (0.4.2) to give this a try on 0.93.2. I'm running in docker, so just did a pip install -upgrade for bellows and zigpy.

The device pairs and the quirk loads properly but I don't see any zha_events in my log with button presses.

`2019-06-02 14:07:33 INFO (MainThread) [zigpy.application] Device 0xf175 (ff:ff:00:0f:e7:ff:50:4b) joined the network
2019-06-02 14:07:33 INFO (MainThread) [zigpy.device] [0xf175] Requesting 'Node Descriptor'
2019-06-02 14:07:34 INFO (MainThread) [zigpy.application] Device 0xf175 (ff:ff:00:0f:e7:ff:50:4b) joined the network
2019-06-02 14:07:42 INFO (MainThread) [zigpy.device] [0xf175] Node Descriptor: <NodeDescriptor byte1=18 byte2=64 mac_capability_flags=128 manufacturer_code=4420 maximum_buffer_size=89 maximum_incoming_transfer_size=61 server_mask=0 maximum_outgoing_transfer_size=61 descriptor_capability_field=0>
2019-06-02 14:07:42 INFO (MainThread) [zigpy.device] [0xf175] Discovering endpoints
2019-06-02 14:07:43 INFO (MainThread) [zigpy.device] [0xf175] Discovered endpoints: [1]
2019-06-02 14:07:43 INFO (MainThread) [zigpy.endpoint] [0xf175:1] Discovering endpoint information
2019-06-02 14:07:43 INFO (MainThread) [zigpy.endpoint] [0xf175:1] Discovered endpoint information: <SimpleDescriptor endpoint=1 profile=49246 device_type=2080 device_version=2 input_clusters=[0, 4096, 65280, 64580] output_clusters=[4096, 3, 6, 8, 4, 5, 0, 65280]>
2019-06-02 14:07:45 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new zha.zha entity: zha.lutron_lzl4bwhl01_remote
2019-06-02 14:07:45 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new binary_sensor.zha entity: binary_sensor.lutron_lzl4bwhl01_remote_e7ff504b_1`

Do I need the dev versions of bellows or zigpy?

bballwiz5, 2019-06-03:
I don't believe it would show up there. Instead, you can go to the events console in the left sidebar under developer tools. Then subscribe to zha_event.

bballwiz5, 2019-06-03:
@Adminiuga it does not have a touchlink cluster as far as I can tell.

bballwiz5, 2019-06-04:
I can confirm it works in my setup and a friend's setup. Oddly, some remotes seem to present as " Lutron" instead of "Lutron".

sreknob, 2019-06-04:
@bballwiz5 thanks for the tip.... that's so much easier :-)
I figured out one of my problems... I copied and pasted the wrong quirk (used the one by roblandry)
*facepalm*
With the proper quirk loaded I am getting the following error though when I re-pair the remote:

```
WARNING (MainThread) [homeassistant.components.zha.core.device] Lutron LZL4BWHL01 Remote channel: on_off-0x891a:1:0x0006 async_configure stage failed ex: 'Device' object has no attribute 'add_to_group'
WARNING (MainThread) [homeassistant.components.zha.core.device] Lutron LZL4BWHL01 Remote channel: level-0x891a:1:0x0008 async_configure stage failed ex: 'Device' object has no attribute 'add_to_group'
```
Still have to figure out what ELSE I'm doing wrong!

bballwiz5, 2019-06-04:
I can't tell you the number of times I've done similar things over the years.

That function was added in zigoy 0.4.2. Can you check your zigpy version?

Adminiuga, 2019-06-04:
@sreknob you would need to use dev version of HA or pin versions in `homeassistant/components/zha/manifest.json` to `zigpy-homeassistant==0.4.2` and `bellows-homeassistant==0.8.0` in other words take manifest.json from dev tree of HA.

sreknob, 2019-06-05:
thanks @Adminiuga and @bballwiz5 for all your help!
Both with bellows and ziqpy updated manually with pip and with manifest.json I wasn't seeing anything. Then I checked the PR and saw that `zhaquirks/__init__.py ` was also updated with the group support. Once I pulled that I started seeing zha_events!

I also wanted to commend you on some fine work - making group support is something that SmartThings hasn't get going in a couple years. Genius work and much appreciated!

Hackashaq666, 2019-06-05:
This is great progress, thank you!  Looking forward to this being added to a future version of home-assistant, hopefully soon.

closed 2019-06-09 by dmulcahey

br0nd, 2019-06-19:
@sreknob @bballwiz5 @Adminiuga I must be missing but I have not been able to get this to work.
Here is what I tried.. 
1) Pulled homeassistant docker with a dev tag (which has the manifest json with the newer zigby and bellows versions
2) manually pulled zhaquirks__init__.py and the zhaquirks/lutron folder to my local zhaquirks pip installation in the docker

it looks like I am successful in pairing the remote using the HA ZHA panel  (logs below), but when I subscribe to zha_event and hit remote buttons, I don't see anything. Any thoughts on what I am missing?

>  Lutron LZL4BWHL01 Remote: started configuration
 Lutron LZL4BWHL01 Remote: channel: zdo- Lutron LZL4BWHL01 Remote_ZDO async_configure stage succeeded
[0x1ad7:zdo] ZDO request 0x0006: [0xfffd, 260, [25], []]
0x9503:1:0x0000: bound  'basic' cluster: Status.SUCCESS
0x9503:1:0x0000: finished channel configuration
0x9503:1:0x1000: bound  'lightlink' cluster: Status.SUCCESS
0x9503:1:0x1000: finished channel configuration
 Lutron LZL4BWHL01 Remote: channel: lightlink-0x9503:1:0x1000 async_configure stage succeeded
0x9503:1:0x0005: bound  'scenes' cluster: Status.SUCCESS
0x9503:1:0x0005: finished channel configuration
 Lutron LZL4BWHL01 Remote: channel: scenes-0x9503:1:0x0005 async_configure stage succeeded
initializing channel: basic from_cache: False
 Lutron LZL4BWHL01 Remote: channel: basic-0x9503:1:0x0000 async_configure stage succeeded
0x9503:1:0x0006: bound  'on_off' cluster: Status.SUCCESS
0x9503:1:0x0006: finished channel configuration
 Lutron LZL4BWHL01 Remote: channel: event_relay-0x9503:1:0x0006 async_configure stage succeeded
0x9503:1:0x0008: bound  'level' cluster: Status.SUCCESS
0x9503:1:0x0008: finished channel configuration
 Lutron LZL4BWHL01 Remote: channel: event_relay-0x9503:1:0x0008 async_configure stage succeeded
 Lutron LZL4BWHL01 Remote: completed configuration
 Lutron LZL4BWHL01 Remote: stored in registry: ZhaDeviceEntry(name=' Lutron LZL4BWHL01 Remote', ieee='ff:ff:00:0f:e7:ff:7a:e7', last_seen=1560966054.4754078)
 Lutron LZL4BWHL01 Remote: started initialization
entry loaded from storage: ZhaDeviceEntry(name=' Lutron LZL4BWHL01 Remote', ieee='ff:ff:00:0f:e7:ff:7a:e7', last_seen=1560966054.4754078)
 Lutron LZL4BWHL01 Remote: channel: zdo- Lutron LZL4BWHL01 Remote_ZDO async_initialize stage succeeded
initializing channel: lightlink from_cache: False
 Lutron LZL4BWHL01 Remote: channel: lightlink-0x9503:1:0x1000 async_initialize stage succeeded
initializing channel: event_relay from_cache: False
 Lutron LZL4BWHL01 Remote: channel: event_relay-0x9503:1:0x0006 async_initialize stage succeeded
initializing channel: event_relay from_cache: False
 Lutron LZL4BWHL01 Remote: channel: event_relay-0x9503:1:0x0008 async_initialize stage succeeded
initializing channel: scenes from_cache: False
 Lutron LZL4BWHL01 Remote: channel: scenes-0x9503:1:0x0005 async_initialize stage succeeded
initializing channel: basic from_cache: False
 Lutron LZL4BWHL01 Remote: channel: basic-0x9503:1:0x0000 async_initialize stage succeeded
 Lutron LZL4BWHL01 Remote: power source: Battery or Unknown
 Lutron LZL4BWHL01 Remote: completed 

> initialization

Adminiuga, 2019-06-19:
> it looks like I am successful in pairing the remote using the HA ZHA panel (logs below), but when I subscribe to zha_event and hit remote buttons, I don't see anything

enable `bellows`  & `zigpy` debug logging (assuming you are using HUSBZB-1) and check if you get anything in the logs when you press buttons. 
If not, then this would require a DEV branch of HA (for zigpy/bellows groups support) and a quirk similar to Ikea 5 button remote.

br0nd, 2019-06-26:
Thanks Adminiuga, I retried with a newer dev branch of HA and it seems to be working now!

mcfarlde, 2020-05-25:
The 110 update of Home Assistant has broken my LCRs integration.  They pair fine, I can rename them and make automations and then after a reboot, the names are reset.  They all appear as Lutron LZL4BWHL01 Remote and do not show as Devices.  They can be removed, renamed and automations made again which will work until the next reboot.

tube0013, 2020-05-25:
there was a bug in 0.110.0/.1  core hat caused devices with no entities to get removed on HA start up, this has been fixed in 0.110.2, I just tested and my LCR's keep names and stay as device options in automations through HA restarts.  (Note this bug also caused issues with Ikea Signal Repeater for example).

PR #89 (2019-06-09): Add support for Lutron Connected Bulb Remote
Remote only supports binding to groups, not devices.
Add a global group id which is subscribed to by
the coordinator and used for receiving broadcasts
from the device.

Fixes #32