zigpy/zha-device-handlers #33: Request support for Quirky tripper door/window sensor

w1ll1am23, 2019-02-21:
Information obtained from debug log in Home Assistant after pairing.

```
2019-02-20 21:54:02 INFO (MainThread) [zigpy.application] Device 0xba4f (7c:e5:24:00:00:09:34:aa) joined the network
2019-02-20 21:54:02 INFO (MainThread) [zigpy.device] [0xba4f] Discovering endpoints
2019-02-20 21:54:02 DEBUG (MainThread) [zigpy.zdo] [0xba4f:zdo] ZDO request 0x0013: [47695, 7c:e5:24:00:00:09:34:aa, 128]
2019-02-20 21:54:02 DEBUG (MainThread) [zigpy.zdo] [0xba4f:zdo] ZDO request 0x0006: [65533, 260, [25], []]
2019-02-20 21:54:03 INFO (MainThread) [zigpy.device] [0xba4f] Discovered endpoints: [1]
2019-02-20 21:54:03 INFO (MainThread) [zigpy.endpoint] [0xba4f:1] Discovering endpoint information
2019-02-20 21:54:03 INFO (MainThread) [zigpy.endpoint] [0xba4f:1] Discovered endpoint information: <SimpleDescriptor endpoint=1 profile=260 device_type=1026 device_version=0 input_clusters=[0, 1, 3, 1280, 32, 2821] output_clusters=[3, 25]>
2019-02-20 21:54:04 DEBUG (MainThread) [zigpy.endpoint] [0xba4f:1] Manufacturer: Sercomm Corp.
2019-02-20 21:54:04 DEBUG (MainThread) [zigpy.endpoint] [0xba4f:1] Model: Tripper
```

dmulcahey, 2019-02-21:
Looks like it should just work, does it not work with ZHA?

w1ll1am23, 2019-02-21:
@dmulcahey sorry about that I guess I didn't wait long enough after pairing for the binary sensor to show up. Seems to be working now. However, the battery level is reporting as other, any fox for that?

dmulcahey, 2019-02-21:
All good, that will need a quirk to do correctly. I’ll see what SmartThings has for it and I’ll add it when I update the lib this weekend.

w1ll1am23, 2019-02-21:
Awesome! Want me to close this? Or leave it open as a reminder for the battery?

dmulcahey, 2019-02-21:
Leave it open

dmulcahey, 2019-02-26:
@w1ll1am23 if you have time I can show you how to do this on discord and you can PR it if you'd like?

w1ll1am23, 2019-04-22:
@dmulcahey did you ever get a chance to get battery reporting working for this?

w1ll1am23, 2019-10-10:
@dmulcahey bumping again. Any update on battery reporting for these?

earlthesquirrel, 2020-03-06:
I have a tripper set up and connected.  It shows the battery device but not a value... thoughts?

42ism, 2021-03-06:
I can confirm, the latest version of HA, just added the Quirky tripper door sensor and seeing this:
[image]

closed 2021-09-09 by github-actions[bot]