zigpy/zha-device-handlers #34: Request for support for Sylvania/Osram Lightify Smart Switch

spencerhooks, 2019-02-25:
Hopefully you don't mind the request. Apologies if I'm over sharing here, but I wanted to make sure you have what you need. Information obtained from debug log in Home Assistant after pairing. Thanks!

```
2019-02-24 21:41:22 INFO (MainThread) [homeassistant.components.zha.api] Permitting joins for 60s
2019-02-24 21:41:25 INFO (MainThread) [zigpy.application] Device 0xc2ff (00:0d:6f:00:0e:c8:ff:33) joined the network
2019-02-24 21:41:25 INFO (MainThread) [zigpy.device] [0xc2ff] Discovering endpoints
2019-02-24 21:41:25 DEBUG (MainThread) [zigpy.zdo] [0xc2ff:zdo] ZDO request 0x0013: [49919, 00:0d:6f:00:0e:c8:ff:33, 128]
2019-02-24 21:41:26 INFO (MainThread) [zigpy.device] [0xc2ff] Discovered endpoints: [1, 2, 3, 4, 5, 6]
2019-02-24 21:41:26 INFO (MainThread) [zigpy.endpoint] [0xc2ff:1] Discovering endpoint information
2019-02-24 21:41:27 INFO (MainThread) [zigpy.endpoint] [0xc2ff:1] Discovered endpoint information: <SimpleDescriptor endpoint=1 profile=260 device_type=2064 device_version=2 input_clusters=[0, 1, 32, 4096, 64768] output_clusters=[3, 4, 5, 6, 8, 25, 768, 4096]>
2019-02-24 21:41:28 DEBUG (MainThread) [zigpy.endpoint] [0xc2ff:1] Manufacturer: OSRAM
2019-02-24 21:41:28 DEBUG (MainThread) [zigpy.endpoint] [0xc2ff:1] Model: Switch 4x-LIGHTIFY
2019-02-24 21:41:28 INFO (MainThread) [zigpy.endpoint] [0xc2ff:2] Discovering endpoint information
2019-02-24 21:41:29 INFO (MainThread) [zigpy.endpoint] [0xc2ff:2] Discovered endpoint information: <SimpleDescriptor endpoint=2 profile=260 device_type=2064 device_version=2 input_clusters=[0, 4096, 64768] output_clusters=[3, 4, 5, 6, 8, 768, 4096]>
2019-02-24 21:41:30 DEBUG (MainThread) [zigpy.endpoint] [0xc2ff:2] Manufacturer: OSRAM
2019-02-24 21:41:30 DEBUG (MainThread) [zigpy.endpoint] [0xc2ff:2] Model: Switch 4x-LIGHTIFY
2019-02-24 21:41:30 INFO (MainThread) [zigpy.endpoint] [0xc2ff:3] Discovering endpoint information
2019-02-24 21:41:31 INFO (MainThread) [zigpy.endpoint] [0xc2ff:3] Discovered endpoint information: <SimpleDescriptor endpoint=3 profile=260 device_type=2064 device_version=2 input_clusters=[0, 4096, 64768] output_clusters=[3, 4, 5, 6, 8, 768, 4096]>
2019-02-24 21:41:32 DEBUG (MainThread) [zigpy.endpoint] [0xc2ff:3] Manufacturer: OSRAM
2019-02-24 21:41:32 DEBUG (MainThread) [zigpy.endpoint] [0xc2ff:3] Model: Switch 4x-LIGHTIFY
2019-02-24 21:41:32 INFO (MainThread) [zigpy.endpoint] [0xc2ff:4] Discovering endpoint information
2019-02-24 21:41:33 INFO (MainThread) [zigpy.endpoint] [0xc2ff:4] Discovered endpoint information: <SimpleDescriptor endpoint=4 profile=260 device_type=2064 device_version=2 input_clusters=[0, 4096, 64768] output_clusters=[3, 4, 5, 6, 8, 768, 4096]>
2019-02-24 21:41:34 DEBUG (MainThread) [zigpy.endpoint] [0xc2ff:4] Manufacturer: OSRAM
2019-02-24 21:41:34 DEBUG (MainThread) [zigpy.endpoint] [0xc2ff:4] Model: Switch 4x-LIGHTIFY
2019-02-24 21:41:34 INFO (MainThread) [zigpy.endpoint] [0xc2ff:5] Discovering endpoint information
2019-02-24 21:41:35 INFO (MainThread) [zigpy.endpoint] [0xc2ff:5] Discovered endpoint information: <SimpleDescriptor endpoint=5 profile=260 device_type=2064 device_version=2 input_clusters=[0, 4096, 64768] output_clusters=[3, 4, 5, 6, 8, 768, 4096]>
2019-02-24 21:41:36 DEBUG (MainThread) [zigpy.endpoint] [0xc2ff:5] Manufacturer: OSRAM
2019-02-24 21:41:36 DEBUG (MainThread) [zigpy.endpoint] [0xc2ff:5] Model: Switch 4x-LIGHTIFY
2019-02-24 21:41:36 INFO (MainThread) [zigpy.endpoint] [0xc2ff:6] Discovering endpoint information
2019-02-24 21:41:37 INFO (MainThread) [zigpy.endpoint] [0xc2ff:6] Discovered endpoint information: <SimpleDescriptor endpoint=6 profile=260 device_type=2064 device_version=2 input_clusters=[0, 4096, 64768] output_clusters=[3, 4, 5, 6, 8, 768, 4096]>
2019-02-24 21:41:38 DEBUG (MainThread) [zigpy.endpoint] [0xc2ff:6] Manufacturer: OSRAM
2019-02-24 21:41:38 DEBUG (MainThread) [zigpy.endpoint] [0xc2ff:6] Model: Switch 4x-LIGHTIFY
2019-02-24 21:41:38 DEBUG (MainThread) [zigpy.quirks] Considering <class 'zigpy.quirks.xiaomi.TemperatureHumiditySensor'>
2019-02-24 21:41:38 DEBUG (MainThread) [zigpy.quirks] Fail because endpoint list mismatch: dict_keys([1, 2, 3]) {1, 2, 3, 4, 5, 6}
2019-02-24 21:41:38 DEBUG (MainThread) [zigpy.quirks] Considering <class 'zigpy.quirks.smartthings.SmartthingsTemperatureHumiditySensor'>
2019-02-24 21:41:38 DEBUG (MainThread) [zigpy.quirks] Fail because endpoint list mismatch: dict_keys([1]) {1, 2, 3, 4, 5, 6}
2019-02-24 21:41:39 DEBUG (MainThread) [zigpy.quirks] Considering <class 'zigpy.quirks.kof.CeilingFan'>
2019-02-24 21:41:39 DEBUG (MainThread) [zigpy.quirks] Fail because endpoint list mismatch: dict_keys([1]) {1, 2, 3, 4, 5, 6}
2019-02-24 21:41:39 DEBUG (MainThread) [zigpy.quirks] Considering <class 'zhaquirks.centralite.3130.CentraLite3130'>
2019-02-24 21:41:39 DEBUG (MainThread) [zigpy.quirks] Fail because endpoint list mismatch: dict_keys([1]) {1, 2, 3, 4, 5, 6}
2019-02-24 21:41:39 DEBUG (MainThread) [zigpy.quirks] Considering <class 'zhaquirks.centralite.3300S.CentraLite3300S'>
2019-02-24 21:41:39 DEBUG (MainThread) [zigpy.quirks] Fail because endpoint list mismatch: dict_keys([1, 2]) {1, 2, 3, 4, 5, 6}
2019-02-24 21:41:39 DEBUG (MainThread) [zigpy.quirks] Considering <class 'zhaquirks.centralite.3305S.CentraLite3305S'>
2019-02-24 21:41:39 DEBUG (MainThread) [zigpy.quirks] Fail because endpoint list mismatch: dict_keys([1, 2]) {1, 2, 3, 4, 5, 6}
2019-02-24 21:41:39 DEBUG (MainThread) [zigpy.quirks] Considering <class 'zhaquirks.centralite.3315S.CentraLite3315S'>
2019-02-24 21:41:39 DEBUG (MainThread) [zigpy.quirks] Fail because endpoint list mismatch: dict_keys([1, 2]) {1, 2, 3, 4, 5, 6}
2019-02-24 21:41:39 DEBUG (MainThread) [zigpy.quirks] Considering <class 'zhaquirks.centralite.3320.CentraLite3320'>
2019-02-24 21:41:39 DEBUG (MainThread) [zigpy.quirks] Fail because endpoint list mismatch: dict_keys([1, 2]) {1, 2, 3, 4, 5, 6}
2019-02-24 21:41:39 DEBUG (MainThread) [zigpy.quirks] Considering <class 'zhaquirks.centralite.3326.CentraLite3326'>
2019-02-24 21:41:39 DEBUG (MainThread) [zigpy.quirks] Fail because endpoint list mismatch: dict_keys([1, 2]) {1, 2, 3, 4, 5, 6}
2019-02-24 21:41:39 DEBUG (MainThread) [zigpy.quirks] Considering <class 'zhaquirks.centralite.motion.CentraLiteMotionSensor'>
2019-02-24 21:41:39 DEBUG (MainThread) [zigpy.quirks] Fail because endpoint list mismatch: dict_keys([1, 2]) {1, 2, 3, 4, 5, 6}
2019-02-24 21:41:39 DEBUG (MainThread) [zigpy.quirks] Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
2019-02-24 21:41:39 DEBUG (MainThread) [zigpy.quirks] Fail because endpoint list mismatch: dict_keys([1]) {1, 2, 3, 4, 5, 6}
2019-02-24 21:41:39 DEBUG (MainThread) [zigpy.quirks] Considering <class 'zhaquirks.philips.rwl021.PhilipsRWL021'>
2019-02-24 21:41:39 DEBUG (MainThread) [zigpy.quirks] Fail because endpoint list mismatch: dict_keys([1, 2]) {1, 2, 3, 4, 5, 6}
2019-02-24 21:41:39 DEBUG (MainThread) [zigpy.quirks] Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
2019-02-24 21:41:39 DEBUG (MainThread) [zigpy.quirks] Fail because endpoint list mismatch: dict_keys([1]) {1, 2, 3, 4, 5, 6}
2019-02-24 21:41:39 DEBUG (MainThread) [zigpy.quirks] Considering <class 'zhaquirks.xiaomi.aqara_body_sensor.AqaraBodySensor'>
2019-02-24 21:41:39 DEBUG (MainThread) [zigpy.quirks] Fail because endpoint list mismatch: dict_keys([1]) {1, 2, 3, 4, 5, 6}
2019-02-24 21:41:39 DEBUG (MainThread) [zigpy.quirks] Considering <class 'zhaquirks.xiaomi.aqara_button.AqaraButton'>
2019-02-24 21:41:39 DEBUG (MainThread) [zigpy.quirks] Fail because endpoint list mismatch: dict_keys([1]) {1, 2, 3, 4, 5, 6}
2019-02-24 21:41:39 DEBUG (MainThread) [zigpy.quirks] Considering <class 'zhaquirks.xiaomi.aqara_cube.AqaraCube'>
2019-02-24 21:41:39 DEBUG (MainThread) [zigpy.quirks] Fail because endpoint list mismatch: dict_keys([1, 2, 3]) {1, 2, 3, 4, 5, 6}
2019-02-24 21:41:39 DEBUG (MainThread) [zigpy.quirks] Considering <class 'zhaquirks.xiaomi.aqara_leak_sensor.AqaraLeakSensor'>
2019-02-24 21:41:39 DEBUG (MainThread) [zigpy.quirks] Fail because endpoint list mismatch: dict_keys([1]) {1, 2, 3, 4, 5, 6}
2019-02-24 21:41:39 DEBUG (MainThread) [zigpy.quirks] Considering <class 'zhaquirks.xiaomi.aqara_open_close_sensor.AqaraOpenCloseSensor'>
2019-02-24 21:41:39 DEBUG (MainThread) [zigpy.quirks] Fail because endpoint list mismatch: dict_keys([1]) {1, 2, 3, 4, 5, 6}
2019-02-24 21:41:39 DEBUG (MainThread) [zigpy.quirks] Considering <class 'zhaquirks.xiaomi.aqara_temperature_humidity_sensor.AqaraTemperatureHumiditySensor'>
2019-02-24 21:41:39 DEBUG (MainThread) [zigpy.quirks] Fail because endpoint list mismatch: dict_keys([1]) {1, 2, 3, 4, 5, 6}
2019-02-24 21:41:39 DEBUG (MainThread) [zigpy.quirks] Considering <class 'zhaquirks.xiaomi.aqara_vibration_sensor.AqaraVibrationSensor'>
2019-02-24 21:41:39 DEBUG (MainThread) [zigpy.quirks] Fail because endpoint list mismatch: dict_keys([1, 2]) {1, 2, 3, 4, 5, 6}
```

dlashua, 2019-02-25:
There are two versions of this device seen in the wild. The European version is fairly easy to support. The US version, which your logs appear to be from, is particularly difficult and will require more reverse engineering of the device.

spencerhooks, 2019-02-25:
Bummer. I'm not much of a developer, but is there anything I can do to help?

dmulcahey, 2019-02-25:
@spencerhooks if this is pairing (looks like it pairs fine) you should see zha_events for these already without a quirk. Go to the event tool in the UI (introduced in .88) in HA and subscribe to zha_event. Then interact with the device. You should see events show up for each on/off and each level control action (dimming)

spencerhooks, 2019-02-25:
Ok, thanks. I'll try the event tool to see what I see. I had tried using the binary sensors to create automations, but they were't working - either not reliably or not at all depending on the button.

dlashua, 2019-02-25:
@spencerhooks This device doesn't conform to "zigbee standards". You'll find the top left button fires an "ON" event on Endpoint 1. And the bottom left button fires an "OFF" event on Endpoint 3.

The Top Right and Bottom Right buttons are indicated on Endpoints 2 and 4, respectively. But, without a proper quirks implementation, you'll see no event for these buttons at all.

Additionally, you may see odd behavior with this device. If Endpoints 1 and 3 are not bound to properly during the initialization (the device goes out of range or something during this process), the device will be joined to the network but the endpoints won't be bound. The default for this device, when the endpoints are not bound, is to turn OFF and ON ALL lights. So if you see all your other zigbee lights turning on and off, that's why.

I've been working on a device handler for this device for a few days now. Unfortunately, I can't find good documentation on Zigpy and I'm unable to find examples of other devices that operate the way this one does, so the development is slow going. Which is a shame because these devices are quite inexpensive, fairly attractive, and offer quite a bit of functionality.

dmulcahey, 2019-02-25:
@dlashua HA will get events for all occurrences of clusters 6 and 8 that are output clusters without a quirk. This device should be emitting events as is.

dmulcahey, 2019-02-25:
Have a link to amazon for this device? I’ll pick one up to experiment with too.

spencerhooks, 2019-02-25:
Here's where I ordered from: https://www.amazon.com/gp/product/B01M3OK6J7

I totally agree about the device being cheap, attractive, etc.

dlashua, 2019-02-25:
https://www.amazon.com/SYLVANIA-LIGHTIFY-Function-Gateway-Assistant/dp/B01M3OK6J7/ref=sr_1_2?keywords=sylvania+4+button&qid=1551118583&s=gateway&sr=8-2

When you get it and you try to pair, look at the model reported by the device. If it states "Switch 4x-LIGHTIFY" then you have the same one I do. If it states "Switch 4x EU-LIGHTIFY" then you have the much easier to support device that I can't seem to get my hands on.

dlashua, 2019-02-25:
I'm also having a hard time figuring out how to write a quirk for this device, because I've yet to find another quirk for a device like it to use as an example.

There are three ways to represent the device... 

1) as four buttons that have no state and generate events to indicate when they are pressed, held, and released. With this method, HASS entities would be only a battery sensor. Everything else would be an event.

2) As something similar to the hue dimmer switch, but two of them. HASS would see two entities, plus a battery sensor. The left side of the device would be one entity (a binary sensor if done as the hue dimmer is) that can be "on" or "off" and has a brightness attribute. And the right side would be another entity just like it.

3) As LIGHTIFY intended it: the left side of the device controlling on and off, and the right side of the device controlling hue. With this, HASS would see two entities, a binary sensor with an on and off state, a brightness attribute and a hue attribute. The device handler would be responsible for keeping track of the brightness level and hue value and altering it with each button press/hold.

The trouble I'm having is with two things:

1) To create two entities (like option 2 above), I would need to combine data from two endpoints into a single entity. I have no idea what this looks like in quirks code. With each endpoint on the device ending up as an entity, as is, we see 6 entities (plus a battery sensor). One of those entities only ever turns on (never off). And another only turns off (never on).

2) How to get "Color" from zigpy.zcl.clusters.lighting to do whatever thing is needed to show as an entity in HASS.

dlashua, 2019-02-25:
> @dlashua HA will get events for all occurrences of clusters 6 and 8 that are output clusters without a quirk. This device should be emitting events as is.

Yes... and, while it looks funny in HA, because of this, the top left and bottom left buttons are "usable", just weird. The top right and bottom right buttons, however, seem to report on cluster 0x0300 (decimal 768) (from what I've seen in other places).  Events don't seem to be emitted for it. So, either this doesn't happen without a quirk, or, the device doesn't report changes on thost clusters without preconfiguration. And performing that preconfiguration is another difficult point I'm having because I don't see any other devices needing this.

Adminiuga, 2019-02-25:
0x300/768 is the color cluster. Try adding `EVENT_RELAY_CLUSTERS.append(zcl.clusters.lighting.Color.cluster_id)` to  [components/zha/core/gateway.py](https://github.com/home-assistant/home-assistant/blob/6626e5c4a4ac8684329298ef28a83a2175f9aafa/homeassistant/components/zha/core/gateway.py#L443)

dlashua, 2019-02-25:
Doing so causes HA to hang on start up. Despite having 

````
logger:
  default: debug
````

There is no useful error message. Simply...

````
Feb 25 13:16:13 maverick systemd[1]: Started Home Assistant.

Feb 25 13:16:15 maverick hass[21000]: 2019-02-25 13:16:15 INFO (MainThread) [homeassistant.loader] Loaded sensor from homeassistant.components.sensor
Feb 25 13:16:15 maverick hass[21000]: 2019-02-25 13:16:15 INFO (MainThread) [homeassistant.loader] Loaded input_boolean from homeassistant.components.input_boolean
Feb 25 13:16:15 maverick hass[21000]: 2019-02-25 13:16:15 INFO (MainThread) [homeassistant.loader] Loaded timer from homeassistant.components.timer
Feb 25 13:16:15 maverick hass[21000]: 2019-02-25 13:16:15 INFO (MainThread) [homeassistant.loader] Loaded cast from homeassistant.components.cast
Feb 25 13:16:15 maverick hass[21000]: 2019-02-25 13:16:15 INFO (MainThread) [homeassistant.loader] Loaded logger from homeassistant.components.logger
Feb 25 13:16:15 maverick hass[21000]: 2019-02-25 13:16:15 INFO (MainThread) [homeassistant.loader] Loaded updater from homeassistant.components.updater
Feb 25 13:16:15 maverick hass[21000]: 2019-02-25 13:16:15 INFO (MainThread) [homeassistant.loader] Loaded xiaomi_aqara from homeassistant.components.xiaomi_aqara
Feb 25 13:16:15 maverick hass[21000]: 2019-02-25 13:16:15 INFO (MainThread) [homeassistant.loader] Loaded input_number from homeassistant.components.input_number
Feb 25 13:16:15 maverick hass[21000]: 2019-02-25 13:16:15 INFO (MainThread) [homeassistant.loader] Loaded openuv from homeassistant.components.openuv
Feb 25 13:16:15 maverick hass[21000]: 2019-02-25 13:16:15 INFO (MainThread) [homeassistant.loader] Loaded notify from homeassistant.components.notify
Feb 25 13:16:15 maverick hass[21000]: 2019-02-25 13:16:15 INFO (MainThread) [homeassistant.loader] Loaded recorder from homeassistant.components.recorder
...
Feb 25 13:16:15 maverick hass[21000]: 2019-02-25 13:16:15 INFO (MainThread) [homeassistant.loader] Loaded camera from homeassistant.components.camera
Feb 25 13:16:15 maverick hass[21000]: 2019-02-25 13:16:15 INFO (MainThread) [homeassistant.loader] Loaded alarm_control_panel from homeassistant.components.alarm_control_panel
Feb 25 13:16:15 maverick hass[21000]: 2019-02-25 13:16:15 INFO (MainThread) [homeassistant.loader] Loaded google_assistant from homeassistant.components.google_assistant
Feb 25 13:16:15 maverick hass[21000]: Config directory: /home/hass/run/homeassistant-config
````

And then it sits there... no more logs... and never fully starts.

dlashua, 2019-02-25:
> And then it sits there... no more logs... and never fully starts.

Nevermind... I found a better place to add it in the file and it seems to start now. Since it was just setting a value, I assumed it was safe to add this line at the top of the file. It was not. :)

dlashua, 2019-02-25:
> 0x300/768 is the color cluster. Try adding `EVENT_RELAY_CLUSTERS.append(zcl.clusters.lighting.Color.cluster_id)` to [components/zha/core/gateway.py](https://github.com/home-assistant/home-assistant/blob/6626e5c4a4ac8684329298ef28a83a2175f9aafa/homeassistant/components/zha/core/gateway.py#L443)

Thanks for taking a stab at it. Sadly, that didn't help. 

The device doesn't respond well to the initializing process. Lots of network errors and it takes multiple "reconfigures" before I see that every bind and report command made it through at least once. But even once that's done, the right side buttons generate no attribute reports or change events that I can see. 

Endpoints 1 and 3 control ALL lights until they are bound to the controller. Then they operate in an odd, but usable manner, both press and hold for each button. But the right side buttons may require some additional configuration before they will work. I don't have any lights that accept a binding to 0x0300/768, so I can't tell if they are controlling everything as well and the binding just didn't take or if there's something else.

Pairing this device with it's native controller (the Osram Lightify Hub) I can see those two buttons are "disabled" by default. I've tried enabling them there, but, when I pair the device back to ZHA it seems to go back to its original settings.

I'm guessing there are some manufacturer specific things that need to be done to bind the buttons to the controller. And a regular old "bind" isn't enough.

dmulcahey, 2019-02-25:
I’m gonna pick one up. I’ll let you know what I find out.

dmulcahey, 2019-02-26:
the device arrives Thursday... I'll see what I can do from a quirks perspective. That said, if anyone wants to be brave I have branches of the backend and frontend of HA in flight to support device binding: 

backend: https://github.com/dmulcahey/home-assistant/tree/dm/zha-direct-binding

frontend:  https://github.com/dmulcahey/home-assistant-polymer/tree/dm/zha-direct-device-binding

This will let you directly pair the remote to lights and use it without quirks, events or automations

dmulcahey, 2019-03-03:
> Yes... and, while it looks funny in HA, because of this, the top left and bottom left buttons are "usable", just weird. The top right and bottom right buttons, however, seem to report on cluster [...]

the color clusters don't report at all... the device is kinda wacky. Binding is the way to go for this. It is already merged to dev. Closing because there is nothing useful that can really be done.

closed 2019-03-03 by dmulcahey

spencerhooks, 2019-03-04:
Thanks for picking up the device to take a look.

Hopefully this isn't a dumb question, but where will device binding show up when it is released? ZHA configuration?

dmulcahey, 2019-03-04:
The config panel in the UI

spencerhooks, 2019-03-04:
Got it. Thanks again!

rgthree, 2019-04-04:
Sorry to ping a closed thread, but I noticed "zhaquirks/osram/lightifyx4.py" was added a few days _after_ this was closed and wondered if that meant this device should be supported after all?

FWIW, I have purchased one, and it's assigned the 
zhaquirks.osram.lightifyx4.LightifyX4 quirk in HA, but doesn't work as expected--6 binary_sensors are added as "device_class: opening" and clicking any of the 4 buttons turns the first sensor to on/open and then it just stays like that. Any further presses, holds, etc. don't change any other states, or that first state back to off.

dmulcahey, 2019-04-04:
It was an experiment. Binding is the best way to use this.

rgthree, 2019-04-05:
Bummer... I was hoping to use it to manually trigger non-zha automations in HA.

Thanks for all the work anyway!

DaAwesomeP, 2019-06-15:
>It was an experiment. Binding is the best way to use this.

What do you mean by binding?

DaAwesomeP, 2019-06-16:
Also, not sure which config panel I'm looking for. I can't get it ZHA to add the device.

BarrettLowe, 2019-07-16:
I am able to bind the Lightify 4x switch with a device. Seems like only the buttons on the left side do anything though. Single press top and bottom buttons turns the bound device on and off. Holding top and bottom fades up and down respectively. I can't get buttons on the right size to do anything though. Any thoughts?

youareme7, 2019-07-17:
> I am able to bind the Lightify 4x switch with a device. Seems like only the buttons on the left side do anything though. Single press top and bottom buttons turns the bound device on and off. [...]

I can't even get the Lightify 4x switch to bind to anything it fails in the attempt according to my log, I have the switch added to ZHA and the left side reports events for tap and for hold which seems to be the norm that the right side switches don't work with this implementation. Means it's half functional to me which isn't too bad given the super low price but it's a shame that the right side doesn't behave too.
```
2019-07-17 14:36:41 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection.140114964268200] Error handling message: Unknown error
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/decorators.py", line 17, in _handle_async_response
    await func(hass, connection, msg)
  File "/usr/src/homeassistant/homeassistant/components/zha/api.py", line 386, in websocket_bind_devices
    zha_gateway, source_ieee, target_ieee, BIND_REQUEST)
  File "/usr/src/homeassistant/homeassistant/components/zha/api.py", line 447, in async_binding_operation
    await asyncio.gather(*bind_tasks)
  File "/usr/local/lib/python3.7/site-packages/zigpy/device.py", line 147, in request
    expect_reply=expect_reply,
  File "/usr/local/lib/python3.7/site-packages/bellows/zigbee/application.py", line 365, in request
    res = await asyncio.wait_for(req.send, timeout=APS_ACK_TIMEOUT)
  File "/usr/local/lib/python3.7/asyncio/tasks.py", line 416, in wait_for
    return fut.result()
zigpy.exceptions.DeliveryError: [0xe405:0:0x0021]: message send failure: EmberStatus.DELIVERY_FAILED
```