zigpy/zha-device-handlers #36: What to do if quirks do not get applied?

glassbase, 2019-02-25:
[My comment](https://github.com/home-assistant/home-assistant/issues/21345#issuecomment-466835453) from an issue over in HA github.

HA 0.88.1, Conbee stick.

What do we do to get the quirks to apply? I have 7 Xiaomi temp sensors (all the same version/model). 5 have quirks, 2 do not. 
1. I removed the 2 from HA / ZHA page
2. Double checked the zigbee.db and they were removed. 
3. Restart HA
4. Permit/Added one sensor to HA
5. Shows as "None None" node but has the quirks (same as my other 5)
6. Restart HA
7. Node name is now "LUMI lumi.weather" but has quirks "zigpy.device.Device" (which I assume if default?)

I also have 2 vibration sensors with same issue. 

Should this issue be here or HA or zigpy-deconz?

dmulcahey, 2019-02-25:
Here for now. The issue is either here or in HA. I’ll hopefully have more info tonight.

dmulcahey, 2019-02-26:
@glassbase do me a favor. Can you dump the devices from zigbee.db so that I can see what they have for signatures?

glassbase, 2019-02-26:
@dmulcahey Sure, I will re-add the temp sensors and dump in the next little bit.

glassbase, 2019-02-26:
So annoying but both temp sensors added with quirks this time (but not the other 10000000 times I tried over the last few days). 

... Moving on to vibration sensors.

glassbase, 2019-02-26:
And vibration sensors added fine this time too.... so weird. 

I think the only thing different was:
1. ensure removed node from HA
2. restart HA 
3. check zigbee.db to ensure removed
4. hold pair button for 5 sec and ensure light blinks back only ONCE to ensure it is reset/default (repeat, if it blinks multiple times)
5. HA permit
6. pair device

dmulcahey, 2019-02-26:
@glassbase good to close this?

glassbase, 2019-02-26:
Yes. Thanks

closed 2019-02-26 by glassbase