zigpy/zha-device-handlers #39: Support for lumi.lock.v1

Nemesis24, 2019-03-08:
It will be very fine to support Xiaomi lumi.lock.v1.
I can help if you explain me !
Thank you.

dmulcahey, 2019-04-30:
We don't have support for the lock domain yet in HA.

Nemesis24, 2019-04-30:
This lock is a zigbee "sensor" and not serve as lock.

dmulcahey, 2019-04-30:
I’m confused... what type of device are we actually talking about? If it is really a door lock we’ll wait to support it correctly rather than hack it as a sensor.

Nemesis24, 2019-04-30:
This device is a simple lock like you've got on all the doors...
But it can detect what key is inserted and it's got 5 keys.
It can detect too if a wrong key or disbled key was inserted.

dmulcahey, 2019-04-30:
Have a link to the product?

Nemesis24, 2019-04-30:
https://www.youtube.com/watch?v=0pGiTnyLpU4

Nemesis24, 2020-09-24:
Perhaps some good news https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1977#issuecomment-698438985

pwet45, 2020-12-13:
Hello everyone, let me revive the subject. can you integrate lumi.lock.v1 in ZHA? Please
Tell me what you need.
thank you in advance

pwet45, 2020-12-23:
Could you integrate this lock please?

A link that can help you, in any case I hope

https://easydomoticz.com/forum/viewtopic.php?f=28&t=10587

Fenix999, 2021-06-21:
> We don't have support for the lock domain yet in HA.

According to the integrations page Locks are supported
https://www.home-assistant.io/integrations/zha/

I don't know if I can do anything to help, but if I can...

ec-blaster, 2021-07-24:
I'm also interested in this integration.
As a matter of fact, this is the only device I'm missing to get to ZHA from Zigbee2Mqtt.
I can contribute if you guide me.
Thanks!

adi90x, 2022-02-01:
Hello, 

Any news on this ? 

Regards

Nemesis24, 2022-02-02:
I hope it could be integrate because this is the only device i can't connect on my Home assistant
If i can help say me something.
Thanks for all.

adi90x, 2022-02-11:
> I'm also interested in this integration. As a matter of fact, this is the only device I'm missing to get to ZHA from Zigbee2Mqtt. I can contribute if you guide me. Thanks!

Is it working fine on zigbee2mqtt ?

Nemesis24, 2022-02-12:
https://zigbee.blakadder.com/Xiaomi_Vima_A6121.html
This link says it's working.

adi90x, 2022-03-15:
Any news on that ? Any chance to have it working in zha ?

Nemesis24, 2022-05-24:
I would have this integrated in my home assistant, what can i do for helping ? Thanks for your answer.

Nemesis24, 2022-09-16:
Hi, any news about that, how can i help ?

closed 2023-03-22 by github-actions[bot]

Drun555, 2023-12-22:
Zigbee2MQTT has support for this lock for ages - so I'll stick to that for a time.

https://www.zigbee2mqtt.io/devices/A6121.html