zigpy/zha-device-handlers #43: Samjin button signature

Socalix, 2019-03-11:
Samjin button apparently has versions with and without `DIAGNOSTICS_CLUSTER_ID` in the input_clusters here:
https://github.com/dmulcahey/zha-device-handlers/blob/master/zhaquirks/samjin/button.py#L69

Mine happens to be without it, so the quirk is not being applied by HA.
Is there a way to define 2 possible signatures or would we need to have a copy of this quirk with a different signature?

dmulcahey, 2019-03-11:
Not yet, but I am pushing for it

ryanwinter, 2019-03-27:
I also have the button without the DIAGNOSTICS_CLUSTER_ID as an input. Should I submit a PR with the additional button signiture?

dmulcahey, 2019-03-27:
Nah, I’ll add it to the next release

dmulcahey, 2019-03-27:
#55 will be in next release. Please test it if you can and ensure it works.

closed 2019-03-27 by dmulcahey

Socalix, 2019-03-29:
@dmulcahey #55 looks good. Thank you!