zigpy/zha-device-handlers #44: Add lumi.sensor_86sw1 (WXKG03LM) support

vmanuel, 2019-03-12:


dmulcahey, 2019-04-30:
Can't do much without information or a device or a tester w/ a device. If you want this worked on please open a new issue with the simple descriptors from the device and clearly articulate what needs to be added to make this function. Meaning what doesn't work in HA that the quirk would need to provide as far as functionality is concerned.

closed 2019-04-30 by dmulcahey

vmanuel, 2019-06-06:
Sorry... I had no time to give more info...

I made it work. I created aqara_WXKG03LM.py:

`"""Xiaomi aqara WXKG03LM sensor."""
import logging
from zigpy.profiles import zha
from zigpy.zcl.clusters.general import Basic, Groups, OnOff
from zhaquirks.xiaomi import BasicCluster, PowerConfigurationCluster,\
    TemperatureMeasurementCluster, XiaomiCustomDevice

BUTTON_DEVICE_TYPE = 0x5F01
XIAOMI_CLUSTER_ID = 0xFFFF

_LOGGER = logging.getLogger(__name__)

class WXKG03LM(XiaomiCustomDevice):
    """Aqara WXKG03LM device."""

    signature = {
        # <SimpleDescriptor endpoint=1 profile=260 device_type=24321
        # device_version=1
        # input_clusters=[0, 6, 65535]
        # output_clusters=[0, 4, 65535]>
        1: {
            'manufacturer': 'LUMI',
            'model': 'lumi.sensor_86sw1',
            'profile_id': zha.PROFILE_ID,
            'device_type': BUTTON_DEVICE_TYPE,
            'input_clusters': [
                Basic.cluster_id,
                OnOff.cluster_id,
                XIAOMI_CLUSTER_ID
            ],
            'output_clusters': [
                Basic.cluster_id,
                Groups.cluster_id,
                XIAOMI_CLUSTER_ID
            ],
        },
    }

    replacement = {
        'endpoints': {
            1: {
                'manufacturer': 'LUMI',
                'model': 'lumi.sensor_86sw1',
                'device_type': zha.DeviceType.REMOTE_CONTROL,
                'input_clusters': [
                    BasicCluster,
                    PowerConfigurationCluster,
                    TemperatureMeasurementCluster,
                    XIAOMI_CLUSTER_ID
                ],
                'output_clusters': [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    XIAOMI_CLUSTER_ID,
                    OnOff.cluster_id,
                ],
            }
        },
    }`

This gadget appears like a binary sensor in HomeAssistant. If I subscribe to a zha_event with its device_ieee, i can control a light (toggle light).

vmanuel, 2019-06-30:
I updated the code:

`
"""Xiaomi aqara single key switch device."""
import logging

from zigpy.profiles import zha
from zigpy.zcl.clusters.general import (
    AnalogInput, Basic, Groups, Identify, MultistateInput, Ota, Scenes, OnOff)

from zhaquirks import CustomCluster
from zhaquirks.xiaomi import (
    BasicCluster, PowerConfigurationCluster, XiaomiCustomDevice
)

XIAOMI_DEVICE_TYPE = 0x5F01
XIAOMI_DEVICE_TYPE2 = 0x5F02
XIAOMI_DEVICE_TYPE3 = 0x5F03
XIAOMI_CLUSTER_ID = 0xFFFF
STATUS_TYPE_ATTR = 0x0055  # decimal = 85

PRESS_TYPE = {
    0: 'long press',
    1: 'single',
    2: 'double'
}

_LOGGER = logging.getLogger(__name__)

class Remote86sw1(XiaomiCustomDevice):
    """Aqara single key switch device."""

    class MultistateInputCluster(CustomCluster, MultistateInput):
        """Multistate input cluster."""

        cluster_id = MultistateInput.cluster_id

        def __init__(self, *args, **kwargs):
            """Init."""
            self._current_state = None
            super().__init__(*args, **kwargs)

        def _update_attribute(self, attrid, value):
            super()._update_attribute(attrid, value)
            if attrid == STATUS_TYPE_ATTR:
                self._current_state = PRESS_TYPE.get(value)
                event_args = {
                    'press_type': self._current_state,
                    'attr_id': attrid,
                    'value': value
                }
                self.listener_event(
                    'zha_send_event',
                    self,
                    self._current_state,
                    event_args
                )
                # show something in the sensor in HA
                super()._update_attribute(
                    0,
                    self._current_state
                )

    signature = {
        # <SimpleDescriptor endpoint=1 profile=260 device_type=24321
        # device_version=1
        # input_clusters=[0, 3, 25, 65535, 18]
        # output_clusters=[0, 4, 3, 5, 25, 65535, 18]>
        'models_info': [
            ('LUMI', 'lumi.sensor_86sw1')
        ],
        'endpoints': {
            1: {
                'profile_id': zha.PROFILE_ID,
                'device_type': XIAOMI_DEVICE_TYPE,
                'input_clusters': [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Ota.cluster_id,
                    XIAOMI_CLUSTER_ID,
                    MultistateInputCluster.cluster_id
                ],
                'output_clusters': [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    Ota.cluster_id,
                    XIAOMI_CLUSTER_ID,
                    MultistateInputCluster.cluster_id
                ],
            },
            # <SimpleDescriptor endpoint=2 profile=260 device_type=24322
            # device_version=1
            # input_clusters=[3, 18]
            # output_clusters=[4, 3, 5, 18]>
            2: {
                'profile_id': zha.PROFILE_ID,
                'device_type': XIAOMI_DEVICE_TYPE2,
                'input_clusters': [
                    Identify.cluster_id,
                    MultistateInputCluster.cluster_id
                ],
                'output_clusters': [
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    MultistateInputCluster.cluster_id
                ],
            },
            # <SimpleDescriptor endpoint=3 profile=260 device_type=24323
            # device_version=1
            # input_clusters=[3, 12]
            # output_clusters=[4, 3, 5, 12]>
            3: {
                'profile_id': zha.PROFILE_ID,
                'device_type': XIAOMI_DEVICE_TYPE3,
                'input_clusters': [
                    Identify.cluster_id,
                    AnalogInput.cluster_id
                ],
                'output_clusters': [
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    AnalogInput.cluster_id
                ],
            },
        }
    }

    replacement = {
        'endpoints': {
            1: {
                'input_clusters': [
                    BasicCluster,
                    Identify.cluster_id,
                    PowerConfigurationCluster,
                    XIAOMI_CLUSTER_ID
                ],
                'output_clusters': [
                    BasicCluster,
                    Groups.cluster_id,
                    OnOff.cluster_id,
                    XIAOMI_CLUSTER_ID
                ],
            }
        },
    }
`

dmulcahey, 2019-06-30:
Please open a PR if you would like this device added.