zigpy/zha-device-handlers #45: Support for lumi.sensor_smoke

Nemesis24, 2019-03-15:
t will be very fine to support lumi.sensor_smoke
I can help if you explain me !
Thank you.

Adminiuga, 2019-03-15:
Do you have a link to product? 
Xiaomi smoke detectors work out of the box though, unless anything changed with the latest release.

Nemesis24, 2019-03-15:
https://www.gearbest.com/alarm-systems/pp_615081.html?lkid=10368929

Yes it works fine, but we can have the battery type, and battery %, .

dmulcahey, 2019-04-30:
Please provide the simple descriptors for the device.

prairiesnpr, 2019-04-30:
This should be fixed by https://github.com/dmulcahey/zha-device-handlers/pull/46

dmulcahey, 2019-04-30:
@Nemesis24 please confirm

Nemesis24, 2019-05-01:
It's ok !!

closed 2019-05-01 by Nemesis24

Nemesis24, 2019-05-03:
> Please provide the simple descriptors for the device.

<SimpleDescriptor endpoint=1 profile=260 device_type=1026 device_version=1 input_clusters=[0, 3, 18, 1280, 12, 1] output_clusters=[25]>