zigpy/zha-device-handlers #51: Support for lumi.ctrl_ln2.aq1

Nemesis24, 2019-03-20:
If it's possible to have Energy consumption of the day ... energy

Nemesis24, 2019-03-20:
I found this if it could help .... https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/converters/fromZigbee.js
```
QBKG12LM_power: {
        cid: 'genBasic',
        type: ['attReport', 'readRsp'],
        convert: (model, msg, publish, options) => {
            if (msg.data.data['65281']) {
                const data = msg.data.data['65281'];
                return {
                    power: precisionRound(data['152'], 2),
                    consumption: precisionRound(data['149'], 2),
                    temperature: precisionRoundOptions(data['3'], options, 'temperature'),
                };
            }
        },
    },
```

dmulcahey, 2019-04-30:
Please provide the simple descriptors for the device.

Nemesis24, 2019-05-03:
<SimpleDescriptor endpoint=1 profile=260 device_type=81 device_version=1 input_clusters=[0, 4, 3, 6, 16, 5, 10, 1, 2] output_clusters=[25, 10]>

Nemesis24, 2019-05-03:
https://github.com/dresden-elektronik/deconz-rest-plugin/commit/b3f44876f2d7259631f42d2559f15f8a906463fb

closed 2020-04-23 by Nemesis24

Hedda, 2021-10-06:
FYI, it looks like support for more Zigbee clusters attributes is now being worked on for [ZHA integration](https://www.home-assistant.io/integrations/zha) in Home Assistant core, see:

https://github.com/home-assistant/core/pull/56909

and

https://github.com/home-assistant/core/pull/56666

Also, note the last comment from Adminiuga in PR for ZHA in Home Assistant core -> https://github.com/home-assistant/core/pull/56909

"**_Support for other attributes could be extended later, as I couldn't find too many devices other than Sinopé RM3250ZB Electrical Load Controller – 50A – Zigbee https://www.amazon.com/dp/B07CGJ6XH2/ reporting anything interesting on the electrical measurement cluster._**"

So maybe now you could try to get this working with the latest Home Assistant beta or wait for the next upcoming release:

https://www.home-assistant.io/faq/release/

PS: There is much more discussion about ZHA energy cluster and attributes for smart power metering here -> https://github.com/zigpy/zha-device-handlers/issues/605 and https://github.com/zigpy/zha-device-handlers/issues/1045