zigpy/zha-device-handlers #62: Commercial Electric ETI Zigbee lights

jdub1810, 2019-04-30:
I have a number of Zigbee ETI Downlights. 

https://www.homedepot.com/p/Commercial-Electric-Wink-Compatible-5-in-and-6-in-65w-Equivalent-White-LED-Smart-Recessed-Trim-with-Color-Tunable-Feature-2700K-to-5000K-53166161/206029900

They worked perfectly for a bit, but recently I am having issues.  The lights (in a fresh install) still pair and still show the color temp slider and brightness slider, but when I adjust the light brightness in the UI and instead of adjusting it snaps back (in the UI) to it's existing (physical) setting.  I looked at the available quirks to see if the "quirks not getting applied" thread would apply, but I don't see those lights in there.  

Any possibility of getting those working again??

Thanks.

Adminiuga, 2019-04-30:
> but when I adjust the light brightness in the UI and instead of adjusting it snaps back (in the UI) to it's existing (physical) setting

So the actual physical brightness does not change?

call `loggers.set_level` service with the following service data:
```
{
  "homeassistant.core": "debug",
  "homeassistant.components.zha": "debug",
  "zigpy": "debug",
  "bellows.zigpy.application": "debug",
  "bellows.ezsp": "debug"
}
```

Change brightness for the light a few times and give about 10s between changes and post debug log here.

Adminiuga, 2019-04-30:
Assuming you are using EZSP based radio, like HUSBZB-1

dmulcahey, 2019-04-30:
This probably belongs as an issue with brightness handling in HA and not here... If you agree please create a new issue in the HA repo and close this.

jdub1810, 2019-05-01:
I will open in the HA repo. Thank you both for the attention!

closed 2019-05-01 by jdub1810