zigpy/zha-device-handlers #70: Xiaomi Aqara Wall Switch Wireless single button (WXKG03LM) recognized as Weather

snail59, 2019-05-03:
I opened this issue because I could not migrate the previous I opened in homeassistant bug tracker( [#23628](https://github.com/home-assistant/home-assistant/issues/23628) )

**Home Assistant release with the issue:**
The release version is 0.92.1

**Last working Home Assistant release (if known):**
Unknown

**Operating environment (Hass.io/Docker/Windows/etc.):**
raspberry pi 3/ FreeBSD / aarch64 / python 3.6 virtualenv / Conbee II

**Component/platform:**
zha

**Description of problem:**
Xiaomi Aqara wireless switch (model WXKG03LM)  is recongnized as a zhaquirks.xiaomi.mija.sensor_ht.Weather. 
The nodename is LUMI lumi.remote.b186acn01

**Problem-relevant `configuration.yaml` entries and (fill out even if it seems unimportant):**
```yaml
zha:
  usb_path: /dev/cuaZigbee
  database_path: /home/homeassistant/.homeassistant/zigbee.db
  radio_type: deconz
```

**Traceback (if applicable):**
_Adding a new device prints this :_ 
```
[0x0d8b:1] Discovering endpoint information
[0x0d8b:1] Discovered endpoint information: <SimpleDescriptor endpoint=1 profile=260 device_type=24321 device_version=1 input_clusters=[0, 3, 25, 65535, 18] output_clusters=[0, 4, 3, 5, 25, 65535, 18]>
[0x0d8b:1] Manufacturer: LUMI
[0x0d8b:1] Model: lumi.remote.b186acn01
[0x0d8b:2] Discovering endpoint information
[0x0d8b:2] Discovered endpoint information: <SimpleDescriptor endpoint=2 profile=260 device_type=24322 device_version=1 input_clusters=[3, 18] output_clusters=[4, 3, 5, 18]>
[0x0d8b:3] Discovering endpoint information
0xa707:3:0x0b04 async_update
0xe6a9:3:0x0b04 async_update
[0x0d8b:3] Discovered endpoint information: <SimpleDescriptor endpoint=3 profile=260 device_type=24323 device_version=1 input_clusters=[3, 12] output_clusters=[4, 3, 5, 12]>
Considering <class 'zigpy.quirks.smartthings.SmartthingsMultiPurposeSensor'>
Fail because endpoint list mismatch: dict_keys([1]) {1, 2, 3}
[...]
Considering <class 'zhaquirks.xiaomi.mija.sensor_ht.Weather'>
**Found custom device replacement for 00:15:8d:00:02:bf:ea:5e: <class 'zhaquirks.xiaomi.mija.sensor_ht.Weather'>**
LUMI lumi.remote.b186acn01: started configuration
node descriptor: [<Status.SUCCESS: 0>, 3467, <NodeDescriptor byte1=2 byte2=64 mac_capability_flags=128 manufacturer_code=4151 maximum_buffer_size=127 maximum_incoming_transfer_size=100 server_mask=0 maximum_outgoing_transfer_size=100 descriptor_capability_field=0>]
LUMI lumi.remote.b186acn01: channel: zdo-LUMI lumi.remote.b186acn01_ZDO async_configure stage succeeded
0x0d8b:2:0x0012: bound  'multistate_input' cluster: Status.SUCCESS
[0x0d8b:2:0x0012] 0 is not a valid attribute id
LUMI lumi.remote.b186acn01 channel: attribute-0x0d8b:2:0x0012 async_configure stage failed ex: object NoneType can't be used in 'await' expression
0x0d8b:1:0x0000: finished channel configuration
LUMI lumi.remote.b186acn01: channel: basic-0x0d8b:1:0x0000 async_configure stage succeeded
0x0d8b:3:0x000c: bound  'analog_input' cluster: Status.SUCCESS
[0x0d8b:3:0x000c] 0 is not a valid attribute id
LUMI lumi.remote.b186acn01 channel: attribute-0x0d8b:3:0x000c async_configure stage failed ex: object NoneType can't be used in 'await' expression
0x0d8b:1:0x0405: Failed to bind 'humidity' cluster:
0x0d8b:1:0x0001: bound  'power' cluster: Status.SUCCESS
0x0d8b:1:0x0012: Failed to bind 'multistate_input' cluster:
[0x0d8b:1:0x0012] 0 is not a valid attribute id
LUMI lumi.remote.b186acn01 channel: attribute-0x0d8b:1:0x0012 async_configure stage failed ex: object NoneType can't be used in 'await' expression
0x0d8b:1:0x0405: reporting 'measured_value' attr on 'humidity' cluster: 30/900/50: Result: '[6, <Status.UNSUP_GENERAL_COMMAND: 130>]'
0x0d8b:1:0x0001: reporting 'battery_voltage' attr on 'power' cluster: 30/900/1: Result: '[6, <Status.UNSUP_GENERAL_COMMAND: 130>]'
0x0d8b:1:0x0402: bound  'temperature' cluster: Status.SUCCESS
0x0d8b:1:0x0405: finished channel configuration
LUMI lumi.remote.b186acn01: channel: attribute-0x0d8b:1:0x0405 async_configure stage succeeded
0x0d8b:1:0x0402: reporting 'measured_value' attr on 'temperature' cluster: 30/900/50: Result: '[6, <Status.UNSUP_GENERAL_COMMAND: 130>]'
0x0d8b:1:0x0402: finished channel configuration
LUMI lumi.remote.b186acn01: channel: attribute-0x0d8b:1:0x0402 async_configure stage succeeded
0x0d8b:1:0x0001: reporting 'battery_percentage_remaining' attr on 'power' cluster: 30/900/1: Result: '[6, <Status.UNSUP_GENERAL_COMMAND: 130>]'
0x0d8b:1:0x0001: finished channel configuration
LUMI lumi.remote.b186acn01: channel: power-0x0d8b:1:0x0001 async_configure stage succeeded
LUMI lumi.remote.b186acn01: completed configuration
LUMI lumi.remote.b186acn01: stored in registry: ZhaDeviceEntry(name='LUMI lumi.remote.b186acn01', ieee='00:15:8d:00:02:bf:ea:5e', power_source=0, manufacturer_code=4151, last_seen=1556637448.5699582)
LUMI lumi.remote.b186acn01: started initialization
entry loaded from storage: ZhaDeviceEntry(name='LUMI lumi.remote.b186acn01', ieee='00:15:8d:00:02:bf:ea:5e', power_source=0, manufacturer_code=4151, last_seen=1556637448.5699582)
LUMI lumi.remote.b186acn01: channel: zdo-LUMI lumi.remote.b186acn01_ZDO async_initialize stage succeeded
LUMI lumi.remote.b186acn01: channel: power-0x0d8b:1:0x0001 async_initialize stage succeeded
LUMI lumi.remote.b186acn01: channel: attribute-0x0d8b:3:0x000c async_initialize stage succeeded
LUMI lumi.remote.b186acn01: channel: attribute-0x0d8b:1:0x0012 async_initialize stage succeeded
LUMI lumi.remote.b186acn01: channel: attribute-0x0d8b:1:0x0405 async_initialize stage succeeded
0xa707:3:0x0b04 async_update
0xe6a9:3:0x0b04 async_update
LUMI lumi.remote.b186acn01: channel: basic-0x0d8b:1:0x0000 async_initialize stage succeeded
LUMI lumi.remote.b186acn01: channel: attribute-0x0d8b:2:0x0012 async_initialize stage succeeded
LUMI lumi.remote.b186acn01: channel: attribute-0x0d8b:1:0x0402 async_initialize stage succeeded
LUMI lumi.remote.b186acn01: power source: Battery or Unknown
LUMI lumi.remote.b186acn01: completed initialization
[0x1a37:1:0x0402] ZCL request 0x000a: [[<Attribute attrid=0 value=<zigpy.zcl.foundation.TypeValue object at 0x4a52c2b0>>]]
[0x1a37:1:0x0402] Attribute report received: 0=2147
[0x1a37:1:0x0405] ZCL request 0x000a: [[<Attribute attrid=0 value=<zigpy.zcl.foundation.TypeValue object at 0x4a52c7f0>>]]
[0x1a37:1:0x0405] Attribute report received: 0=5218
[0x1a37:1:0x0403] ZCL request 0x000a: [[<Attribute attrid=0 value=<zigpy.zcl.foundation.TypeValue object at 0x4a527e10>>, <Attribute attrid=20 value=<zigpy.zcl.foundation.TypeValue object at 0x4a508e48>>, <Attribute attrid=16 value=<zigpy.zcl.foundation.TypeValue object at 0x4a52c2b0>>]]
[0x1a37:1:0x0403] Attribute report received: 0=1019, 20=-1, 16=10195
```

**Additional information:**
I copied remote_b286acn01.py to remote_b186acn01.py and remplaced every occurence of b286acn01 by b186acn01 but the result is :
- 3 binary_sensors (none working)
- 4 sensors (only one working)
- 1 temperature sensor ( which seems to work but the temperature is not accurate)

The working sensor sends those events:
-  left_single
- left_double
- left_long press ( there is no underscore between "long" and "press" )

snail59, 2019-05-03:
First comment added :

@dmulcahey : First, thank for your work.
I tried your [fix](https://github.com/dmulcahey/zha-device-handlers/pull/68) and the switch is recognized without any quirk.
But now it shows 3 sensors, all reporting a unknown status enven after 1, 2 short presses or a long press.
It does not report any binary_sensor. This is maybe normal, but people using a xiaomi gateway seem to have at least one.
The temperature sensor has disappeared, which seems normal.

Let me know if and how I can help.

snail59, 2019-05-03:
2nd comment added:

There are two models of WXKG03LM :
- [2016 model](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Xiaomi-WXKG03LM-2016)
- [2018 model](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Xiaomi-WXKG03LM-2018)

The same thing happens for the WXKG02LM
- the [2016 model](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Xiaomi-WXKG02LM-2016) is recognized as a lumi.sensor_86sw2
- the [2018 model](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Xiaomi-WXKG02LM-2018) is recognized as a lumi.remote.b286acn01 

The WXKG02LM 2018 has a dedicated quirk (zhaquirks/xiaomi/aqara/remote_b286acn01.py), so I suppose that XKG03LM 2018 needs a dedicated quirk too.

I tried to create one from remote_b286acn01.py but my knowledge is limited on this topic.
I get these problems :

1) there is no binary_sensor created, while according to the  homeassistant's xiaomi binary sensor [webpage](https://www.home-assistant.io/components/binary_sensor.xiaomi_aqara), there are some.
2) There are 3 endpoints in the signature. For b286acn01, the first is for the left button, the 2nd is for the right one and the third is for both. But for b186acn01, 2 and 3 are obviously useless. That is maybe a  b286acn01 "amputated".

Nemesis24, 2019-05-03:
WXKG11LM too

dmulcahey, 2019-05-05:
Fixed by https://github.com/dmulcahey/zha-device-handlers/pull/68

dmulcahey, 2019-05-05:
The weather sensor defines a manufacturer and model now so it will not incorrectly match anything else. Note: all button type devices emit events. The binary sensors are a legacy artifact and are subject to removal in the future. To see the events, subscribe to zha_event in the event tool in HA.

closed 2019-05-05 by dmulcahey