zigpy/zha-device-handlers #71: AduroSmart ERIA Smart Wireless Dimming Switch Remote

tube0013, 2019-05-03:
https://www.amazon.com/gp/product/B07HJHJWGT/ref=ppx_yo_dt_b_asin_title_o05_s00?ie=UTF8&psc=1

Is this enough info to create a quirk for this remote? If not let me know and I can look for more logs.  I'm @ tube on discord.

```
[zigpy.endpoint] [0x1d0f:1] Discovered endpoint information: <SimpleDescriptor endpoint=1 profile=260 device_type=2080 device_version=2 input_clusters=[0, 3, 8, 4096, 64716] output_clusters=[3, 4, 6, 8, 4096, 64716]>
```
On:

```
2019-05-03 09:33:43 DEBUG (MainThread) [zigpy.zcl] [0x1d0f:1:0x0006] ZCL request 0x0101: []
2019-05-03 09:33:43 DEBUG (MainThread) [zigpy.zcl] [0x1d0f:1:0x0006] No handler for cluster command 1
2019-05-03 09:33:43 WARNING (MainThread) [zigpy.zcl] Unknown cluster-specific command 0
2019-05-03 09:33:43 DEBUG (MainThread) [zigpy.zcl] [0x1d0f:1:0xfccc] ZCL request 0x0100: b'\x00\x00\x00'
2019-05-03 09:33:43 DEBUG (MainThread) [zigpy.zcl] [0x1d0f:1:0xfccc] No handler for cluster command 0
```

Up:
```
2019-05-03 09:32:44 DEBUG (MainThread) [zigpy.zcl] [0x1d0f:1:0x0008] ZCL request 0x0102: [0, 16, 9]
2019-05-03 09:32:44 DEBUG (MainThread) [zigpy.zcl] [0x1d0f:1:0x0008] No handler for cluster command 2
2019-05-03 09:32:44 WARNING (MainThread) [zigpy.zcl] Unknown cluster-specific command 0
2019-05-03 09:32:44 DEBUG (MainThread) [zigpy.zcl] [0x1d0f:1:0xfccc] ZCL request 0x0100: b'\x00\x01\x00'
2019-05-03 09:32:44 DEBUG (MainThread) [zigpy.zcl] [0x1d0f:1:0xfccc] No handler for cluster command 0
```
Down:
```
2019-05-03 09:34:52 DEBUG (MainThread) [zigpy.zcl] [0x1d0f:1:0x0008] ZCL request 0x0102: [1, 16, 9]
2019-05-03 09:34:52 DEBUG (MainThread) [zigpy.zcl] [0x1d0f:1:0x0008] No handler for cluster command 2
2019-05-03 09:34:52 WARNING (MainThread) [zigpy.zcl] Unknown cluster-specific command 0
2019-05-03 09:34:52 DEBUG (MainThread) [zigpy.zcl] [0x1d0f:1:0xfccc] ZCL request 0x0100: b'\x00\x02\x00'
2019-05-03 09:34:52 DEBUG (MainThread) [zigpy.zcl] [0x1d0f:1:0xfccc] No handler for cluster command 0
```

Off:
```
2019-05-03 09:35:15 DEBUG (MainThread) [zigpy.zcl] [0x1d0f:1:0x0006] ZCL request 0x0100: []
2019-05-03 09:35:15 DEBUG (MainThread) [zigpy.zcl] [0x1d0f:1:0x0006] No handler for cluster command 0
2019-05-03 09:35:15 WARNING (MainThread) [zigpy.zcl] Unknown cluster-specific command 0
2019-05-03 09:35:15 DEBUG (MainThread) [zigpy.zcl] [0x1d0f:1:0xfccc] ZCL request 0x0100: b'\x00\x03\x00'
2019-05-03 09:35:15 DEBUG (MainThread) [zigpy.zcl] [0x1d0f:1:0xfccc] No handler for cluster command 0
```

tube0013, 2019-05-16:
I just pulled this out again and noticed that I get a binary sensor that shows open/closed based on the on or off button press:

[image]

Also I am getting zha_events for the open and close buttons, just not the up and down:

```
Event 1 fired 10:53 PM:
{
    "event_type": "zha_event",
    "data": {
        "unique_id": "0x1d0f:1:0x0006",
        "device_ieee": "00:15:8d:00:01:98:d2:cf",
        "command": "off",
        "args": []
    },
    "origin": "LOCAL",
    "time_fired": "2019-05-16T02:53:29.728295+00:00",
    "context": {
        "id": "d42550a80b4948efb03a6a26925c2153",
        "parent_id": null,
        "user_id": null
    }
}
Event 0 fired 10:53 PM:
{
    "event_type": "zha_event",
    "data": {
        "unique_id": "0x1d0f:1:0x0006",
        "device_ieee": "00:15:8d:00:01:98:d2:cf",
        "command": "on",
        "args": []
    },
    "origin": "LOCAL",
    "time_fired": "2019-05-16T02:53:28.748698+00:00",
    "context": {
        "id": "f297bc16fa3b425296c3d90cff2aa487",
        "parent_id": null,
        "user_id": null
    }
}
```

dmulcahey, 2019-06-07:
this looks like it should just work. cluster 8 is level control and there should be events for that by default if the device actually uses the cluster. I would assume up/down are for level? Your posted traffic even shows traffic on cluster 8. can you try removing and re-pairing the device?

dmulcahey, 2019-06-07:
@tube0013 I picked one up... should be in tomorrow and i'll mess w/ it.

tube0013, 2019-06-07:
Cool - gotta love 1 day shipping.

I was home today so just tried to re-pair the remote, and get the same result.  the On and Off buttons work as expected, but the up/down just show in the zigpy debug log, not as events.

dmulcahey, 2019-06-09:
@tube0013 https://github.com/dmulcahey/zha-device-handlers/commit/94b5fef6405573c22712b83b4841846a83e3c2bb

buttons work now... gotta look into this more. This shouldn't be necessary from what I see.

tube0013, 2019-06-09:
Yea weird I’ll give this a try when I get a few. Thank you.

dmulcahey, 2019-06-25:
will be in HA .95

closed 2019-06-25 by dmulcahey