zigpy/zha-device-handlers #76: Xiaomi Aqara Wall Switch Wireless single button (WXKG03LM) not working

snail59, 2019-05-06:
Hello,

As my previous issue was closed [issue 70(https://github.com/dmulcahey/zha-device-handlers/issues/70), I did not know what to do, except opening this new bug. If this is not the right way, please tell me.

My comments in issue 70 are still valid. To summarize:
- I manually applied your patch and there are now 3 sensors, but their status is "unknown" .
- I created the file``remote_b186acn01.py`` from ``remote_b286acn01.py`` and while there are too many sensors, one is properly reporting events. Those events are misnamed (left_single, double_single etc) but that is not the most important.

I do not see "zha_event in the event tool in HA.". But I will search.

Again, let me know if you need more informations or if I can help in any way.

dmulcahey, 2019-05-06:
This functions as designed with the new release on PyPi for the quirks lib. I’m not sure what you’re expecting but the events are what should be used for this device. I’m not sure what you think the actual issue is? As far as I am concerned everything is working as it should...

snail59, 2019-05-06:
I thought I was clear but it seems I am not -), so I will rephrase, OK ?
My expectations are/were the following :

1. recognition
**expectation** 
having this module to be properly recognized because it was recognized as a weather module
**result**
It is now recognized thanks to your patch, and shows 3 sensors.

2. sensors
**expectation** 
Those 3 sensors to print values.
**result**
Unfortunately, they don't and report nothing but "unknown", so I believe something is wrong.

3. sensors again
**expectation** 
Getting those 3 sensors to print values.
**tries**
The absence of value from the sensors seemed wrong, so I looked at the code (I do not know python that much) and saw a file named "remote_b286acn01.py". 
As the signature looks exactly the same except for the model, I duplicated it to a file named "remote_b186acn01.py" (not existing before) and replaced each occurence of "b**2**86acn01" by "b**1**86acn01".
**result**
With zhaquirks 0.0.12, I get 4 sensors:
- 2 are still reporting a state of "unknown". I can live with this, but I would happy to know why they are here. Result of a bad configuration ? Should they even exist ?
- 1 is reporting temperature ( which seems strange to me, bad configuration again?)
- 1 is the winner. It shows status accordingly to the different presses ( single, double and long press). 
That is why I believe that this module needs a dedicated quirk, as b286acn01 does. 
The only problem is that It shows them as left_xxx because the origin file was written for the two buttons model, while this one has only 1 button, and so does not need to have a left_xxx right_xxx of both_xxx status. Only single, double and long are relevant status.

4. binary_sensors
**expectation** 
With zhaquirks 0.0.8, there are 4 binary_sensors too.
I was hoping I could use one of them to get a on/off or clicked/not_clicked status becauseI do not need double and long presses.
**result**
You wrote that binary_sensors are expected to be removed and that I should use events.
I accept that. I just have to figure out how to find those events.

If I am wrong somewhere, please tell me why. I like/need to understand things :-) .

darootler, 2019-05-06:
Hi!

I have exactly the same situation. I am just received a WXKG03LM button. I applied the latest patch, paired the button and getting 3 sensons. All sensors are reporting unknown.

Am I doing something wrong or should one of those sensors report anything back on a click? At the moment i have paired the button but cannot do anything with it.

Regards
Richard

dmulcahey, 2019-05-07:
I ordered a WXKG03LM to explore this. It will be a few weeks. Stay tuned.

dmulcahey, 2019-05-13:
Just updating... still waiting for this to arrive.

dmulcahey, 2019-05-23:
@snail59 take a look at: https://github.com/dmulcahey/zha-device-handlers/pull/83

snail59, 2019-05-23:
@dmulcahey tested and approved ;-).
However, I have 2 questions. 
There are 2 unknown sensors; could you please explain why they appear? 
Plus, there is a temperature sensor; why ?
I am asking this only by curiosity.

closed 2019-05-24 by dmulcahey

darootler, 2019-06-06:
When will this fix be included in hass.io?

Regards
Richard

PR #83 (2019-05-24): Add support for Aqara lumi b186acn01
fixes: #76