zigpy/zha-device-handlers #79: <not an issue> question about install/usage

pbathuk, 2019-05-13:
Hi,

I am currently using a zigate device, which is ok, however I have a number of devices that I have brought that are not supported & it would appear that zigate -> home assistant support is limited, as there are only two contributors & I can't find out how to add new templates.. 
So I am going to be buying a conbee ii dongle & migrating to zha, now I have a number of devices that I believe quirks for (xiaomi, ikea etc), to be able to use the quirks do I :
setup zha in HA with my dongle
git clone the rep to a custom component 
restart & thats it? 

Or am I missing something simple like zha already has them baked in?
If zha does have them baked in, how would I create my own local quirk before submitting it for others? i.e. is there a set folder it goes into be picked up?

Sorry for the questions, I did read the contribute & tried the wiki, however the wiki links back to the front page.

Cheers

dmulcahey, 2019-05-13:
It’s enabled by default. No need for a custom component either.

dmulcahey, 2019-05-13:
As for adding local files... depends on HA installation method. When you’re ready ask on #zigbee on discord and someone will assist.

pbathuk, 2019-05-13:
Brilliant thanks. 
Will wait on my usb stick and once I hit one that does not work head to discord :) thanks so much

closed 2019-05-13 by pbathuk

dmulcahey, 2019-05-13:
Any time... handle is the same on discord btw