zigpy/zha-device-handlers #81: Sengled Element A19NAE26 color temperature zha handler thingy

backcountrymountains, 2019-05-14:
I've got several Sengled Element tunable white bulbs that I'd like to get working in HA with color temperature.

I haven't removed/rejoined the device so I don't have a "SimpleDescriptor" but I need it to match exactly for the quirk to work. I will get it.

That one time that I got a bellows description of the device:
> Device:
>   NWK: 0xb94e
>   IEEE: b0:ce:18:14:03:01:43:a9
>   Endpoints:
>     1: profile=0x104, device_type=DeviceType.DIMMABLE_LIGHT
>       Input Clusters:
>         Basic (0)
>         Identify (3)
>         Groups (4)
>         Scenes (5)
>         On/Off (6)
>         Level control (8)
>         Color Control (768)
>         Metering (1794)
>         Diagnostic (2821)
>       Output Clusters:
>         Ota (25)

Current zigbee.db entry:

> ieeeeeeeeeeeeee endpiont clust attrid value
> b0:ce:18:14:03:01:43:a9	1	0	4	sengled
> b0:ce:18:14:03:01:43:a9	1	0	5	Z01-A19NAE26
(how the crap do you format a table in here)

It seems like I should be able to modify the Osram A19 tunable white to work for the Sengled Element bulbs. I have no idea how to do that.

Everything is on endpoint 1. 

bellows commands on the color cluster work:
> bellows -v debug -d /dev/ttyUSB1 zcl -D /config/zigbee.db b0:ce:18:14:03:01:43:a9 1 768 command move_to_color_temp 300 5

The metering cluster puts out some kind of virtual watts, like the apparent wattage were the bulb an incandescent.

I have no idea what happens on the diagnostic cluster.

So I pretty much copy/pasted the OSRAM and made a sengled_element.py:

>  """Sengled Element tunable white device."""
> from zigpy.profiles import zha
> from zigpy.quirks import CustomDevice
> from zigpy.zcl.clusters.general import (
>     Basic, Groups, Identify, LevelControl, OnOff, Ota, Scenes)
> from zigpy.zcl.clusters.homeautomation import ElectricalMeasurement
> from zigpy.zcl.clusters.lighting import Color
> 
> class A19TunableWhite(CustomDevice):
>     """Sengled A19 tunable white device."""
> 
>     signature = {
>         # <SimpleDescriptor endpoint=1 profile=260 device_type=257
>         # device_version=1 input_clusters=[0, 3, 4, 5, 6, 8, 768, 1794, 2821]
>         # output_clusters=[25]>
>         1: {
>             'manufacturer': 'sengled',
>             'model': 'Z01-A19NAE26',
>             'profile_id': zha.PROFILE_ID,
>             'device_type': zha.DeviceType.DIMMABLE_LIGHT,
>             'input_clusters': [
>                 Basic.cluster_id,
>                 Identify.cluster_id,
>                 Groups.cluster_id,
>                 Scenes.cluster_id,
>                 OnOff.cluster_id,
>                 LevelControl.cluster_id,
>                 Color.cluster_id,
>                 ElectricalMeasurement.cluster_id,
>                 Diagnostic.cluster_id
>             ],
>             'output_clusters': [
>                 Ota.cluster_id
>             ],
>         }
>     }
> 
>     replacement = {
>         'endpoints': {
>             1: {
>                 'profile_id': zha.PROFILE_ID,
>                 'device_type': zha.DeviceType.COLOR_DIMMABLE_LIGHT,
>                 'input_clusters': [
>                     Basic.cluster_id,
>                     Identify.cluster_id,
>                     Groups.cluster_id,
>                     Scenes.cluster_id,
>                     OnOff.cluster_id,
>                     LevelControl.cluster_id,
>                     Color.cluster_id,
>                     ElectricalMeasurement.cluster_id,
>                     Diagnostic.cluster_id
>                 ],
>                 'output_clusters': [
>                     Ota.cluster_id
>                 ],
>             },
>         }
>     }

Do I have to coerce the device_type from DIMMABLE_LIGHT to COLOR_DIMMABLE_LIGHT so I have access to cluster 768?

Where do I put the .py file? /custom_components/zha_device_handlers/sengled/sengled_element.py?

Will it work?

P.S. I don't know where to put this other issue this so I'm hoping somebody can tell me:

If I try to issue the color cluster command in the HA service tab or using the ZHA integration page, I get the following error:

> zha/core/device.py", line 366, in issue_cluster_command
>     expect_reply=True)
> ValueError: Wrong number of parameters for request, expected 2 argument(s)

The 768 color cluster is available on the integration page, but it takes two arguments: color temp and transition time. I can't input two arguments on the ZHA integrations page and the service call throws the same "expected two arguments failure"

> {"ieee":"b0:ce:18:14:03:01:43:a9",
> "endpoint_id":"1",
> "cluster_id":"768",
> "cluster_type":"in",
> "command":"10",
> "command_type":"client",
> "args":"200 3",
> "manufacturer":""
> }

pps: Is it a client or server command?

Thanks for the help.

dmulcahey, 2019-05-14:
Wow, a lot here... try the beta... this should just work now!

backcountrymountains, 2019-05-14:
Beta HA? i'm running hassio in virtualbox on a pc. 
Beta fixes sengled element or color cluster command?

dmulcahey, 2019-05-14:
Yes

backcountrymountains, 2019-05-14:
okay, thanks.

dmulcahey, 2019-05-14:
You may have to remove and re join the device after updating

backcountrymountains, 2019-05-14:
yeah. I'd probably have to do that in any case

backcountrymountains, 2019-05-14:
backing everything up now to switch to beta.

backcountrymountains, 2019-05-14:
93b3 Works! The element bulbs showed up with color temperature control and I didn't even have to re-join! Yay!

closed 2019-05-14 by backcountrymountains