zigpy/zha-device-handlers #86: Hue Dimmer Long Press Events

andrewpc, 2019-05-28:
Ive successfully joined a hue dimmer (RWL021) to ZHA. The basic events are present, these being single tap on/off, single tap dim up/down and hold dim up/down step. These can be mapped through to HA for automations.

Reading into the Zigbee2MQTT support for this dimmer, it appears that this device is also capable of providing discrete long press and multi-press events. To enable this, from reading the code and the related issue, it appears that when initialized, the dimmer needs an attribute written to it. 

The relevant code reference is here: https://github.com/Koenkk/zigbee-shepherd-converters/blob/69befbcce5d744cf5f81e1cdf6f59cd5bcb5917e/devices.js#L909

The Issue link is here: https://github.com/Koenkk/zigbee2mqtt/issues/585

Is this assumption correct, and how difficult is this to add to ZHA? Im happy to try and have a go at this, but want to get some validation on whats needed and any ZHA limitations that might prevent this?

dmulcahey, 2019-05-28:
> The Issue link is here: Koenkk/zigbee2mqtt#585

This should be fairly trivial. You may be able to write the attribute via the config panel already. That said, I do see a bind for a mfg specific cluster there. Not sure what that’s about. I don’t have one of these... do you have a link to the exact model? I can pick one up to mess with this a bit.

David F. Mulcahey

jonathanweinberg, 2019-05-28:
Not the OP, but this should be the model: https://www.amazon.com/Philips-Dimmer-Switch-Installation-Free-Exclusively/dp/B076MGKTGS

As far as I know, there's only one real model of the switch, different SKU's for Europe / US + Style ( I/O vs On/Off printed on the remote).

Found this: https://www.reddit.com/r/Hue/comments/7ia4f5/is_there_a_second_version_of_the_hue_dimmer_switch/

andrewpc, 2019-05-29:
Yes, thanks @jonathanweinberg Confirming thats the one I am using.

I did try sending the configuration using the config panel, but either didnt get the values write or got some warnings about the values being invalid? I wasnt sure if there was some kind of schema limiting what needed to be sent or I had converted them incorrectly.

andrewpc, 2019-06-04:
Just an update, I did re-try setting the attribute using the service, but I believe it failed because the attribute is manufacturer specific, and therefore didnt match the schema. If I can get around that, by setting some other flag, I am happy to try again? It might be that it should be done in code?

dmulcahey, 2019-06-07:
@andrewpc I picked one up. Should be here tomorrow.

andrewpc, 2019-06-08:
Great! Looking forward to hearing how your progress. Please ket me know if I can help test any changes.

andrewpc, 2019-06-19:
@dmulcahey Ive noticed you've pushed some changes for this remote. Does this enable the multi-press features? I tested quickly by swapping for the new quirks file and saw the manufacturer specific cluster/attributes appear, but wasnt sure whether something additional was needed to enable the feature? Thanks

dmulcahey, 2019-06-25:
I am hacking on this w/ a HA branch too. when time permits: https://github.com/dmulcahey/home-assistant/tree/dm/zha-phillips-remote. It uses a mfg specific cluster after the attribute is updated for the "advanced" functions. This will require some heavy customization to get right as well as ignoring the events on the regular ZCL clusters. Feel free to take that branch and mess around.

bpottle1, 2020-01-25:
Hey guys, just wondering if any progress had been made on this request? I would love to be able to have the long press and/or double click on the hue dimmer switches. I’ve been searching everywhere to try and figure it out and then I found this issue open on github from a few months back.

jcallaghan, 2020-06-22:
@dmulcahey @andrewpc were you able to get long press and double click events with this device in the end? I've just moved 130~ devices over from deCONZ and am going to really miss this feature.

dmulcahey, 2020-06-22:
I haven’t touched this in forever. I’ll see if I can mess with it some more soon.

fintmu, 2020-06-27:
Following this. I have been testing zha after deconz few days now and this missing feature is big one for me. I really need that long press with hue dimmer switch somehow if I decide to stay with zha.

Related to this light commands are much slower with zha to complete than with deconz. I get lot of "Script already running" warnings if the buttons are pressed repeatedly.

PR #388 (2020-07-03): [BREAKING CHANGE] Phillips Hue remote manufacturer specific cluster handling
This PR adds support for 4 actions per button: press, hold, short release and long release. It needs https://github.com/home-assistant/core/pull/37172 in order to function. 

Fixes: #86 

Things still needed:

- [x] device trigger support
- [x] deduplicate code between remote versions
closed 2020-07-03 by dmulcahey