zigpy/zha-device-handlers #87: Support for Centralite 3405-L

jonathanweinberg, 2019-05-28:
Hi there! Was wondering about support for the Centralite 3405-L Security Keypad.

This is known as the "Lowes Iris Keypad V2" (there is a V1, which is the proprietary Lowes Iris / AlertMe zigbee, and the v3, also from Centralite, which has different features).

[Lowes Iris Keypad V2 Manual (pdf warning) ](http://pdf.lowes.com/useandcareguides/812489023049_use.pdf)

[SmartThings Groovy Support Handler for the Keypad](https://github.com/miriad/Centralite-Keypad/blob/master/devicetypes/mitchpond/centralite-keypad.src/centralite-keypad.groovy)

Log output from adding the device via hassio / Home-Assistant 93.2 via a Nortek HUSBZB-1 (removed the most of the list of failures to match quirks).

```
[0x0000:zdo] ZDO request 0x0036: [60, <Bool.false: 0>]
Device 0xa734 (00:0d:6f:00:0b:63:2e:9f) joined the network
[0xa734] Requesting 'Node Descriptor'
[0xa734:zdo] ZDO request 0x0013: [42804, 00:0d:6f:00:0b:63:2e:9f, 128]
[0xa734] Node Descriptor: <NodeDescriptor byte1=2 byte2=64 mac_capability_flags=128 manufacturer_code=4174 maximum_buffer_size=82 maximum_incoming_transfer_size=82 server_mask=0 maximum_outgoing_transfer_size=82 descriptor_capability_field=0>
[0xa734] Discovering endpoints
[0xa734] Discovered endpoints: [1]
[0xa734:1] Discovering endpoint information
[0xa734:1] Discovered endpoint information: <SimpleDescriptor endpoint=1 profile=260 device_type=1025 device_version=0 input_clusters=[0, 1, 3, 32, 1026, 1280, 1281, 2821, 64516, 64517] output_clusters=[25, 1281]>
[0xa734:1] Manufacturer: CentraLite
[0xa734:1] Model: 3405-L
Considering <class 'zigpy.quirks.smartthings.SmartthingsMultiPurposeSensor'>
Fail because device_type mismatch on at least one endpoint
...
Considering <class 'zhaquirks.xiaomi.mija.smoke.MijiaHoneywellSmokeDetectorSensor'>
Fail because device_type mismatch on at least one endpoint
CentraLite 3405-L: started configuration
node descriptor: [<Status.SUCCESS: 0>, 42804, <NodeDescriptor byte1=2 byte2=64 mac_capability_flags=128 manufacturer_code=4174 maximum_buffer_size=82 maximum_incoming_transfer_size=82 server_mask=0 maximum_outgoing_transfer_size=82 descriptor_capability_field=0>]
CentraLite 3405-L: channel: zdo-CentraLite 3405-L_ZDO async_configure stage succeeded
0xa734:1:0x0000: finished channel configuration
initializing channel: basic from_cache: False
CentraLite 3405-L: channel: basic-0xa734:1:0x0000 async_configure stage succeeded
[0xa734:1:0x0000] Unexpected ZCL reply 0x0001: [[<ReadAttributeRecord attrid=7 status=0 value=3>]]
0xa734:1:0x0500: started IASZoneChannel configuration
0xa734:1:0x0001: bound  'power' cluster: Status.SUCCESS
0xa734:1:0x0402: bound  'temperature' cluster: Status.SUCCESS
0xa734:1:0x0500: bound  'ias_zone' cluster: Status.SUCCESS
0xa734:1:0x0001: reporting 'battery_voltage' attr on 'power' cluster: 30/900/1: Result: '[[<ConfigureReportingResponseRecord status=0 direction=0 attrid=0>]]'
0xa734:1:0x0402: reporting 'measured_value' attr on 'temperature' cluster: 30/900/50: Result: '[[<ConfigureReportingResponseRecord status=0 direction=0 attrid=0>]]'
0xa734:1:0x0500: wrote cie_addr: 00:0d:6f:00:0a:ff:73:58 to 'ias_zone' cluster: [<WriteAttributesStatusRecord status=Status.SUCCESS attrid=0>]
0xa734:1:0x0500: finished IASZoneChannel configuration
CentraLite 3405-L: channel: ias_zone-0xa734:1:0x0500 async_configure stage succeeded
0xa734:1:0x0402: finished channel configuration
CentraLite 3405-L: channel: attribute-0xa734:1:0x0402 async_configure stage succeeded
0xa734:1:0x0001: reporting 'battery_percentage_remaining' attr on 'power' cluster: 30/900/1: Result: '[[<ConfigureReportingResponseRecord status=134 direction=0 attrid=33>]]'
0xa734:1:0x0001: finished channel configuration
CentraLite 3405-L: channel: power-0xa734:1:0x0001 async_configure stage succeeded
CentraLite 3405-L: completed configuration
CentraLite 3405-L: stored in registry: ZhaDeviceEntry(name='CentraLite 3405-L', ieee='00:0d:6f:00:0b:63:2e:9f', power_source=0, manufacturer_code=4174, last_seen=1559071414.6262376)
CentraLite 3405-L: started initialization
entry loaded from storage: ZhaDeviceEntry(name='CentraLite 3405-L', ieee='00:0d:6f:00:0b:63:2e:9f', power_source=0, manufacturer_code=4174, last_seen=1559071414.6262376)
CentraLite 3405-L: channel: zdo-CentraLite 3405-L_ZDO async_initialize stage succeeded
initializing channel: basic from_cache: False
CentraLite 3405-L: channel: basic-0xa734:1:0x0000 async_initialize stage succeeded
initializing channel: attribute from_cache: False
CentraLite 3405-L: channel: attribute-0xa734:1:0x0402 async_initialize stage succeeded
initializing channel: ias_zone from_cache: False
CentraLite 3405-L: channel: ias_zone-0xa734:1:0x0500 async_initialize stage succeeded
initializing channel: power from_cache: False
CentraLite 3405-L: channel: power-0xa734:1:0x0001 async_initialize stage succeeded
CentraLite 3405-L: power source: Battery or Unknown
CentraLite 3405-L: completed initialization
0x62b0:1:0x0b04 async_update
0x62b0:1:0x0b04 async_update
[0xa734:1:0x0501] ZCL request 0x0107: []
[0xa734:1:0x0501] No handler for cluster command 7
Data remains after deserializing ZCL frame
[0xa734:1:0x0501] ZCL request 0x0100: []
[0xa734:1:0x0501] No handler for cluster command 0
[0xa734:1:0x0501] ZCL request 0x0104: []
[0xa734:1:0x0501] No handler for cluster command 4
Device 0xa734 (00:0d:6f:00:0b:63:2e:9f) joined the network
Skip initialization for existing device 00:0d:6f:00:0b:63:2e:9f
[0xa734:zdo] ZDO request 0x0013: [42804, 00:0d:6f:00:0b:63:2e:9f, 128]
[0xa734:1:0x0501] ZCL request 0x0107: []
[0xa734:1:0x0501] No handler for cluster command 7
[0xa734:1:0x0501] ZCL request 0x0107: []
[0xa734:1:0x0501] No handler for cluster command 7
[0xa734:1:0x0501] ZCL request 0x0107: []
[0xa734:1:0x0501] No handler for cluster command 7
[0xa734:1:0x0501] ZCL request 0x0104: []
[0xa734:1:0x0501] No handler for cluster command 4
[0x448c:1:0x0001] ZCL request 0x000a: [[<Attribute attrid=32 value=<TypeValue type=uint8_t, value=28>>]]
[0x448c:1:0x0001] Attribute report received: battery_voltage=28
```

Hitting buttons will pop up the `No handler...` messages. Eventually I'll get `2019-05-28 15:24:21 WARNING (MainThread) [zigpy.zcl] Data remains after deserializing ZCL frame` in the Home Assistant log and stop receiving reports.

Did I miss any required info? What can I do to help?

dmulcahey, 2019-06-07:
This will take a bit of work... and won't be possible w/o the device (well... it's possible but not worth trying) because of the complexity. Do you know where I can pick one up?

jonathanweinberg, 2019-06-07:
That might be tough. Are you in the US?

dmulcahey, 2019-06-07:
Yes

presslab-us, 2019-06-13:
I have one of these, and was planning to implement it eventually...  but I'm pretty busy lately so it might be a while.

nphil, 2019-11-27:
@dmulcahey are you still interested in getting these supported? I have a extra that I can ship you. I also have an Iris Keyfob that's missing support in ZHA.

dmulcahey, 2020-03-02:
I started the implementation for this. Should be in HA 107 or HA 108

nerdosity, 2020-09-16:
Do you need help in testing @dmulcahey ? I just received a Technicolor TKA105 (Xfinity XHK1-TC).

dmulcahey, 2020-09-16:
I need to find time to work on this :) it’s on the list though. I’ll let you know when I have something

nerdosity, 2020-09-16:
https://github.com/Koenkk/zigbee-herdsman-converters/commit/223e543a362e60897d41f3a5af4430077d7b16f4 this is latest commit for zigbee2mqtt, it is working on integration a similar (almost identical) unit.
I am proceeding to integrate this with HA ZHA events, basic integration seems easy. Let me know if you need some help investigating

nerdosity, 2020-11-27:
Hi, FIY, someone just integrated a XHK1-TC in zigbee2mqtt.
I hope his device.js helps, he just said that everything is up to spec.

https://drive.google.com/file/d/1EApTMI9YTDy3-_loF7yzOjrpxdz-5F_q/view

h4rlw00d, 2021-02-13:
Hello everyone, is there any news on this issue?

Hedda, 2021-04-26:
FYI, looks like dmulcahey is working on alarm control panel support to ZHA for Home Assistant core -> https://github.com/home-assistant/core/pull/49080

nerdosity, 2021-04-27:
Hi, how can I test it? I would gladly help.

Hedda, 2021-04-27:
> how can I test it? I would gladly help.

dmulcahey development branch for it is found here -> https://github.com/dmulcahey/home-assistant/tree/dm/zha-ias-ace-support

Comments if testing before merged should go into the pull request -> https://github.com/home-assistant/core/pull/49080

Hedda, 2021-04-30:
FYI; Home Assistant 2021.5 (planned for release May 5, 2021) Beta release notes: "**_ZHA now has support for alarm control panels_**".

https://rc.home-assistant.io/blog/2021/04/28/release-20215/ -> https://github.com/home-assistant/core/pull/49080

closed 2021-11-03 by github-actions[bot]

u144175, 2022-04-06:
Has anyone gotten the CentraLite 3405-L working in HA?  I have been able to pair the device and get the temperature sensor reading, as well as setting an away-armed state by pressing the ON button on the physical keypad.  However, I cannot figure out how to send a disarm command via the keypad.  Probably because I have not been able to set a PIN.  No idea how to do that or where to do that.  Power (battery) level indicate is not functioning but returns 'unknown'.  I am interested in using this keypad to enter a four-digit PIN, which if entered correctly, an entity change could be used to drive certain automations.  Anyone gotten something like this working with the 3405-L?  How did you set the PIN?

h4rlw00d, 2022-04-06:
I did set up the pin code in HA by going to the configuration menu, then devices and services. 
In the list of integrations find Zigbee Home Automation (ZHA) and then click on configure. By scrolling at the bottom of the page you should find the 'Alarm Control Panel Options' where is possible to set up the master code. 
There are also other options as the number of trials before triggering the alarm and a checkbox to select if the code is needed to arm the alarm. 
Click on update the configuration and it should work fine.

u144175, 2022-04-06:
@h4rlw00d - Thank you!  That worked.  I never would have found it there without your direction.  I expected to find the PIN under the keypad's configuration.
Back years ago when I used this Iris branded keypad with my Lowe's Iris hub, it chirped any time a door was opened or closed.  It would be simple to create an automation to look for any door open/closed event but I have no idea how to send a command to the keypad to chirp nor what that command would be.  Any ideas on this?  Thanks again!

h4rlw00d, 2022-04-09:
@u144175 Sorry, but I don't know if is possible to send a chirp command to the keypad (I don't believe that it has been implemented yet). A similar solution would be to use a smart speaker and to send an audio file when the open/close event occurs.