zigpy/zha-device-handlers #90: CentraLite 3320-L refusing to complete pairing

rpitera, 2019-06-05:
HASS.io version: 0.93.2
Zigbee Hub Device: Conbee II, latest firmware version
Quirks mode manually specified in configuration.yaml (Just in case)

Hope someone can help; I've been struggling with this for weeks.  I've tried at least 4 different sensors.  The first was from an unsealed box so I tried a factory reset but no luck.  Thinking it might be defective I then tried 3 from factory sealed boxes but got the same error sequence 
 each time (only the addresses are different, obviously):

```
Device 0x12c4 (00:0d:6f:00:0b:be:55:cb) joined the network
[0x12c4:zdo] ZDO request 0x0013: [48725, 00:00:80:00:0d:6f:00:0b, 0]
[0x12c4] Requesting 'Node Descriptor'
[0x12c4] Requesting 'Node Descriptor'
[0x12c4] Node Descriptor: <NodeDescriptor byte1=2 byte2=64 mac_capability_flags=128 manufacturer_code=49887 maximum_buffer_size=82 maximum_incoming_transfer_size=82 server_mask=0 maximum_outgoing_transfer_size=82 descriptor_capability_field=0>
Error calling listener.node_descriptor_updated: FOREIGN KEY constraint failed
[0x12c4:zdo] ZDO request 0x0006: [65533, 260, [], [1280]]
[0x12c4] Node Descriptor: <NodeDescriptor byte1=2 byte2=64 mac_capability_flags=128 manufacturer_code=49887 maximum_buffer_size=82 maximum_incoming_transfer_size=82 server_mask=0 maximum_outgoing_transfer_size=82 descriptor_capability_field=0>
[0x12c4] Discovering endpoints
[0x12c4] Discovered endpoints: [1, 2]
[0x12c4:1] Discovering endpoint information
[0x12c4:1] Discovered endpoint information: <SimpleDescriptor endpoint=1 profile=260 device_type=1026 device_version=0 input_clusters=[0, 1, 3, 32, 1026, 1280, 2821] output_clusters=[25]>
[0x12c4:1] Manufacturer: CentraLite
[0x12c4:1] Model: 3320-L
[0x12c4:2] Discovering endpoint information
[0x12c4:2] Discovered endpoint information: <SimpleDescriptor endpoint=2 profile=49887 device_type=12 device_version=0 input_clusters=[0, 1, 3, 2821, 64527] output_clusters=[3]>
[0x12c4] Endpoint 2 initialization failure: 
Removing device 0x12c4 (00:0d:6f:00:0b:be:55:cb)
Device 0x4f9f (00:0d:6f:00:0b:be:55:cb) joined the network
Device 00:0d:6f:00:0b:be:55:cb changed id (0x12c4 => 0x4f9f)
[0x4f9f:zdo] ZDO request 0x0013: [48725, 00:00:80:00:0d:6f:00:0b, 0]
[0x4f9f:2] Discovering endpoint information
[0x4f9f:2] Discovered endpoint information: <SimpleDescriptor endpoint=2 profile=49887 device_type=12 device_version=0 input_clusters=[0, 1, 3, 2821, 64527] output_clusters=[3]>
[0x4f9f:zdo] ZDO request 0x0006: [65533, 260, [], [1280]]
Sending 'zdo_leave_req' failed: 
[0x4f9f] Endpoint 2 initialization failure: 
Device not found for removal: 00:0d:6f:00:0b:be:55:cb
```

I know the Conbee II is working as I am able to pair GE Link lights immediately and was also able to pair a Quirky Tripper - with both working well afterwards.  I see that the 3320-L is listed as being supported so I am confused as to why it seems to attempt the pairing but in each case it seems to fail.  I have the Conbee II attached via a USB extension as advised by the vendor and have placed each sensor right next to the Conbee II to eliminate reception issues.

I researched every source I could before posting the issue here so I've run out of resources and ideas.  Any help or insight you might be able to provide would be greatly appreciated.  Please let me know if I can provide any additional information.

dmulcahey, 2019-06-07:
this isn't a quirks issue... but happy to help. Please set your log level to debug and try again. at a minimum: zigpy, zigpy-deconz and zha please.

rpitera, 2019-06-07:
I appreciate it; I honestly didn't think it was since so many other folks are using them without an issue.

I'll get the debug logs to you as soon as possible.  Thanks for your help!

rpitera, 2019-06-07:
Noob question:  Can you provide the proper syntax for setting debug on just those three?  I've done this with components before but never with libraries.  I tried setting the whole log to debug but I don't want you to have to wade through all the meaningless info.

dmulcahey, 2019-06-07:
https://www.home-assistant.io/components/logger/

Adminiuga, 2019-06-07:
`logger.set_level` service call to temporarily bump levels
```
{
    "asyncio": "debug"
    "homeassistant.components.zha": "debug"
    "zigpy": "debug"
    "bellows": "debug"
    "zigpy_xbee": "debug"
    "zigpy_deconz": "debug"
}
```

`logger:` `configuration.yaml` to set level permanently
```
logger:
  default: info
  logs:
    asyncio: debug
    homeassistant.components.zha: debug
    zigpy: debug
    bellows: debug
    zigpy_xbee: debug
    zigpy_deconz: debug
```

Adminiuga, 2019-06-07:
For whatever reason it failed to get back from the device the model and manufacturer for endpoint id 2. I'd guess it got a timeout reading those attributes, which is not seen here because of the logging levels. Without model/manufacturer endpoint is considered uninitialized (incomplete discovers) so it removes the device.

```
[0x12c4:1] Manufacturer: CentraLite
[0x12c4:1] Model: 3320-L
[0x12c4:2] Discovering endpoint information
[0x12c4:2] Discovered endpoint information: <SimpleDescriptor endpoint=2 profile=49887 device_type=12 device_version=0 input_clusters=[0, 1, 3, 2821, 64527] output_clusters=[3]>
[0x12c4] Endpoint 2 initialization failure: 
```

rpitera, 2019-06-07:
Thanks @Adminiuga - I knew how to set error levels, just not for these components in particular.

Sorry for the delay but here's the logs from startup up to while attempting to pair the device:

```
2019-06-07 11:20:17 DEBUG (MainThread) [zigpy_deconz.uart] Connection made
2019-06-07 11:20:17 DEBUG (MainThread) [zigpy.appdb] Loading application state from /config/zigbee.db
2019-06-07 11:20:17 DEBUG (MainThread) [zigpy.appdb] Attribute id: 4 value: b'GE_Appliances'
2019-06-07 11:20:17 DEBUG (MainThread) [zigpy.appdb] Attribute id: 5 value: b'ZLL Light'
2019-06-07 11:20:17 DEBUG (MainThread) [zigpy.appdb] Attribute id: 4 value: b'Sercomm Corp.'
2019-06-07 11:20:17 DEBUG (MainThread) [zigpy.appdb] Attribute id: 5 value: b'Tripper'
2019-06-07 11:20:17 DEBUG (MainThread) [zigpy.appdb] Attribute id: 1 value: 21
2019-06-07 11:20:17 DEBUG (MainThread) [zigpy.appdb] Attribute id: 7 value: 3
2019-06-07 11:20:17 DEBUG (MainThread) [zigpy.appdb] Attribute id: 49 value: 2
2019-06-07 11:20:17 DEBUG (MainThread) [zigpy.appdb] Attribute id: 51 value: 2
2019-06-07 11:20:17 DEBUG (MainThread) [zigpy.appdb] Attribute id: 2 value: 49
2019-06-07 11:20:17 DEBUG (MainThread) [zigpy.appdb] Attribute id: 0 value: 1
2019-06-07 11:20:17 DEBUG (MainThread) [zigpy.appdb] Attribute id: 32 value: 26
2019-06-07 11:20:17 DEBUG (MainThread) [zigpy.appdb] Attribute id: 7 value: 1
2019-06-07 11:20:17 DEBUG (MainThread) [zigpy.appdb] Attribute id: 0 value: 0
2019-06-07 11:20:17 DEBUG (MainThread) [zigpy.appdb] Attribute id: 0 value: 218
2019-06-07 11:20:17 DEBUG (MainThread) [zigpy.quirks] Considering <class 'zigpy.quirks.smartthings.SmartthingsMultiPurposeSensor'>
2019-06-07 11:20:17 DEBUG (MainThread) [zigpy.quirks] Fail because device_type mismatch on at least one endpoint
2019-06-07 11:20:17 DEBUG (MainThread) [zigpy.quirks] Considering <class 'zigpy.quirks.kof.CeilingFan'>
2019-06-07 11:20:17 DEBUG (MainThread) [zigpy.quirks] Fail because device_type mismatch on at least one endpoint
2019-06-07 11:20:17 DEBUG (MainThread) [zigpy.quirks] Considering <class 'zigpy.quirks.keen.KeenTemperatureHumiditySensor'>
2019-06-07 11:20:17 DEBUG (MainThread) [zigpy.quirks] Fail because device_type mismatch on at least one endpoint
2019-06-07 11:20:17 DEBUG (MainThread) [zigpy.quirks] Considering <class 'zigpy.quirks.ikea.TradfriPlug'>
2019-06-07 11:20:17 DEBUG (MainThread) [zigpy.quirks] Fail because endpoint list mismatch: dict_keys([1, 2, 242]) {1}
2019-06-07 11:20:17 DEBUG (MainThread) [zigpy.quirks] Considering <class 'zhaquirks.centralite.3130.CentraLite3130'>
2019-06-07 11:20:17 DEBUG (MainThread) [zigpy.quirks] Fail because device_type mismatch on at least one endpoint
2019-06-07 11:20:17 DEBUG (MainThread) [zigpy.quirks] Considering <class 'zhaquirks.centralite.3300S.CentraLite3300S'>
2019-06-07 11:20:17 DEBUG (MainThread) [zigpy.quirks] Fail because endpoint list mismatch: dict_keys([1, 2]) {1}
2019-06-07 11:20:17 DEBUG (MainThread) [zigpy.quirks] Considering <class 'zhaquirks.centralite.3305S.CentraLite3305S'>
2019-06-07 11:20:17 DEBUG (MainThread) [zigpy.quirks] Fail because endpoint list mismatch: dict_keys([1, 2]) {1}
2019-06-07 11:20:17 DEBUG (MainThread) [zigpy.quirks] Considering <class 'zhaquirks.centralite.3310S.CentraLite3310S'>
2019-06-07 11:20:17 DEBUG (MainThread) [zigpy.quirks] Fail because device_type mismatch on at least one endpoint
2019-06-07 11:20:17 DEBUG (MainThread) [zigpy.quirks] Considering <class 'zhaquirks.centralite.3315S.CentraLite3315S'>
2019-06-07 11:20:17 DEBUG (MainThread) [zigpy.quirks] Fail because endpoint list mismatch: dict_keys([1, 2]) {1}
2019-06-07 11:20:17 DEBUG (MainThread) [zigpy.quirks] Considering <class 'zhaquirks.centralite.3320.CentraLite3320'>
2019-06-07 11:20:17 DEBUG (MainThread) [zigpy.quirks] Fail because endpoint list mismatch: dict_keys([1, 2]) {1}
2019-06-07 11:20:17 DEBUG (MainThread) [zigpy.quirks] Considering <class 'zhaquirks.centralite.3321S.CentraLite3321S'>
2019-06-07 11:20:17 DEBUG (MainThread) [zigpy.quirks] Fail because endpoint list mismatch: dict_keys([1, 2]) {1}
2019-06-07 11:20:17 DEBUG (MainThread) [zigpy.quirks] Considering <class 'zhaquirks.centralite.ias.CentraLiteIASSensor'>
2019-06-07 11:20:17 DEBUG (MainThread) [zigpy.quirks] Fail because endpoint list mismatch: dict_keys([1, 2]) {1}
2019-06-07 11:20:17 DEBUG (MainThread) [zigpy.quirks] Considering <class 'zhaquirks.centralite.motion.CentraLiteMotionSensor'>
2019-06-07 11:20:17 DEBUG (MainThread) [zigpy.quirks] Fail because endpoint list mismatch: dict_keys([1, 2]) {1}
2019-06-07 11:20:17 DEBUG (MainThread) [zigpy.quirks] Considering <class 'zhaquirks.gledopto.gls007z.GLS007Z'>
2019-06-07 11:20:17 DEBUG (MainThread) [zigpy.quirks] Fail because endpoint list mismatch: dict_keys([12, 11, 13]) {1}
2019-06-07 11:20:17 DEBUG (MainThread) [zigpy.quirks] Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
2019-06-07 11:20:17 DEBUG (MainThread) [zigpy.quirks] Fail because endpoint list mismatch: dict_keys([11, 13]) {1}
2019-06-07 11:20:17 DEBUG (MainThread) [zigpy.quirks] Considering <class 'zhaquirks.hivehome.mot003V0.MOT003'>
2019-06-07 11:20:17 DEBUG (MainThread) [zigpy.quirks] Fail because endpoint list mismatch: dict_keys([6]) {1}
2019-06-07 11:20:17 DEBUG (MainThread) [zigpy.quirks] Considering <class 'zhaquirks.hivehome.mot003V6.MOT003'>
2019-06-07 11:20:17 DEBUG (MainThread) [zigpy.quirks] Fail because endpoint list mismatch: dict_keys([6]) {1}
[... cut ...]
  File "/usr/local/lib/python3.7/site-packages/zigpy/device.py", line 129, in request
    expect_reply=expect_reply,
  File "/usr/local/lib/python3.7/site-packages/zigpy_deconz/zigbee/application.py", line 120, in request
    return await asyncio.wait_for(reply_fut, timeout)
  File "/usr/local/lib/python3.7/asyncio/tasks.py", line 423, in wait_for
    raise futures.TimeoutError()
concurrent.futures._base.TimeoutError
2019-06-07 11:24:58 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x1c72000c00050002176fffdf

```

dmulcahey, 2019-06-09:
@Adminiuga thoughts?

rpitera, 2019-06-09:
@dmulcahey and @Adminiuga just wanted to thank you both for even taking a look at this.  I can always go buy other sensors, but my worry is that there is something inherently wrong with my setup since this sensor seems to work for others.

Let me know if there is any other background info or data I can supply.

Adminiuga, 2019-06-10:
@dmulcahey need to take a second pass on the log. On the 1st glance, I see it gets two `device_joined` calls kicking off device discovery with the 2nd call cancelling the 1st device initialization. I don't think I see any timeouts and it makes me wonder why is it failing.

Adminiuga, 2019-06-11:
@rpitera 
- how far are the sensors from Deconz?
- can you use a short USB extension cable to move deconz a little bit further from the server?

Adminiuga, 2019-06-11:
and please try to factory reset the devices prior joining.  I see it was able talking to coordinator, prior you issued `zha.permit` service call, means the device was still on the network. I see incoming `ZDO Device Announcement` at least 4 times and each of those is kicking off a device discovery routine but it also has to cancel the previous routine. Not sure why do you get so many. Are you removing the battery?

Adminiuga, 2019-06-12:
@rpitera you are running HASSOS image, right? not a hassio script on a linux host? Do you know how to install a dev version of `zigpy-deconz`? 
Those sensors are Zigbe End devices, so they need a parent to communicate. The version of zigpy-deconz you are on, does not support broadcasts and I suspect that does not open other routers for joining, which means the only parent device those sensors could use is deconz itself and hence the question how far are the sensors from deconz.

I'd like you to try installing dev version of zigpy-deconz, to see if if helps with the issue.

rpitera, 2019-06-12:
> * how far are the sensors from Deconz?
> * can you use a short USB extension cable to move deconz a little bit further from the server?

1. Right next to the Conbee II stick
2. When I first installed the stick, the instructions say to put it on an extension cable so it's been on a long extension cable since day one.  Yesterday, before reading this I had tried pairing from a 6 foot distance from the server but it didn't make a difference.

> and please try to factory reset the devices prior joining. I see it was able talking to coordinator, prior >you issued zha.permit service call, means the device was still on the network. I see [...]

Every time I have tried pairing, I started with a factory reset.  With the Iris contact sensor, this requires you to remove the battery and hold down the button while replacing it so I have been removing the battery.   I have also been trying with three different Iris sensors so that may account for what you are seeing.  Also, I have been removing the batteries after the attempts since I believed there was no point in having the batteries in the sensor when they were being stored.  Was this the wrong thing to do?

BTW, I paired another Quirky Tripper contact sensor in the meantime - it paired almost instantly with no issue.

>@rpitera you are running HASSOS image, right? not a hassio script on a linux host? Do you know how >to install a dev version of zigpy-deconz?

I'm running hass.io on a virtualbox host, so not sure how (or if) I could run a dev version of zigpy-deconz but if it's possible I would definitely be up for trying it.

Adminiuga, 2019-06-12:
yeah, I can confirm the problem is somewhere in `zigpy-deconz` but not sure exactly what is causing it. 
Ordered a 3320-L so I could test with it. Please open an issue in `zigpy-deconz`  and move this thread there.

Adminiuga, 2019-06-12:
as a workaround, here's a dump for my sensor:
```
INSERT INTO devices VALUES('00:0d:6f:aa:bb:cc:dd:ee',17636,2);
INSERT INTO attributes VALUES('00:0d:6f:aa:bb:cc:dd:ee',1,0,4,X'43656e7472614c697465');
INSERT INTO attributes VALUES('00:0d:6f:aa:bb:cc:dd:ee',1,0,5,X'333332302d4c');
INSERT INTO attributes VALUES('00:0d:6f:aa:bb:cc:dd:ee',2,0,4,X'43656e7472614c697465');
INSERT INTO attributes VALUES('00:0d:6f:aa:bb:cc:dd:ee',2,0,5,X'333332302d4c');
INSERT INTO attributes VALUES('00:0d:6f:aa:bb:cc:dd:ee',1,1,51,1);
INSERT INTO attributes VALUES('00:0d:6f:aa:bb:cc:dd:ee',1,1,49,2);
INSERT INTO attributes VALUES('00:0d:6f:aa:bb:cc:dd:ee',1,1280,2,32);
INSERT INTO attributes VALUES('00:0d:6f:aa:bb:cc:dd:ee',1,1280,0,0);
INSERT INTO attributes VALUES('00:0d:6f:aa:bb:cc:dd:ee',1,1,62,0);
INSERT INTO attributes VALUES('00:0d:6f:aa:bb:cc:dd:ee',1,0,7,3);
INSERT INTO attributes VALUES('00:0d:6f:aa:bb:cc:dd:ee',2,0,7,3);
INSERT INTO attributes VALUES('00:0d:6f:aa:bb:cc:dd:ee',1,1280,1,21);
INSERT INTO attributes VALUES('00:0d:6f:aa:bb:cc:dd:ee',1,1026,0,2175);
INSERT INTO attributes VALUES('00:0d:6f:aa:bb:cc:dd:ee',1,1,32,28);
INSERT INTO attributes VALUES('00:0d:6f:aa:bb:cc:dd:ee',1,1,33,200);
INSERT INTO node_descriptors VALUES('00:0d:6f:aa:bb:cc:dd:ee',X'024080dfc25252000000520000');
INSERT INTO endpoints VALUES('00:0d:6f:aa:bb:cc:dd:ee',1,260,1026,1);
INSERT INTO endpoints VALUES('00:0d:6f:aa:bb:cc:dd:ee',2,49887,12,1);
INSERT INTO clusters VALUES('00:0d:6f:aa:bb:cc:dd:ee',1,0);
INSERT INTO clusters VALUES('00:0d:6f:aa:bb:cc:dd:ee',1,1);
INSERT INTO clusters VALUES('00:0d:6f:aa:bb:cc:dd:ee',1,1026);
INSERT INTO clusters VALUES('00:0d:6f:aa:bb:cc:dd:ee',1,3);
INSERT INTO clusters VALUES('00:0d:6f:aa:bb:cc:dd:ee',1,2821);
INSERT INTO clusters VALUES('00:0d:6f:aa:bb:cc:dd:ee',1,32);
INSERT INTO clusters VALUES('00:0d:6f:aa:bb:cc:dd:ee',1,1280);
INSERT INTO clusters VALUES('00:0d:6f:aa:bb:cc:dd:ee',2,0);
INSERT INTO clusters VALUES('00:0d:6f:aa:bb:cc:dd:ee',2,1);
INSERT INTO clusters VALUES('00:0d:6f:aa:bb:cc:dd:ee',2,3);
INSERT INTO clusters VALUES('00:0d:6f:aa:bb:cc:dd:ee',2,2821);
INSERT INTO clusters VALUES('00:0d:6f:aa:bb:cc:dd:ee',2,64527);
INSERT INTO output_clusters VALUES('00:0d:6f:aa:bb:cc:dd:ee',1,25);
INSERT INTO output_clusters VALUES('00:0d:6f:aa:bb:cc:dd:ee',2,3);
```
save into a file and replace with your IEEE address `00:0d:6f:00:0b:be:55:cb` then import it with `sqlite3 zigbee.db <centralite-3320-l.txt` after that
- remove the battery
- zha.permit
- factory reset by holding button

I'm hoping since the device is already in the database, it just going to update NWK and would work. Lemme know

rpitera, 2019-06-12:
> as a workaround, here's a dump for my sensor:

@Adminiuga - Opened the new issue in the zigpy-deconz repo and I'll give your workaround a shot and reply with my findings.  It will be tomorrow before I get a shot at it, and I really do appreciate the help.  In the meantime, I've added a few more GE Link Lights without any problems, so I'll do a snapshot and a save of the db before I test this out.

@dmulcahey -  I wanted to thank you again for your time and effort in this issue; I was pretty stumped.   I'll leave it up to you if you want to close this (since we have the other issue linked) or if you want it open until I post my results of the workaround tomorrow.    Your call, sir!

Adminiuga, 2019-06-12:
Not the quirks issue. Something up with zigpy-deconz, which causes it to fail discovery, because it doesn't get all the data. and could be related to some other reported issues with deconz.
You may close this one here, since unrelated to zhaquirks

rpitera, 2019-06-12:
Closing this then and thanks to you both.

closed 2019-06-12 by rpitera