zigpy/zha-device-handlers #94: Support: ORVIBO - Motion Sensor - SN10ZW-1CO

NickolaosC, 2019-06-13:
**Pairing**
First time I tried to add the device to Home Assistant nothing showed up on the screen.  Therefore I factory reset the device and pulled out my SmartThings hub to try it on there.  It was found on the first try and worked flawlessly.  Although I did not need to use the custom DHT, this was provided on the seller’s portal:
https://github.com/Kriskit/SmartThingsPublic/blob/master/devicetypes/kriskit/orvibo/orvibo-motion-sensor.groovy

**Success?**
After successfully removing the device from SmartThings I was able to successfully add the motion detector to Home Assistant using the built in ZHA support.  I am not sure what I did differently this time, I was simply happy it “worked”. It “detects” movements and I can see the nwk, ieee, rssi and battery level statuses.

**Issue**
The device indicates that it is unavailable until a detects a motion
It will remain online for about 2 hours during which time it continues to indicate that motion is detected even if I hide it in a box.  When the device finally times out ( or goes into sleep mode?) it will become once again unavailable.  Once motion there is any kind of motion presented to the device, the status changes to "Detected" once again and the 2 hour wait starts all over again.

Of note in the SmarThings DHT file linked above line 33 of the groovy code states the following
https://github.com/Kriskit/SmartThingsPublic/blob/master/devicetypes/kriskit/orvibo/orvibo-motion-sensor.groovy

"These devices don't report when motion stops, so it's necessary to have a timer to report that motion has stopped. You can adjust how long this is below."

My guess it is this behavior that is causing the problem.  Although my code / driver / device handler writing skills are terrible, I am assuming it should be an easy fix.  I simply am not sure where to start.

The only time it shows that the device is in the “Clear” status is right after joining the zigbee network.  Otherwise it will get stuck at “Detected” until it becomes “Unavailable” some 2hours and 20 minutes later.

Please note that I have already added a few Zigbee devices to HA including the Livingsmart Door/Windows Contact Sensor model SM10ZW (also rebranded from Orvibo) and a Multi Temperature White LED ceiling light.  All of them worked right out of the box.

**My Setup**
Home Assistant 0.93.2 using Hass.io installed on Proxmox using whiskerz007 script
ZWave & Zigbee connectivity: Nortek HUSBZB-1

**configuration.yaml (relevant section)**
```
zha:
  usb_path: /dev/ttyUSB1
  database_path: /config/zigbee.db
```
Manufacturer: ORVIBO
Model:		SN10ZW-1CO	(Livingsmart rebrand of Orvibo device)

**Logs**
Not sure what information is going to be relevant, but here is some information from the logs that I believe might be important.
```
Device 0x765d (00:15:xx:xx:xx:xx:xx:xx) left the network
Device 0x008e (00:15:xx:xx:xx:xx:xx:xx) joined the network
Device 00:15:xx:xx:xx:xx:xx:xx changed id (0x765d => 0x008e)
Canceling old initialize call
Failed ZDO request during device initialization: 
[0x008e] Requesting 'Node Descriptor'
[0x008e:zdo] ZDO request 0x0013: [142, 00:15:xx:xx:xx:xx:xx:xx, 128]
[0x008e] Node Descriptor: <NodeDescriptor byte1=2 byte2=64 mac_capability_flags=128 manufacturer_code=4151 maximum_buffer_size=127 maximum_incoming_transfer_size=100 server_mask=0 maximum_outgoing_transfer_size=100 descriptor_capability_field=0>
[0x008e] Discovering endpoints
[0x008e] Discovered endpoints: [1]
[0x008e:1] Discovering endpoint information
[0x008e:1] Discovered endpoint information: <SimpleDescriptor endpoint=1 profile=260 device_type=1026 device_version=1 input_clusters=[0, 3, 1280, 65535, 1] output_clusters=[0, 4, 3, 5, 1]>
[0x008e:1] Manufacturer: ORVIBO
[0x008e:1] Model: 895a2d80097f4ae2b2d40500d5e03dcc

ORVIBO 895a2d80097f4ae2b2d40500d5e03dcc: started configuration
node descriptor: [<Status.SUCCESS: 0>, 142, <NodeDescriptor byte1=2 byte2=64 mac_capability_flags=128 manufacturer_code=4151 maximum_buffer_size=127 maximum_incoming_transfer_size=100 server_mask=0 maximum_outgoing_transfer_size=100 descriptor_capability_field=0>]
ORVIBO 895a2d80097f4ae2b2d40500d5e03dcc: channel: zdo-ORVIBO 895a2d80097f4ae2b2d40500d5e03dcc_ZDO async_configure stage succeeded
0x008e:1:0x0000: finished channel configuration
0x008e:1:0x0500: started IASZoneChannel configuration
initializing channel: basic from_cache: False
ORVIBO 895a2d80097f4ae2b2d40500d5e03dcc: channel: basic-0x008e:1:0x0000 async_configure stage succeeded
0x008e:1:0x0500: bound  'ias_zone' cluster: Status.SUCCESS
0x008e:1:0x0500: wrote cie_addr: 00:0d:6f:00:12:00:b6:a4 to 'ias_zone' cluster: [<WriteAttributesStatusRecord status=Status.SUCCESS attrid=0>]
0x008e:1:0x0500: finished IASZoneChannel configuration
ORVIBO 895a2d80097f4ae2b2d40500d5e03dcc: channel: ias_zone-0x008e:1:0x0500 async_configure stage succeeded
Data remains after deserializing ZCL frame
[0x008e:1:0x0500] ZCL request 0x0101: []
Enroll requested
[0x008e:1:0x0500] No handler for cluster command 1
0x008e:1:0x0001: Failed to bind 'power' cluster: 
[0x008e:1:0x0500] ZCL request 0x0100: [1, 0, 0, 0]
Updated alarm state: 1
[0x008e:1:0x0500] No handler for cluster command 0
0x008e:1:0x0001: reporting 'battery_voltage' attr on 'power' cluster: 30/900/1: Result: '[[<ConfigureReportingResponseRecord status=140 direction=0 attrid=32>]]'
```

liorm0505, 2019-09-06:
Hi Everyone, 
I also have such sensor and would realy love to have it working with ZHA. 
Would like to help implementing this.

jlongman, 2020-07-27:
I had these working with `zigbee2mqtt`,  is there a migration path for work from there to ZHA?

jlongman, 2020-07-27:
Note, I see that the Aqara one has the timeout also:
https://github.com/zigpy/zha-device-handlers/blob/b96ea1afe70d51a16dbe85b56a9c32ac08fa259d/zhaquirks/xiaomi/__init__.py#L288

jlongman, 2020-07-29:
In #426 I added some files related to this device, so far no joy.  Based on testing as a result of the great response from contrubutors/collaborators I did make some progress: my code is loaded for the device, but it's not working.

my original code referred to motion detection and I needed to change that to an IAS_ZONE device - I think I knew that and it was an oversite.  The aqara code I copied had a timeout (which is why I copied it) of 120s.  Looking at the groovy code it had _2s_.  I tried that and it didn't work anyways.

The error messages are confusing, `Received frame from unknown device: 0xb42d`,  some noise in ZCL messages, quite confusing.  So I thought I'd add the previous known working solution, zigbee2mqtt (etc) references for when I can get back to this:

- Note it's a livingwise device there:
  - https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices.js#L8059
  - ```        supports: 'occupancy',
        fromZigbee: [fz.battery, fz.ias_occupancy_alarm_1_with_timeout],```
 - it has a reference to a `ias_occupancy_alarm_1_with_timeout`
   - which is referened https://github.com/Koenkk/zigbee-herdsman-converters/blob/259acf98cb8f0f0254c4a7aadc0d5b2347cd9667/converters/fromZigbee.js#L747
    - ```            if (timeout !== 0) {
                store[deviceID] = setTimeout(() => {
                    publish({occupancy: false});
                    store[deviceID] = null;
                }, timeout * 1000);
            }
```

I also noticed, and it's worth clarifying, that the SmartThings device listed at the top is not the one I've been using, the difference being I have model `895a2d80097f4ae2b2d40500d5e03dcc` and the smarthings model is `585fdfb8c2304119a2432e9845cf2623` and the cluster IDs do not match.

jlongman, 2020-08-01:
My work in progress is [here](https://github.com/jlongman/zha-device-handlers/commit/48f2222d938d30e74dbffd02b5707d22cea471d6#diff-084196f388a63553e659de245ac0d962).

It pairs and is identified as the correct quick but it still stays in one state and doesn't reset.  I played with the [timer](https://github.com/jlongman/zha-device-handlers/commit/48f2222d938d30e74dbffd02b5707d22cea471d6#diff-084196f388a63553e659de245ac0d962R60) a bunch  (120, 2, 15) but no joy.

liorm0505, 2020-08-03:
I am using a sensor with the same model number as the original code. 
I have modified the model number in the code but i am not sure the correct quirk is identified. 
Here is the log from the pairing process:
[log.txt](https://github.com/zigpy/zha-device-handlers/files/5018049/log.txt)

This is how it eventually beeing paird:
[image]

Is there anything else i need to change in the code in order for the right quirk to be identified? 

BTW - i forked your code and made my own changes here:

[https://github.com/liorm0505/zha-device-handlers/tree/orvibo/zhaquirks/orvibo](url)

Thanks,

jlongman, 2020-08-04:
Hi, so it looks like we have two different devices. It’s fine for testing but will have to fix it eventually if we ever get it working. Also I’ve updated my code since you forked it because, as shown in the new documentation, I have the wrong cluster used in the `replacement` section.  Not working yet unfortunately.

cydia2020, 2020-12-28:
Hi, my Orvibo SN10ZW has the model ID 700ae5aab3414ec09c1872efe7b8755, seems like there are multiple variants of this device.

closed 2021-08-20 by github-actions[bot]