zigpy/zha-device-handlers #95: Support for llkzmk11lm 2 Channels relay Xiaomi Aqara

Nemesis24, 2019-06-14:
[(https://www.banggood.com/fr/XIAOMI-Aqara-LLKZMK11LM-Two-way-Control-Module-Wireless-Relay-Controller-2-Channels-Work-For-Mijia-APP-Home-Kit-Control-Module-p-1441658.html?gmcCountry=FR&currency=EUR&createTmp=1&utm_source=googleshopping&utm_medium=cpc_bgs&utm_content=frank&utm_campaign=pla-mix-fr-pc-0308&ad_id=337073179860&gclid=EAIaIQobChMIld6sqqLp4gIVT1XTCh0v4QBlEAQYAyABEgIFo_D_BwE&cur_warehouse=CN)]

dmulcahey, 2019-06-14:
Works ootb

Nemesis24, 2019-06-14:
Sorry what's ootb ?

dmulcahey, 2019-06-14:
Out of the box. It just works

Nemesis24, 2019-06-14:
Ok so I've got a problem , i will try to find.

Nemesis24, 2019-06-15:
It doesn't work for me.
```
[0xc9a7:1:0x0400] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=uint16_t, value=0>>]]
[0xc9a7:1:0x0400] Attribute report received: measured_value=0
[0xc9a7:1:0x0406] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=bitmap8, value=1>>]]
[0xc9a7:1:0x0406] Attribute report received: occupancy=1
Updated alarm state: 1
Device 0xc19d (00:15:8d:00:03:cb:e2:9f) joined the network
Device 00:15:8d:00:03:cb:e2:9f changed id (0xdaee => 0xc19d)
[0xc19d:zdo] ZDO request 0x0013: [52194, 00:00:8e:00:15:8d:00:03, 0]
[0xc19d] Requesting 'Node Descriptor'
[0xc19d] Requesting 'Node Descriptor'
[0xc19d] Requesting Node Descriptor failed: 
[0xc19d] Discovering endpoints
[0xe96f:1:0x0400] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=uint16_t, value=0>>]]
[0xe96f:1:0x0400] Attribute report received: measured_value=0
[0xe96f] Requesting 'Node Descriptor'
[0xe96f:1:0x0406] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=bitmap8, value=1>>]]
[0xe96f:1:0x0406] Attribute report received: occupancy=1
Updated alarm state: 1
[0xfe80:1:0x0000] ZCL request 0x000a: [[<Attribute attrid=65281 value=<TypeValue type=bytes, value=b"d\x10\x00e\x10\x00\x03(\x16\x989\x00\x00\x00\x00\x959\xb3P\xec>\x05!&\x00\x9a \x00\x08!\x1f\x11\x07'\x00\x00\x00\x00\x00\x00\x00\x00\t!\x01\x06">>]]
[0xfe80:1:0x0000] Attribute report received: 65281=b"d\x10\x00e\x10\x00\x03(\x16\x989\x00\x00\x00\x00\x959\xb3P\xec>\x05!&\x00\x9a \x00\x08!\x1f\x11\x07'\x00\x00\x00\x00\x00\x00\x00\x00\t!\x01\x06"
[0x0de0:1:0x0402] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=int16s, value=1982>>]]
[0x0de0:1:0x0402] Attribute report received: measured_value=1982
[0x0de0] Requesting 'Node Descriptor'
[0x0de0:1:0x0405] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=uint16_t, value=6968>>]]
[0x0de0:1:0x0405] Attribute report received: measured_value=6968
[0x0de0:1:0x0403] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=int16s, value=1014>>, <Attribute attrid=20 value=<TypeValue type=int8s, value=-1>>, <Attribute attrid=16 value=<TypeValue type=int16s, value=10147>>]]
[0x0de0:1:0x0403] Attribute report received: measured_value=1014, 20=-1, 16=10147
[0xa363] Requesting Node Descriptor failed: 
```

dmulcahey, 2019-06-16:
it's not pairing for you... nothing I can do there. try moving the stick closer to the relay to pair it.

Adminiuga, 2019-06-16:
what's the radio? Enable debug logging and provide full logs, but from the log snippet you provided 
I see at least three devices being discovered and at least two are not responding.

Nemesis24, 2019-06-17:
It's ok, i found the problem, Mi Home was running an update when I removed the relay, so 
I added it and deleted it again, after that everything worked.
The radio is a raspbee.
For the others devices i think this is my lumi-lock, or the midding channels for lumi.ctrl_ln2.aq1.

I think this device needs a quirk because we don't have the power consumption.

Thanks for all.

Nemesis24, 2019-08-10:
https://github.com/zigpy/zigpy-deconz/issues/59

closed 2020-04-23 by Nemesis24