zigpy/zha-device-handlers #96: battery level incorrect for Xiaomi Aqara Door/Window Sensor

lolouk44, 2019-06-14:
Reported Battery level is always 100% (same as Phoscon/deCONZ) when Xiaomi Aqara hub reports (what I think is) the correct value of 41%.
In addition reported battery type is CR2016 when actual battery is CR2032.
Actual battery has been on shelf several years and in use in Aqara sensor for 6+ months hence my belief the value reported by Aqara hub is correct (and defo not 100%)
I use a ConbeeII with the latest firmware (deCONZ_ConBeeII_0x264a0700) on Ubuntu 16.04
Home-Assistant 0.94.3
Hope I've given you everything you need.

dmulcahey, 2019-08-10:
The aqara door sensors are oval shaped and too small to accept a CR2032. The battery is actually a CR1632. 
[image]

scoobydoofus, 2019-08-15:
I see a similar issue with the battery levels reported in Home Assistant for both the door/window sensors and motion sensors.

They're connected via ZHA with a HUSBZB-1 stick in Hassio 97.2, working fine for detecting motion/opening and closing doors, but all the motion sensors show battery at 100% and the door sensors both show 91%, and it's been like that for weeks, no change at all. The motion sensors are around 6 months old so I doubt they are actually still 100%.

I looked at the ZHA info, the motion sensors are using "zhaquirks.xiaomi.aqara.motion_aq2.MotionAQ2" and the door sensors are using "zhaquirks.xiaomi.aqara.magnet_aq2.MagnetAQ2".

Let me know if you need any more info to investigate this.

dmulcahey, 2019-10-16:
#174 potentially fixes the battery level issue.

closed 2019-10-16 by dmulcahey

rpatterson, 2019-10-17:
This is still happening for me and I think #174 doesn't fix this.  While wrapping my head around #176 I learned how to lookup device cluster attributes in the ZHA Configuration UI and I can't find any that return another other than `None`, including [the attribute IDs used in the quirk](https://github.com/rpatterson/zha-device-handlers/blob/dev/zhaquirks/xiaomi/__init__.py#L200-L202).  Looking at the SmartThings handler I was using before moving to HA, it seems like [it may not be using attributes at all](https://github.com/bspranger/Xiaomi/blob/master/devicetypes/bspranger/xiaomi-aqara-door-window-sensor.src/xiaomi-aqara-door-window-sensor.groovy#L169-L189).  So I think this should be reopened, @dmulcahey.

rpatterson, 2019-10-17:
Odd, I just checked again and my door sensor battery has since updated from 100% to 49%.  So it may be that switching to a checkout of this repo addressed the issue, the PR above does fix it, and the empty values from the attribute queries don't indicate a problem.

dmulcahey, 2019-10-17:
Xiaomi doesn’t use standard clusters and attributes so what you did to read attrs for centralite/sylvania won’t work for Xiaomi. Glad this is working.

highground88, 2019-11-15:
Forgive my ignorance, but is this merge still working? I am also getting battery readings of 100% across all my Aqara sensors (HUSBZB-1), and am unsure of how to fix this.  I'm running HASSIO 0.101.3 in Synology Docker.

dmulcahey, 2019-11-15:
The changes are in the beta not in the current release

highground88, 2019-11-15:
Great thanks - that makes sense - I'm not seeing any values under the battery_voltage etc clusters either. I will await the release! Thanks!