zigpy/zha-device-handlers #98: Lutron Auroa

tube0013, 2019-06-22:
So I pre-ordered 2 of these and just received them.

They pair by double clicking the button.

here is the debug log from pairing with HA 0.94.4, and some debug logging for actions after pairing.  Let me know if I can do any more! thanks!

```
[0x0000:zdo] ZDO request 0x0036: [60, <Bool.false: 0>]
[0x14d3:1:0x0000] ZCL request 0x000a: [[<Attribute attrid=65281 value=<TypeValue type=bytes, value=b'\x01!\xdb\x0b\x03(\x19\x04!\xa8\x13\x05!$\x01\x06$\x01\x00\x00\x00\x00\n!9\x0fd\x10\x00'>>]]
[0x14d3:1:0x0000] Attribute report received: 65281=b'\x01!\xdb\x0b\x03(\x19\x04!\xa8\x13\x05!$\x01\x06$\x01\x00\x00\x00\x00\n!9\x0fd\x10\x00'
0xb055:3:0x0b04 async_update
0xabfb:3:0x0b04 async_update
0x7bc5:3:0x0b04 async_update
Device 0x7129 (ff:ff:00:0f:e7:fe:e1:9d) joined the network
[0x7129] Requesting 'Node Descriptor'
Device 0x7129 (ff:ff:00:0f:e7:fe:e1:9d) joined the network
Skip initialization for existing device ff:ff:00:0f:e7:fe:e1:9d
[0x7129:zdo] ZDO request 0x0013: [28969, ff:ff:00:0f:e7:fe:e1:9d, 128]
[0x6ec4:zdo] ZDO request 0x0000: [00:0d:6f:00:0a:ff:73:23, 0, 0]
[0x6ec4:zdo] ZDO request 0x0013: [28356, 90:fd:9f:ff:fe:65:df:8e, 142]
[0xe189:zdo] ZDO request 0x0013: [57737, f0:d1:b8:00:00:02:32:96, 142]
[0xe189:1:0x0006] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=Bool, value=Bool.true>>]]
[0xe189:1:0x0006] Attribute report received: on_off=Bool.true
[0xe189:1:0x0008] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=uint8_t, value=94>>]]
[0xe189:1:0x0008] Attribute report received: current_level=94
0xe189:1:0x0008: received attribute: 0 update with value: 94
[0x7129:zdo] ZDO request 0x0002: [0]
[0x7129:zdo] Unsupported ZDO request 0x0002
[0x9591:1:0x0019] ZCL request 0x0101: [0, 4454, 260, 285409283, 0]
[0x9591:1:0x0019] No handler for cluster command 1
[0x7129:zdo] ZDO request 0x0002: [0]
[0x7129:zdo] Unsupported ZDO request 0x0002
[0x6ec4:zdo] ZDO request 0x0000: [00:0d:6f:00:0a:ff:73:23, 0, 0]
[0x6ec4:1:0x0006] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=Bool, value=Bool.false>>]]
[0x6ec4:1:0x0006] Attribute report received: on_off=Bool.false
[0x7129:zdo] ZDO request 0x0002: [0]
[0x7129:zdo] Unsupported ZDO request 0x0002
Unknown ZDO cluster 0x8038
[0x6ec4:zdo] Unexpected ZDO reply 0x8038: b'\x00\x00\xf8\xff\x07\x14\x00\x0c\x00\x10\xa9\xad\xb1\xa7\xa8\xa5\xae\xb1\xa7\xa6\xa2\xb2\xb7\x9e\x9c\x9c'
[0x6ec4:zdo] ZDO request 0x0000: [00:0d:6f:00:0a:ff:73:23, 0, 0]
[0x0000:zdo] ZDO request 0x0001: [28356, 0, 0]
[0x6ec4:zdo] Unexpected ZDO reply 0x8001: [<Status.SUCCESS: 0>, 90:fd:9f:ff:fe:65:df:8e, 28356, 0, 0, []]
[0x6ec4:1:0x0008] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=uint8_t, value=254>>]]
[0x6ec4:1:0x0008] Attribute report received: current_level=254
0xdd57:1:0x0008: received attribute: 0 update with value: 254
[0x7129] Requesting Node Descriptor failed: [0x7129:0:0x0002]: message send failure: EmberStatus.DELIVERY_FAILED
[0x7129] Discovering endpoints
[0x7129] Discovered endpoints: [1]
[0x7129:1] Discovering endpoint information
[0x7129:1] Discovered endpoint information: <SimpleDescriptor endpoint=1 profile=260 device_type=2080 device_version=1 input_clusters=[0, 1, 3, 4096, 64512] output_clusters=[3, 4, 6, 8, 25, 4096]>
[0x7129:1] Manufacturer: Lutron
[0x7129:1] Model: Z3-1BRL
Considering <class 'zigpy.quirks.smartthings.SmartthingsMultiPurposeSensor'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zigpy.quirks.kof.CeilingFan'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zigpy.quirks.keen.KeenTemperatureHumiditySensor'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zigpy.quirks.ikea.TradfriPlug'>
Fail because endpoint list mismatch: dict_keys([1, 2, 242]) {1}
Considering <class 'zhaquirks.centralite.3130.CentraLite3130'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.centralite.3300S.CentraLite3300S'>
Fail because endpoint list mismatch: dict_keys([1, 2]) {1}
Considering <class 'zhaquirks.centralite.3305S.CentraLite3305S'>
Fail because endpoint list mismatch: dict_keys([1, 2]) {1}
Considering <class 'zhaquirks.centralite.3310S.CentraLite3310S'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.centralite.3315S.CentraLite3315S'>
Fail because endpoint list mismatch: dict_keys([1, 2]) {1}
[... 198 lines cut ...]
2019-06-22 13:50:12 DEBUG (MainThread) [zigpy.zcl] [0x16e5:1:0x0008] ZCL request 0x0104: [132, 2]
2019-06-22 13:50:12 DEBUG (MainThread) [zigpy.zcl] [0x16e5:1:0x0008] ZCL request 0x0104: [139, 2]
2019-06-22 13:50:13 DEBUG (MainThread) [zigpy.zcl] [0x16e5:1:0x0008] ZCL request 0x0104: [153, 2]
2019-06-22 13:50:13 DEBUG (MainThread) [zigpy.zcl] [0x16e5:1:0x0008] No handler for cluster command 4
2019-06-22 13:50:13 DEBUG (MainThread) [zigpy.zcl] [0x16e5:1:0x0008] ZCL request 0x0104: [175, 2]
2019-06-22 13:50:13 DEBUG (MainThread) [zigpy.zcl] [0x16e5:1:0x0008] ZCL request 0x0104: [190, 2]
2019-06-22 13:50:13 DEBUG (MainThread) [zigpy.zcl] [0x16e5:1:0x0008] ZCL request 0x0104: [204, 2]
2019-06-22 13:50:13 DEBUG (MainThread) [zigpy.zcl] [0x16e5:1:0x0008] ZCL request 0x0104: [219, 2]
2019-06-22 13:50:14 DEBUG (MainThread) [zigpy.zcl] [0x16e5:1:0x0008] ZCL request 0x0104: [226, 2]
2019-06-22 13:50:14 DEBUG (MainThread) [zigpy.zcl] [0x16e5:1:0x0008] No handler for cluster command 4
2019-06-22 13:50:14 DEBUG (MainThread) [zigpy.zcl] [0x16e5:1:0x0008] ZCL request 0x0104: [234, 2]
```

Counterclockwise turn:

```
2019-06-22 13:54:29 DEBUG (MainThread) [zigpy.zcl] [0x16e5:1:0x0008] ZCL request 0x0000: [[0]]
2019-06-22 13:54:29 DEBUG (MainThread) [zigpy.zcl] [0x16e5:1:0x0008] No handler for general command 0
2019-06-22 13:54:29 DEBUG (MainThread) [zigpy.zcl] [0x16e5:1:0x0008] ZCL request 0x0104: [197, 2]
2019-06-22 13:54:29 DEBUG (MainThread) [zigpy.zcl] [0x16e5:1:0x0008] No handler for cluster command 4
2019-06-22 13:54:29 DEBUG (MainThread) [zigpy.zcl] [0x16e5:1:0x0008] ZCL request 0x0104: [124, 2]
2019-06-22 13:54:29 DEBUG (MainThread) [zigpy.zcl] [0x16e5:1:0x0008] ZCL request 0x0104: [66, 2]
2019-06-22 13:54:29 DEBUG (MainThread) [zigpy.zcl] [0x16e5:1:0x0008] ZCL request 0x0104: [22, 2]
2019-06-22 13:54:30 DEBUG (MainThread) [zigpy.zcl] [0x16e5:1:0x0008] ZCL request 0x0104: [2, 2]
```

Adminiuga, 2019-06-22:
Enable `homeassistant.core` debug logging, I think you should be getting
zha events

dmulcahey, 2019-06-25:
this shouldn't need a quirk. Binding and events should work OOTB.

tube0013, 2019-06-26:
Okay finally got around to testing.  I think it took a restart of HA to see the events but I'm only seeing them with the dimmer - turning it works great the args count up/down from 2 to 255 as you turn!

However the on/off tap action is not firing an event.  

https://youtu.be/62eszCs5P58

GodOfWar526, 2019-08-17:
Just commenting to refresh the issue. I've recently purchased one of these devices and have noticed the same exact issue. The toggle seems to be on ARG 7 which isn't being consistently read despite the device remembering previous toggle state.

lordkev, 2020-04-16:
@dmulcahey Any ideas why binding wouldn't work for this device? The events come through fine, but no matter if I try to bind it to a group or an individual device it doesn't seem to work. Though I'm having the same issues with my Hue Dimmer Switch, so I'm wondering if there's a broader bug with binding in the latest stable? Or possibly just an issue with my setup?

I see messages such as the following when I tap the button:

```
[0xbe33:1:0x0008] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=True> manufacturer=None tsn=83 command_id=4>
[0xbe33:1:0x0008] ZCL request 0x0004: [255, 7]
[0xbe33:1:0x0008] No handler for cluster command 4
```

dmulcahey, 2020-04-16:
I think it may be something with hue and lutron specifically. They do things a bit different. I have to look into it when I have a moment.

lordkev, 2020-04-16:
@dmulcahey Thanks, I appreciate it. FWIW, my Lutron Connected Bulb Remote which uses the quirk that utilizes the GroupBoundCluster function does work as expected. I was thinking about trying that method out for these devices. Happy to submit PRs if that works, but I don't know the codebase well enough to know if I'd be breaking some other functionality with them.

tube0013, 2020-04-17:
Adding some information about my experience with Lutronn Aurora-

I have 2 Aurora’s that are binded to single bulbs (not groups). They were purchased when they first came out and it is my understanding that there is a newer firmware on devices manufacturers after October 2019.  

See page 16 here:
http://www.lutron.com/TechnicalDocumentLibrary/0301916_Aurora_Advanced_Install_Guide.pdf

Recently I had to update the binding on one of my Auroras as the bulb it was bound to was taken out of commission and replaced. I ended up resetting and repairing it to ha/zha and looked up the the manual to figure out the right steps as it had been a while. 

Step 5 on page 16 of the link manual discus removing the battery after pairing to a “unsupported” hub this is needed for older fw Auroras and not needed for newer fw apparently.

When I did this I was not able to bind the device to any bulb (was not even trying groups). I then reset it and did not do the battery removal step and it subsequently bonded fine to the bulb o was not able to pair it to when I did the instructed battery removal.

I have a new Aurora on order and due in a few days, I’m suspecting it will be on the newer firmware I will test it and try and find time to test the older device as well.  I will also try and ping Lutron to see what they say.

tube0013, 2020-04-20:
I just got in an Aurora that has firmware 3.4 (my others report 3.1) I was able to add it to HA right away and bind it to an individual bulb via HA bind device with no problem, and it worked fine.  I unbinded it from the bulb, add the bulb to a new zigbee group, and then successfully binded the aurora to the group and it functions - ON-OFF is consistent, dimming it not.

FWIW I think I initially added this ticket in hopes of getting Device Automation actions added.

drinfernoo, 2020-10-02:
I can confirm I'm having the exact same issues as above. How irritating.

a-bianucci, 2020-10-13:
I haven't been able to get it to bind at all.  I can see the Aurora in zha triggering events on both the button and the twist, however when I hit the bind button nothing happens.  This applies to both a group or a single light.

Am I missing something other then adding it to HA, and then trying to the bind option from the menu?

WilliamRandol, 2021-06-25:
I found this on the forums about the button press issue.

https://community.home-assistant.io/t/nortek-stick-conbee-2-and-the-lutron-aurora-dimmer-switch-button-push-issue/246992/5

closed 2021-12-29 by github-actions[bot]

Plausible0Bug, 2023-05-09:
I'm suffering from unreliable button press on Lutron aurora dimmer switch(Z3-1BRL) as well.
My hardware is Home assistant yellow.
I'm able to capture debug logs, hopefully someone can help take a look or guide me through creating a quirk, thank you.
[Z3-1BR_logs.txt](https://github.com/zigpy/zha-device-handlers/files/11426384/Z3-1BR_logs.txt)