zigpy/zha-device-handlers #107: Mija temperature sensor only reports 0.5 degree intervals

frog32, 2019-07-01:
When running the sensor `zhaquirks.xiaomi.mija.sensor_ht.Weather` on a xiaomi gateway it will report on 0.1 degrees Celsius intervals when running on ZHA it only pushes an update once it diverted 0.5 degrees from the previous value.

I was unable to manually configure the sensor to behave the same as on a xiaomi gateway and think that the default behavior should be similar as it took me some time to realize that it is just not publishing if there is only a slight temperature change and it not completely broken.

dmulcahey, 2019-07-01:
The device handles the reporting interval. Xiaomi devices do this completely on their own. There is nothing that we can do here.

frog32, 2019-07-01:
However if run on a xiaomi gateway it behaves different and knowing the sleepiness of the device I am certain that the gateway is not polling them.

Is it feasible to sniff on an existing network so we could figure out why the behavior is different on a xiaomi gateway?

dmulcahey, 2019-07-01:
Go for it. It’s not something I have time to invest in.

frog32, 2019-07-01:
I went ahead and ordered hardware to do exactly that. I will likely get back in August.

dmulcahey, 2019-07-18:
did you figure anything out here?

frog32, 2019-07-18:
I am still waiting for hardware to sniff the pairing between the sensor and the xiaomi gateway. Currently I think the solution could be with using the commented out cluster that allows this to be changed.

https://github.com/zigpy/zigpy/blob/dev/zigpy/zcl/clusters/measurement.py#L49

frog32, 2019-07-18:
By adding this line (type is t.int16s it seems) on my dev system and then setting the value to 0010 I was able to set one of those sensors to report in 0.1°C intervals. I will try to do the same on my production system but since my production uses hassio it is going to need some docker trickery to apply the change and validate.

dmulcahey, 2019-07-20:
cool... that would mean we need to PR Zigpy

Adminiuga, 2019-07-21:
> By adding this line (type is t.int16s it seems) 

Can you post the diff?

dmulcahey, 2019-07-21:
@Adminiuga I think he is referencing the 2 commented out lines in the temp cluster that are linked above.

Adminiuga, 2019-07-21:
Hrm, I'd love to know where those are coming from? Don't see those attributes in ZCL specs. And even if you define additional attributes for the cluster, temp sensor is still based on the attribute id 0x0000, right?

frog32, 2019-07-21:
Somehow my test device went back to report 0.5° intervals. Now that I have all the hardware to sniff traffic I will try to sniff pairing with the xiaomi gateway to see what it does.

frog32, 2019-08-11:
Sorry I was OOO but could now capture pairing of the same sensor to a xiaomi gateway. However I did not spot anything useful. I am willing to share the capture with people directly but will not post it here since it contains the network key.

Shulyaka, 2019-11-19:
Does RelativeHumidity cluster have similar attributes?

dmulcahey, 2020-03-06:
is this actually an issue?

Shulyaka, 2020-03-06:
Not really, I just wonder if precision could be improved for humidity as well

roger-, 2020-05-31:
Any progress on this?

TheJulianJES, 2020-12-29:
This should be fixed by https://github.com/home-assistant/core/pull/43442 if I'm correct.

closed 2021-08-20 by github-actions[bot]