zigpy/zha-device-handlers #115: Add support for xiaomi lumi plug?

barradan, 2019-07-19:
Wondering if support will ever be added? 

https://www.aliexpress.com/item/32876832843.html?spm=2114.search0302.3.45.2e4f4c9ezks19v&ws_ab_test=searchweb0_0%2Csearchweb201602_0_10130_453_454_10618_10132_536_10131_317_537_10133_319_10059_10696_10084_10083_10547_10887_10307_321_10548_322_10065_10068_10103_10884%2Csearchweb201603_0%2CppcSwitch_0&algo_pvid=c7d827dd-9cb6-4fc4-8031-3b304bd91f63&algo_expid=c7d827dd-9cb6-4fc4-8031-3b304bd91f63-5

Was using a thrown together version here: https://github.com/dmulcahey/zha-device-handlers/pull/53 however update from 0.93 to 0.96 has broken it.

dmulcahey, 2020-03-06:
2 of these and the relay are now supported

closed 2020-03-06 by dmulcahey

Draluak, 2020-07-06:
Is the support for lumi.plug limited to toggle on/off? When adding this device in HA (0.112.2) through a ConbeeII (newest firmware) i dont get any sensors or attributes for current and total consumption. I tried a few times, but no luck so far. 
By looking into the cluster i can see the endpoints and read the attribute  present_value, so it should be there generally. 

AnalogInput (Endpoint id: 2, Id: 0x000c, Type: in)
AnalogInput (Endpoint id: 2, Id: 0x000c, Type: out)
AnalogInput (Endpoint id: 3, Id: 0x000c, Type: in)
AnalogInput (Endpoint id: 3, Id: 0x000c, Type: out)

[image]

dmulcahey, 2020-07-06:
I need the signature for the device. Seems like you have a different model. Please open a new issue with the full signature