zigpy/zha-device-handlers #116: Smartthings moisture sensor v1 not returning battery (0.96)

ryanwinter, 2019-07-19:
I'm sure this was working in the past, however now I'm seeing 'unknown' battery for this device since upgrading to 0.96.

When I query the 'battery_voltage' attribute, I get 26 which indicates 100%.

dmulcahey, 2019-07-19:
Can I have the signature for the device?

ryanwinter, 2019-07-19:
here!
manufacturer=Smartthings
model=moisturev4

input_clusters=[0, 1, 3, 15, 32, 1026, 1280]
output_clusters=[25]

Ah, so it looks like the same signature as the motion, however the model is different.

dmulcahey, 2019-07-19:
Can I have a screenshot of the device card too in the config panel? Sorry, should have asked for that before.

ryanwinter, 2019-07-19:
Do you mean this?

[image]

dmulcahey, 2019-07-20:
yes. Thanks!

dmulcahey, 2019-07-20:
wanna try #117 and let me know if it fixes your issue?

closed 2019-07-21 by dmulcahey

ryanwinter, 2019-07-22:
> wanna try #117 and let me know if it fixes your issue?

Sorry for slow reply, was out for the weekend, LGTM.

PR #117 (2019-07-21): Rename movionV4 to iasV4 and add moistureV4 model info
Fixes #116