zigpy/zha-device-handlers #128: Centralite 3350 endpoint list mismatch

presslab-us, 2019-08-10:
I have a Centralite 3305 smart button (iris branded from Lowe's).  The battery reporting is broken like a lot of other Centralite devices.  There's already a 3305S quirk, but it doesn't quite match, as shown below.  I guess mine only has 1 endpoint, and the quirk has two.  Otherwise it looks like it would match.

What would be the best way to solve this?  Is it possible to match both devices, essentially making endpoint 2 optional?

```
Device 0x1f29 (00:0d:6f:00:03:19:77:dc) joined the network
[0x1f29] Requesting 'Node Descriptor'
[0x1f29:zdo] ZDO request 0x0006: [0xfffd, 260, [], [64582]]
[0x1f29:zdo] ZDO request 0x0013: [0x1f29, 00:0d:6f:00:03:19:77:dc, 128]
[0x1f29] Node Descriptor: <Optional byte1=2 byte2=64 mac_capability_flags=128 manufacturer_code=49887 maximum_buffer_size=82 maximum_incoming_transfer_size=82 server_mask=0 maximum_outgoing_transfer_size=82 descriptor_capability_field=0>
[0x1f29] Discovering endpoints
[0x1f29] Discovered endpoints: [1]
[0x1f29:1] Discovering endpoint information
[0x1f29:1] Discovered endpoint information: <SimpleDescriptor endpoint=1 profile=260 device_type=1026 device_version=0 input_clusters=[0, 1, 3, 1026, 1280, 32, 2821] output_clusters=[25]>
[0x1f29:1] Manufacturer: CentraLite
[0x1f29:1] Model: 3305
Checking quirks for CentraLite 3305 (00:0d:6f:00:03:19:77:dc)
Considering <class 'zhaquirks.centralite.3305S.CentraLite3305S'>
Fail because endpoint list mismatch: {1, 2} {1}
```

dmulcahey, 2019-08-10:
Use this pattern: https://github.com/dmulcahey/zha-device-handlers/blob/52f480ee595d69f4629c35333436bbf73c0a3170/zhaquirks/lutron/lzl4bwhl01remote.py#L87

And add a 2nd device to the same file.

presslab-us, 2019-08-10:
Fixed by https://github.com/dmulcahey/zha-device-handlers/pull/129

closed 2019-08-10 by presslab-us