zigpy/zha-device-handlers #136: OSRAM LIGHTIFY BR RGBW incorrectly match to tunable white quirk

tube0013, 2019-08-20:
This bulb gets matched incorrectly to the Osram A19 tunable white device. (which if https://github.com/dmulcahey/zha-device-handlers/issues/135 is addressed, would no longer work as this is a Color Bulb.

[image]

Log out put from Joining the bulb.

`[0x0000:zdo] ZDO request 0x0036: [60, <Bool.false: 0>]
Device 0xd1d1 (84:18:26:00:00:08:69:09) joined the network
[0xd1d1] Requesting 'Node Descriptor'
[0xd1d1:zdo] ZDO request 0x0013: [0xd1d1, 84:18:26:00:00:08:69:09, 14]
[0xd1d1:zdo] ZDO request 0x0036: [180, <Bool.false: 0>]
[0xd1d1] Node Descriptor: <Optional byte1=1 byte2=64 mac_capability_flags=142 manufacturer_code=48042 maximum_buffer_size=64 maximum_incoming_transfer_size=0 server_mask=0 maximum_outgoing_transfer_size=0 descriptor_capability_field=3>
[0xd1d1] Discovering endpoints
[0xd1d1] Discovered endpoints: [3]
[0xd1d1:3] Discovering endpoint information
[0xd1d1:3] Discovered endpoint information: <SimpleDescriptor endpoint=3 profile=260 device_type=258 device_version=0 input_clusters=[0, 3, 4, 5, 6, 8, 768, 2820, 64527] output_clusters=[25]>
[0xd1d1:3] Manufacturer: OSRAM
[0xd1d1:3] Model: LIGHTIFY BR RGBW
Checking quirks for OSRAM LIGHTIFY BR RGBW (84:18:26:00:00:08:69:09)
Considering <class 'zigpy.quirks.smartthings.SmartthingsMultiPurposeSensor'>
Fail because endpoint list mismatch: {1} {3}
Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
Fail because endpoint list mismatch: {1} {3}
Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
Fail because endpoint list mismatch: {11, 13} {3}
Considering <class 'zhaquirks.lutron.lzl4bwhl01remote.LutronLZL4BWHL01Remote2'>
Fail because endpoint list mismatch: {1} {3}
Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
Fail because endpoint list mismatch: {1} {3}
Considering <class 'zhaquirks.osram.a19twhite.A19TunableWhite'>
Found custom device replacement for 84:18:26:00:00:08:69:09: <class 'zhaquirks.osram.a19twhite.A19TunableWhite'>
[0xd1d1](LIGHTIFY BR RGBW): started configuration
[0xd1d1:ZDO](LIGHTIFY BR RGBW): channel: 'async_configure' stage succeeded
0xd1d1:3:0x0008: bound  'level' cluster: Status.SUCCESS
0xd1d1:3:0x0006: bound  'on_off' cluster: Status.SUCCESS
0xd1d1:3:0x0008: reporting 'current_level' attr on 'level' cluster: 1/900/1: Result: '[[<ConfigureReportingResponseRecord status=Status.SUCCESS>]]'
[0xd1d1:3:0x0300]: initializing channel: from_cache: False
0xd1d1:3:0x0006: reporting 'on_off' attr on 'on_off' cluster: 0/900/1: Result: '[[<ConfigureReportingResponseRecord status=Status.SUCCESS>]]'
[0xd1d1:3:0x0008]: finished channel configuration
[0xd1d1:3:0x0008]: channel: 'async_configure' stage succeeded
0xd1d1:3:0x0300: bound  'light_color' cluster: Status.SUCCESS
[0xd1d1:3:0x0006]: finished channel configuration
[0xd1d1:3:0x0006]: channel: 'async_configure' stage succeeded
0xd1d1:3:0x0000: bound  'basic' cluster: Status.SUCCESS
[0xd1d1:3:0x0000]: finished channel configuration
0xd1d1:3:0x0300: reporting 'current_x' attr on 'light_color' cluster: 30/900/1: Result: '[[<ConfigureReportingResponseRecord status=Status.SUCCESS>]]'
0xd1d1:3:0x0b04: bound  'electrical_measurement' cluster: Status.SUCCESS
[0xd1d1:3:0x0000]: initializing channel: from_cache: False
[0xd1d1:3:0x0000]: channel: 'async_configure' stage succeeded
0xd1d1:3:0x0b04: reporting 'active_power' attr on 'electrical_measurement' cluster: 30/900/1: Result: '[[<ConfigureReportingResponseRecord status=Status.UNREPORTABLE_ATTRIBUTE direction=0 attrid=1291>]]'
[0xd1d1:3:0x0008] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=uint8_t, value=255>>]]
[0xd1d1:3:0x0008] Attribute report received: current_level=255
[0xd1d1:3:0x0008]: received attribute: 0 update with value: 255
[0xd1d1:3:0x0006] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=Bool, value=Bool.true>>]]
[0xd1d1:3:0x0006] Attribute report received: on_off=Bool.true
[0xd1d1:3:0x0b04]: finished channel configuration
[0xd1d1:3:0x0b04]: channel: 'async_configure' stage succeeded
0xd1d1:3:0x0300: reporting 'current_y' attr on 'light_color' cluster: 30/900/1: Result: '[[<ConfigureReportingResponseRecord status=Status.SUCCESS>]]'
0xd1d1:3:0x0300: reporting 'color_temperature' attr on 'light_color' cluster: 30/900/1: Result: '[[<ConfigureReportingResponseRecord status=Status.INSUFFICIENT_SPACE direction=0 attrid=7>]]'
[0xd1d1:3:0x0300]: finished channel configuration
[0xd1d1:3:0x0300]: channel: 'async_configure' stage succeeded
[0xd1d1](LIGHTIFY BR RGBW): completed configuration
[0xd1d1](LIGHTIFY BR RGBW): stored in registry: ZhaDeviceEntry(name='OSRAM LIGHTIFY BR RGBW', ieee='84:18:26:00:00:08:69:09', last_seen=1566324703.978271)
[0xd1d1](LIGHTIFY BR RGBW): started initialization
[0xd1d1:ZDO](LIGHTIFY BR RGBW): entry loaded from storage: ZhaDeviceEntry(name='OSRAM LIGHTIFY BR RGBW', ieee='84:18:26:00:00:08:69:09', last_seen=1566324703.978271)
[0xd1d1:ZDO](LIGHTIFY BR RGBW): channel: 'async_initialize' stage succeeded
[0xd1d1:3:0x0300]: initializing channel: from_cache: True
[0xd1d1:3:0x0006]: attempting to update onoff state - from cache: False
[0xd1d1:3:0x0b04]: async_update
[0xd1d1:3:0x0006]: initializing channel: from_cache: False
[0xd1d1:3:0x0006]: channel: 'async_initialize' stage succeeded
light.osram_lightify_br_rgbw_00086909_3: polling current state
[0xd1d1:3:0x0000]: initializing channel: from_cache: False
[0xd1d1:3:0x0000]: channel: 'async_initialize' stage succeeded
[0xd1d1:3:0x0300]: channel: 'async_initialize' stage succeeded
[0xd1d1:3:0x0b04]: initializing channel: from_cache: False
[0xd1d1:3:0x0b04]: channel: 'async_initialize' stage succeeded
[0xd1d1:3:0x0008]: received attribute: 0 update with value: 255
[0xd1d1:3:0x0008]: initializing channel: from_cache: False
[0xd1d1:3:0x0008]: channel: 'async_initialize' stage succeeded
[0xd1d1:3:0x0008] Unexpected ZCL reply 0x0001: [[<ReadAttributeRecord attrid=0 status=0 value=255>]]
[0xd1d1](LIGHTIFY BR RGBW): power source: Mains
[0xd1d1](LIGHTIFY BR RGBW): completed initialization
[0xd1d1:zdo] ZDO request 0x0006: [0xfffd, 260, [25], []]
[0xd1d1:zdo] ZDO request 0x0001: [0x0000, 0, 0]
[0xd1d1:3:0x0019] ZCL request 0x0101: [0, 4364, 11, 16908681, None]
[0xd1d1:3:0x0019] No handler for cluster command 1
[0xd1d1:3:0x0300] ZCL request 0x000a: [[<Attribute attrid=3 value=<TypeValue type=uint16_t, value=24939>>]]
[0xd1d1:3:0x0300] Attribute report received: current_x=24939
[0xd1d1:3:0x0300] ZCL request 0x000a: [[<Attribute attrid=4 value=<TypeValue type=uint16_t, value=24701>>]]
[0xd1d1:3:0x0300] Attribute report received: current_y=24701
[0xd1d1:3:0x0b04]: async_update
`

dmulcahey, 2020-12-21:
@tube0013 what happened here? Is a quirk needed?

tube0013, 2020-12-21:
> @tube0013 what happened here? Is a quirk needed?

I don't have any more Osram stuff. I don't know if it is needed any longer so closing.

closed 2020-12-21 by tube0013