zigpy/zha-device-handlers #140: Add support for Sengled 4 button controller

doughecka, 2019-08-25:
Just picked this up from Bestbuy, able to pair it to HA via ZHA, but the buttons don't generate events. This is what I get when pairing:
[0x0000:zdo] ZDO request 0x0036: [60, <Bool.false: 0>]
Device 0x6ae4 (b0:ce:18:14:03:5d:cb:63) joined the network
[0x6ae4:zdo] ZDO request 0x0013: [0x6ae4, b0:ce:18:14:03:5d:cb:63, 128]
[0x6ae4] Requesting 'Node Descriptor'
[0x6ae4] Node Descriptor: <Optional byte1=2 byte2=64 mac_capability_flags=128 manufacturer_code=4448 maximum_buffer_size=82 maximum_incoming_transfer_size=82 server_mask=11264 maximum_outgoing_transfer_size=82 descriptor_capability_field=0>
Error calling listener.node_descriptor_updated: FOREIGN KEY constraint failed
[0x6ae4] Discovering endpoints
[0x6ae4] Discovered endpoints: [1]
[0x6ae4:1] Discovering endpoint information
[0x6ae4:1] Discovered endpoint information: <SimpleDescriptor endpoint=1 profile=260 device_type=260 device_version=0 input_clusters=[0, 1, 3, 32, 64529] output_clusters=[3, 4, 6, 8, 64528]>
[0x6ae4:1] Manufacturer: sengled
[0x6ae4:1] Model: E1E-G7F
Checking quirks for sengled E1E-G7F (b0:ce:18:14:03:5d:cb:63)
Considering <class 'zigpy.quirks.smartthings.SmartthingsMultiPurposeSensor'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
Fail because endpoint list mismatch: {11, 13} {1}
Considering <class 'zhaquirks.lutron.lzl4bwhl01remote.LutronLZL4BWHL01Remote2'>
Fail because profile_id mismatch on at least one endpoint
Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.osram.a19twhite.A19TunableWhite'>
Fail because endpoint list mismatch: {3} {1}
Considering <class 'zhaquirks.philips.rwl021.PhilipsRWL021'>
Fail because endpoint list mismatch: {1, 2} {1}
Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.xbee.xbee_io.XbeeSensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.xiaomi.mija.smoke.MijiaHoneywellSmokeDetectorSensor'>
Fail because device_type mismatch on at least one endpoint
[0x6ae4](E1E-G7F): started configuration
[0x6ae4:ZDO](E1E-G7F): channel: 'async_configure' stage succeeded
0x6ae4:1:0x0000: bound  'basic' cluster: Status.SUCCESS
[0x6ae4:1:0x0000]: finished channel configuration
0x6ae4:1:0x0001: bound  'power' cluster: Status.SUCCESS
0x6ae4:1:0x0006: bound  'on_off' cluster: Status.SUCCESS
[0x6ae4:1:0x0006]: finished channel configuration
[0x6ae4:1:0x0006]: channel: 'async_configure' stage succeeded
[0x6ae4:1:0x0000] Unexpected ZCL reply 0x0001: [[<ReadAttributeRecord attrid=4 status=0 value=sengled>, <ReadAttributeRecord attrid=5 status=0 value=E1E-G7F>]]
0x6ae4:1:0x0008: bound  'level' cluster: Status.SUCCESS
[0x6ae4:1:0x0008]: finished channel configuration
[0x6ae4:1:0x0008]: channel: 'async_configure' stage succeeded
[0x6ae4:1:0x0000]: initializing channel: from_cache: False
[0x6ae4:1:0x0000]: channel: 'async_configure' stage succeeded
0x6ae4:1:0x0001: reporting 'battery_voltage' attr on 'power' cluster: 30/900/1: Result: '[[<ConfigureReportingResponseRecord status=Status.SUCCESS>]]'
0x6ae4:1:0x0001: reporting 'battery_percentage_remaining' attr on 'power' cluster: 30/900/1: Result: '[[<ConfigureReportingResponseRecord status=Status.SUCCESS>]]'
[0x6ae4:1:0x0001]: finished channel configuration
[0x6ae4:1:0x0001]: channel: 'async_configure' stage succeeded
[0x6ae4](E1E-G7F): completed configuration
[0x6ae4](E1E-G7F): stored in registry: ZhaDeviceEntry(name='sengled E1E-G7F', ieee='b0:ce:18:14:03:5d:cb:63', last_seen=1566698421.7629156)
[0x6ae4](E1E-G7F): started initialization
[0x6ae4:ZDO](E1E-G7F): entry loaded from storage: ZhaDeviceEntry(name='sengled E1E-G7F', ieee='b0:ce:18:14:03:5d:cb:63', last_seen=1566698421.7629156)
[0x6ae4:ZDO](E1E-G7F): channel: 'async_initialize' stage succeeded
[0x6ae4:1:0x0006]: initializing channel: from_cache: False
[0x6ae4:1:0x0006]: channel: 'async_initialize' stage succeeded
[0x6ae4:1:0x0008]: initializing channel: from_cache: False
[0x6ae4:1:0x0008]: channel: 'async_initialize' stage succeeded
[0x6ae4:1:0x0000]: initializing channel: from_cache: False
[0x6ae4:1:0x0000]: channel: 'async_initialize' stage succeeded

When pressing buttons, I get the following (single tap each button top to bottom, then double tap top to bottom, then press+hold top to bottom):
[0xbb9a:10:0x0019] ZCL request 0x0101: [0, 4425, 1, 500, None]
[0xbb9a:10:0x0019] No handler for cluster command 1
Unknown cluster-specific command 0
[0x6ae4:1:0xfc10] ZCL request 0x0100: b'\x01\x00\x00\x00'
[0x6ae4:1:0xfc10] No handler for cluster command 0
Unknown cluster-specific command 0
[0x6ae4:1:0xfc10] ZCL request 0x0100: b'\x02\x00\x01\x00'
[0x6ae4:1:0xfc10] No handler for cluster command 0
[0xbb9a:10:0x0019] ZCL request 0x0101: [0, 4425, 1, 500, None]
[0xbb9a:10:0x0019] No handler for cluster command 1
Unknown cluster-specific command 0
[0x6ae4:1:0xfc10] ZCL request 0x0100: b'\x03\x00\x01\x00'
[0x6ae4:1:0xfc10] No handler for cluster command 0
Unknown cluster-specific command 0
[0x6ae4:1:0xfc10] ZCL request 0x0100: b'\x04\x00\x00\x00'
[0x6ae4:1:0xfc10] No handler for cluster command 0
Unknown cluster-specific command 0
[0x6ae4:1:0xfc10] ZCL request 0x0100: b'\x05\x00\x01\x00'
[0x6ae4:1:0xfc10] No handler for cluster command 0
[0xbb9a:10:0x0019] ZCL request 0x0101: [0, 4425, 1, 500, None]
[0xbb9a:10:0x0019] No handler for cluster command 1
Unknown cluster-specific command 0
[0x6ae4:1:0xfc10] ZCL request 0x0100: b'\x02\x00\x01\x00'
[0x6ae4:1:0xfc10] No handler for cluster command 0
Unknown cluster-specific command 0
[0x6ae4:1:0xfc10] ZCL request 0x0100: b'\x03\x00\x01\x00'
[0x6ae4:1:0xfc10] No handler for cluster command 0
Unknown cluster-specific command 0
[0x6ae4:1:0xfc10] ZCL request 0x0100: b'\x02\x00\x02\x00'
[0x6ae4:1:0xfc10] No handler for cluster command 0
Unknown cluster-specific command 0
[0x6ae4:1:0xfc10] ZCL request 0x0100: b'\x07\x00\x01\x00'
[0x6ae4:1:0xfc10] No handler for cluster command 0
[0xbb9a:10:0x0019] ZCL request 0x0101: [0, 4425, 1, 500, None]
[0xbb9a:10:0x0019] No handler for cluster command 1
Unknown cluster-specific command 0
[0x6ae4:1:0xfc10] ZCL request 0x0100: b'\x06\x00\x01\x00'
[0x6ae4:1:0xfc10] No handler for cluster command 0
Unknown cluster-specific command 0
[0x6ae4:1:0xfc10] ZCL request 0x0100: b'\x03\x00\x01\x00'
[0x6ae4:1:0xfc10] No handler for cluster command 0
[0xbb9a:10:0x0019] ZCL request 0x0101: [0, 4425, 1, 500, None]
[0xbb9a:10:0x0019] No handler for cluster command 1
Unknown cluster-specific command 0
[0x6ae4:1:0xfc10] ZCL request 0x0100: b'\x02\x00\x01\x00'
[0x6ae4:1:0xfc10] No handler for cluster command 0
Unknown cluster-specific command 0
[0x6ae4:1:0xfc10] ZCL request 0x0100: b'\x02\x00\x02\x00'
[0x6ae4:1:0xfc10] No handler for cluster command 0
Unknown cluster-specific command 0
[0x6ae4:1:0xfc10] ZCL request 0x0100: b'\x02\x00\x02\x00'
[0x6ae4:1:0xfc10] No handler for cluster command 0
Unknown cluster-specific command 0
[0x6ae4:1:0xfc10] ZCL request 0x0100: b'\x03\x00\x02\x00'
[0x6ae4:1:0xfc10] No handler for cluster command 0
Unknown cluster-specific command 0
[0x6ae4:1:0xfc10] ZCL request 0x0100: b'\x03\x00\x02\x00'
[0x6ae4:1:0xfc10] No handler for cluster command 0
Unknown cluster-specific command 0
[0x6ae4:1:0xfc10] ZCL request 0x0100: b'\x08\x00\x01\x00'
[0x6ae4:1:0xfc10] No handler for cluster command 0
[0xbb9a:10:0x0019] ZCL request 0x0101: [0, 4425, 1, 500, None]
[0xbb9a:10:0x0019] No handler for cluster command 1
[0x6ae4:1:0x0001] ZCL request 0x000a: [[<Attribute attrid=33 value=<TypeValue type=uint8_t, value=200>>]]
[0x6ae4:1:0x0001] Attribute report received: battery_percentage_remaining=200
[0xbb9a:10:0x0019] ZCL request 0x0101: [0, 4425, 1, 500, None]
[0xbb9a:10:0x0019] No handler for cluster command 1

doughecka, 2019-08-25:
This appears to be very similar looking to the Hue controller... perhaps I can just tweak that one and get it to work. This is the first time I've tried anything at this level, when using Hass.io what's the best way to experiment with making a change? Can I drop it in custom_components or would I have to add the entire zha-device-handlers source to custom_components?

dmulcahey, 2019-08-25:
with hass.io it isn’t trivial and I’m not a hass.io user so I’m not going to be much help with that.

tube0013, 2019-08-25:
I used to be able to use it In the deps folder but that doesn't work any more. Now I've had luck getting a console into the homeassistant container (I just use protainer to make that easy) and just updating / adding the quirks there.

doughecka, 2019-08-25:
From what I'm reading perhaps I can just drop a copy of the zha-device-handlers into my custom components? 
https://developers.home-assistant.io/blog/2019/04/12/new-integration-structure.html
Not sure how that will get used by ZHA though, unless I keep the name the same, but it says not to do that :D

doughecka, 2019-08-25:
ok, so I've managed to build a file that's accepted by the HA when adding the device, however it SEEMS like all the button presses are sent under the manufacturer custom cluster... is there a mechanism to 'convert' those to standard button presses or to simply kick out ZHA events? Is there an example quirk I can look at that does the same thing? Honestly if it just generates events containing the raw data that would work. 

[0x8c26:1:0xfc10] ZCL request 0x0100: b'\x01\x00\x00\x00'
[0x8c26:1:0xfc10] No handler for cluster command 0
Unknown cluster-specific command 0
[0x8c26:1:0xfc10] ZCL request 0x0100: b'\x01\x00\x00\x00'
[0x8c26:1:0xfc10] No handler for cluster command 0
Unknown cluster-specific command 0
[0x8c26:1:0xfc10] ZCL request 0x0100: b'\x02\x00\x01\x00'
[0x8c26:1:0xfc10] No handler for cluster command 0
Unknown cluster-specific command 0
[0x8c26:1:0xfc10] ZCL request 0x0100: b'\x06\x00\x01\x00'
[0x8c26:1:0xfc10] No handler for cluster command 0
Unknown cluster-specific command 0
[0x8c26:1:0xfc10] ZCL request 0x0100: b'\x05\x00\x01\x00'
[0x8c26:1:0xfc10] No handler for cluster command 0
Unknown cluster-specific command 0
[0x8c26:1:0xfc10] ZCL request 0x0100: b'\x01\x00\x00\x00'
[0x8c26:1:0xfc10] No handler for cluster command 0
Unknown cluster-specific command 0
[0x8c26:1:0xfc10] ZCL request 0x0100: b'\x02\x00\x01\x00'
[0x8c26:1:0xfc10] No handler for cluster command 0
Unknown cluster-specific command 0
[0x8c26:1:0xfc10] ZCL request 0x0100: b'\x02\x00\x01\x00'
[0x8c26:1:0xfc10] No handler for cluster command 0
Unknown cluster-specific command 0
[0x8c26:1:0xfc10] ZCL request 0x0100: b'\x02\x00\x02\x00'
[0x8c26:1:0xfc10] No handler for cluster command 0
Unknown cluster-specific command 0
[0x8c26:1:0xfc10] ZCL request 0x0100: b'\x02\x00\x01\x00'
[0x8c26:1:0xfc10] No handler for cluster command 0
Unknown cluster-specific command 0
[0x8c26:1:0xfc10] ZCL request 0x0100: b'\x02\x00\x02\x00'
[0x8c26:1:0xfc10] No handler for cluster command 0
Unknown cluster-specific command 0
[0x8c26:1:0xfc10] ZCL request 0x0100: b'\x02\x00\x01\x00'
[0x8c26:1:0xfc10] No handler for cluster command 0
Unknown cluster-specific command 0
[0x8c26:1:0xfc10] ZCL request 0x0100: b'\x02\x00\x02\x00'
[0x8c26:1:0xfc10] No handler for cluster command 0
Unknown cluster-specific command 0
[0x8c26:1:0xfc10] ZCL request 0x0100: b'\x02\x00\x02\x00'
[0x8c26:1:0xfc10] No handler for cluster command 0
Unknown cluster-specific command 0
[0x8c26:1:0xfc10] ZCL request 0x0100: b'\x02\x00\x02\x00'
[0x8c26:1:0xfc10] No handler for cluster command 0
Unknown cluster-specific command 0
[0x8c26:1:0xfc10] ZCL request 0x0100: b'\x02\x00\x02\x00'
[0x8c26:1:0xfc10] No handler for cluster command 0
[0x0000:zdo] ZDO request 0x0036: [60, <Bool.false: 0>]
Unknown cluster-specific command 0
[0x8c26:1:0xfc10] ZCL request 0x0100: b'\x01\x00\x00\x00'
[0x8c26:1:0xfc10] No handler for cluster command 0
Unknown cluster-specific command 0
[0x8c26:1:0xfc10] ZCL request 0x0100: b'\x05\x00\x01\x00'
[0x8c26:1:0xfc10] No handler for cluster command 0
Unknown cluster-specific command 0
[0x8c26:1:0xfc10] ZCL request 0x0100: b'\x06\x00\x01\x00'
[0x8c26:1:0xfc10] No handler for cluster command 0
Unknown cluster-specific command 0
[0x8c26:1:0xfc10] ZCL request 0x0100: b'\x04\x00\x00\x00'
[0x8c26:1:0xfc10] No handler for cluster command 0

dmulcahey, 2019-08-25:
Events are how it should work... you can translate them across clusters. Meaning create the clusters that you want them to fire on and use those clusters to send the appropriate messages. The Xiaomi quirks do this a lot so they’re good examples to follow.

doughecka, 2019-08-25:
Well, this was my attempt... but I'm not a Python Dev and I'm not familiar with what data is flowing where, so still not working, although I feel I'm close. If someone wants to take a glance at this and see what I'm missing, I'd appreciate it. :) 

https://github.com/doughecka/zha-device-handlers/tree/dev/zhaquirks/sengled

Adminiuga, 2019-08-26:
Most likely you don't need https://github.com/doughecka/zha-device-handlers/blob/6047e5a567d62962ab56f1bb77e9da7134e85d1d/zhaquirks/sengled/__init__.py#L26-L77 those are specifically for Xiaomi devices sending invalid data. Haven't seen any other device doing that.

for testing your fix on hassio, you may try an approach similar to https://github.com/zigpy/zigpy-deconz/issues/44#issuecomment-503810629, just make sure your branch of zha-quirks is the same version as version in `manifest.json` of ZHA component, otherwise HA will try to reinstall version from the manifest.

Check if you can find a SmartThings device handler as it may offer more insight into cluster specific command formatting. Per debug log you provided, it seems only the 1st byte and the 3rd bytes are changing. See if you can find a pattern between button presses (which button/short, long press, double tap etc) and the command payload.

blathers123, 2020-11-30:
Just got two of these switches. How can I help with the integration? It looks like there's a working driver here:
https://community.hubitat.com/t/sengled-smart-light-switch/23916/29

compeek, 2021-01-06:
There is also an implementation in zigbee2mqtt, and it looks pretty simple:
https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/converters/fromZigbee.js#L3062
https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices.js#L5914

I would really love to see this device supported as well. @doughecka I'm curious if you ever got anywhere further with this? I might try to get it working myself.

doughecka, 2021-01-06:
> There is also an implementation in zigbee2mqtt, and it looks pretty simple:
> https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/converters/fromZigbee.js#L3062 [...]

doughecka, 2021-01-06:
> There is also an implementation in zigbee2mqtt, and it looks pretty simple:
> https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/converters/fromZigbee.js#L3062 [...]

Unfortunately I never made any additional progress, but I still have it kicking around here somewhere.

compeek, 2021-01-06:
All right, it took me the entire evening and night (and I really need to go to bed now...), but I finally got something working. So far, I have it so simple presses of the On and Off buttons emit events. I still need to do the dim up and down buttons and see about double presses and long presses since I saw the zigbee2mqtt handler seems to support that, but now that I know more what I'm doing, that should be pretty straightforward.

I really struggled to figure out how to get the events to show up in Home Assistant. Nothing I was doing seemed to work, and after reading through a lot of source code, I found that Home Assistant ignores events from clusters it doesn't know about, such as the custom one this device sends button presses on. Then I finally realized that if I pass the information to one of the standard clusters via a bus, I can send a zha_event from there, and it will show up in Home Assistant.

I suppose that's kind of the whole point of the device handlers--translating custom logic and clusters into standard behavior--but it wasn't really clicking for quite a while, and unfortunately the documentation isn't too clear on a lot of this stuff.

Anyway, once I get this more polished, I'll try to see about creating a PR and hopefully getting this into the repo so it's an officially supported device for ZHA, but for now I'm sharing what I have so far in case anybody else wants to play with it.

```
import logging

import zigpy.types as t
from zigpy.profiles import zha
from zigpy.quirks import CustomCluster, CustomDevice
from zigpy.zcl.clusters.general import (
    Basic,
    Groups,
    Identify,
    LevelControl,
    OnOff,
    PollControl,
    PowerConfiguration,
)

from .. import Bus
from ..const import (
    COMMAND_OFF,
    COMMAND_ON,
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
    ZHA_SEND_EVENT,
)

_LOGGER = logging.getLogger(__name__)

class OnOffCluster(CustomCluster, OnOff):
    cluster_id = OnOff.cluster_id

    def __init__(self, *args, **kwargs):
        super().__init__(*args, **kwargs)
        
        self.endpoint.device.on_off_bus.add_listener(self)

class SengledRemoteCluster(CustomCluster):
    cluster_id = 64528

    manufacturer_server_commands = {
        0x0000: (
            "notification",
            (t.uint8_t, t.uint8_t, t.uint8_t, t.uint8_t),
            False,
        )
    }
    
    def handle_cluster_request(self, tsn, command_id, args):
        if args[0] == 1:
            self.endpoint.device.on_off_bus.listener_event("listener_event", ZHA_SEND_EVENT, COMMAND_ON, [])
        elif args[0] == 4:
            self.endpoint.device.on_off_bus.listener_event("listener_event", ZHA_SEND_EVENT, COMMAND_OFF, [])            

class SengledE1EG57F(CustomDevice):
    def __init__(self, *args, **kwargs):
        self.on_off_bus = Bus()
        
        super().__init__(*args, **kwargs)

    signature = {
        # <SimpleDescriptor endpoint=1 profile=260 device_type=260
        # device_version=0
        # input_clusters=[0, 1, 3, 32, 64529]
        # output_clusters=[3, 4, 6, 8, 64528]>
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.DIMMER_SWITCH,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    Identify.cluster_id,
                    PollControl.cluster_id,
                    0xfc11,
                ],
                OUTPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Groups.cluster_id,
                    OnOff.cluster_id,
                    LevelControl.cluster_id,
                    SengledRemoteCluster.cluster_id,
                ],
            },
        }
    }

    replacement = {
        ENDPOINTS: {
            1: {
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    Identify.cluster_id,
                    PollControl.cluster_id,
                    0xfc11,
                ],
                OUTPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Groups.cluster_id,
                    OnOffCluster,
                    LevelControl.cluster_id,
                    SengledRemoteCluster,
                ],
            },
        }
    }
```

To test this locally, I'm roughly following the directions here: https://github.com/zigpy/zha-device-handlers#testing-quirks-in-development-in-docker-based-install

The code above is a file I called `e1e-g7f.py`, and it lives alongside an empty file called `__init__.py`, which are both in a folder called `sengled`, which goes in `/usr/local/lib/python3.8/site-packages/zhaquirks/` in the homeassistant Docker container. The instructions say to copy the files there, but to make it easier, I created the `sengled` folder inside the Home Assistant config folder and then just symlinked `/usr/local/lib/python3.8/site-packages/zhaquirks/` to `/config/sengled` inside the Docker container.

Also, I found that deleting the `pycache` folder didn't seem to be necessary--I just restart Home Assistant after making changes.

blathers123, 2021-01-06:
@compeek awesome work. I'm glad this is finally getting some traction. I'll buy you a beer now and a 6 pack once the PR gets accepted if you'd like

compeek, 2021-01-06:
> Alex Hauck awesome work. I'm glad this is finally getting some traction. I'll buy you a beer now and a 6 pack once the PR gets accepted if you'd like

I appreciate it, but I'm just happy to get it working especially if other people can make use of it.

compeek, 2021-01-08:
All right, I think I got this thing fully working. I submitted a PR: https://github.com/zigpy/zha-device-handlers/pull/707

I see already there are a few things I need to fix to pass the automated checks, but once again I stayed up way too late, so I'll address those this weekend.

I don't know how long it will take for this to make it to Home Assistant once the PR is approved and merged, but hopefully we'll all be able to use this device soon!

blathers123, 2021-01-08:
Awesome, terrific work. Thank you!

NobleWolf, 2021-01-08:
> @compeek awesome work. I'm glad this is finally getting some traction. I'll buy you a beer now and a 6 pack once the PR gets accepted if you'd like

I will do the same if you setup a buymeacoffee.com or something.

compeek, 2021-01-08:
> > Alex Hauck awesome work. I'm glad this is finally getting some traction. I'll buy you a beer now and a 6 pack once the PR gets accepted if you'd like
> 
> I will do the same if you setup a buymeacoffee.com or something.

You guys are too kind. I really don't expect anything, but I set one up: https://www.buymeacoffee.com/compeek

compeek, 2021-01-08:
By the way, the PR is merged! Sounds like it will probably make it into the next release of Home Assistant: https://github.com/zigpy/zha-device-handlers/pull/707#issuecomment-757016275

blathers123, 2021-01-08:
Not sure of the beer to taco conversation rate but I took a stab at it. Awesome work @compeek

compeek, 2021-01-09:
> Not sure of the beer to taco conversation rate but I took a stab at it. Awesome work @compeek

Haha, thank you so much! Be sure to let me know how it works for you once you test it.

compeek, 2021-02-04:
@blathers123 @NobleWolf @doughecka The new device handler is finally showing up in Home Assistant for me now with the 2021.2.0 release! Let me know how it works for you.

NobleWolf, 2021-02-05:
@compeek yes it is working well now. How do you find the device handler version in the HA install? It wasn't in the release notes.

@blathers123 and @doughecka there are two blueprints available for this Sengled E1E-G7F device:
Control each button/option manually: https://community.home-assistant.io/t/zha-sengled-smart-light-switch-e1e-g7f-with-complete-control-of-every-button/276776/5
Easy "light switch" functions: https://community.home-assistant.io/t/zha-sengled-smart-light-switch-e1e-g7f/266441/15

NobleWolf, 2021-02-05:
> You guys are too kind. I really don't expect anything, but I set one up: https://www.buymeacoffee.com/compeek

It's hard work (and as you said, many late nights) that deserves recognition.

compeek, 2021-02-05:
> Alex Hauck yes it is working well now. How do you find the device handler version in the HA install? It wasn't in the release notes.

I'm not sure. :/ I also checked the release notes and didn't see anything. I've just been hoping the new changes would show up eventually. I think maybe they just forgot to put it in the release notes because 2021.2.1 that came out today seems to show version numbers for dependencies they upgraded.

compeek, 2021-02-05:
> It's hard work (and as you said, many late nights) that deserves recognition.

Thank you very much!

MattWestb, 2021-02-05:
And in dependencies  for [zha-device-handlers 0.0.52](https://github.com/zigpy/zha-device-handlers/releases/tag/0.0.52) is ??

doughecka, 2021-02-05:
> @compeek yes it is working well now. How do you find the device handler version in the HA install? It wasn't in the release notes.
>
> @blathers123 and @doughecka there are two blueprints available for [...]

The complete control blueprint works great!
I'm going to go ahead and close this issue, thanks for getting this working. Enjoy your coffee 👍

closed 2021-02-05 by doughecka

compeek, 2021-02-05:
> I'm going to go ahead and close this issue, thanks for getting this working. Enjoy your coffee 👍

You're welcome, and thank you!

compeek, 2021-02-05:
> And in dependencies for [zha-device-handlers 0.0.52](https://github.com/zigpy/zha-device-handlers/releases/tag/0.0.52) is ??

@MattWestb Sorry, I'm not sure what you're asking.

MattWestb, 2021-02-05:
>I'm not sure. :/ I also checked the release notes and didn't see anything. I've just been hoping the new changes would show up eventually. I think maybe they just forgot to put it in the release notes because 2021.2.1 that came out today seems to show version numbers for dependencies they upgraded.

And if you is reading the dependencies in the release notes you is finding the merger of the PR:
"Add quirk for Sengled E1E-G7F (#707) @compeek"

compeek, 2021-02-05:
> > I'm not sure. :/ I also checked the release notes and didn't see anything. I've just been hoping the new changes would show up eventually. I think maybe they just forgot to put it in the release notes because 2021.2.1 that came out today seems to show version numbers for dependencies they upgraded.
> 
> And if you is reading the dependencies in the release notes you is finding the merger of the PR:
> "Add quirk for Sengled E1E-G7F (#707) Alex Hauck"

Yes, but I think the question was: How do we know that Home Assistant is using the new version of zha-device-handlers? It wasn't mentioned in the Home Assistant release notes.

MattWestb, 2021-02-05:
Then you must learning reading the release notes and commitments that is refereed to them or using the try and error method.

Also you can reading the version if you is diving down in your docker container.

NobleWolf, 2021-02-05:
The zigpy/device handler version bump is normally in the HA release notes.