zigpy/zha-device-handlers #142: Aqara button lumi.sensor_switch.aq2 issues (wxkg11lm)

darkskiez, 2019-08-27:
I don't receive button events, just attribute changed events.

Double Clicks result in an unknown attribute id=32768 value=2 message

Single clicks send an on_off attribute change twice.

https://gist.github.com/darkskiez/34afff2942a2968c113b4c8192db18ad

jasonatepaint, 2019-08-31:
I am experiencing the same results on HA .97.2 for the same _wxkg11lm_ switch:

- 2 `zha_event` events being fired for on_off (1/0) with a single press
- Double-click shows unknown attribute w/value=2  

**More info:**
The device is registered and shows the `zhaquirks.xiaomi.aqara.switch_aq2.SwitchAQ2` quirk.  The battery is the only sensor registered, which appears to properly report battery status.

dmulcahey, 2019-09-03:
the only sensor you will get for this is the battery sensor. I'll take a look at the double press event soon.

Nemesis24, 2019-09-23:
for sample what i use in automation for single press, double press, triple press and quadruple press, all works fine.

`- id: '1551960941703'
  alias: Inter_table_1clic
  trigger:
  - event_data:
      device_ieee: 00:15:8d:00:02:13:32:1d
      attribute_name: on_off
      command: attribute_updated
      args:
        attribute_id: 0
        attribute_name: on_off
        value: 1
    event_type: zha_event
    platform: event
  condition: []
  action:
  - entity_id: switch.lumi_lumi_ctrl_ln2_aq1_02488196_2
    service: switch.toggle
- id: '1551960941704'
  alias: Inter_table_2clic
  trigger:
  - event_data:
      device_ieee: 00:15:8d:00:02:13:32:1d
      attribute_name: on_off
      command: attribute_updated
      args:
        attribute_id: 32768
        attribute_name: Unknown
        value: 2
    event_type: zha_event
    platform: event
  condition: []
  action:
  - entity_id: switch.lumi_lumi_ctrl_ln2_aq1_02488196_1
    service: switch.toggle
- id: '1551960941705'
  alias: Inter_table_3clic
  trigger:
  - event_data:
      device_ieee: 00:15:8d:00:02:13:32:1d
      attribute_name: on_off
      command: attribute_updated
      args:
        attribute_id: 32768
        attribute_name: Unknown
        value: 3
    event_type: zha_event
    platform: event
  condition: []
  action:
  - entity_id: switch.lumi_lumi_ctrl_ln2_aq1_023fc9b2_2
    service: switch.toggle
- id: '1551960941706'
  alias: Inter_table_4clic
  trigger:
  - event_data:
      device_ieee: 00:15:8d:00:02:13:32:1d
      attribute_name: on_off
      command: attribute_updated
      args:
        attribute_id: 32768
        attribute_name: Unknown
        value: 4
    event_type: zha_event
    platform: event
  condition: []
  action:
  - entity_id: switch.lumi_lumi_ctrl_ln2_aq1_023fc9b2_1
    service: switch.toggle`

dmulcahey, 2019-09-26:
This is much easier to use in the next release of HA. The events will be irrelevant. HA .100 includes device automations which make this much easier to use: 

[image]

closed 2019-09-26 by dmulcahey