zigpy/zha-device-handlers #144: Enable state reporting

abmantis, 2019-09-01:
How to enable state reporting in, for example, wall power plugs?
I have some that are working, but they don't report their state if I press the button on the plug, nor their current power (W) values.

Zigbee2mqtt has the following lines for that: 
```
(cb) => device.report('seMetering', 'instantaneousDemand', 10, 60, 1, cb),
(cb) => device.report('genOnOff', 'onOff', 1, 60, 1, cb),
```

Any way to do the same here?

abmantis, 2019-09-03:
Just to clarify, I'm asking as a dev, not as a user :) 
I'm ok with implementing the change if possible.

dmulcahey, 2019-09-03:
this is done by default in ZHA in HA. follow the code in async_device_initialized in gateway.py.

abmantis, 2019-09-04:
@dmulcahey I think that ZHA is doing what it should. When I press the Reconfigure button on HA, I get the following logs:
```
2019-09-04 21:43:15 DEBUG (MainThread) [homeassistant.components.zha.api] Reconfiguring node with ieee_address: <REDACTED>
2019-09-04 21:43:15 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xc5e1](ZB-SmartPlug-1.0.0): started configuration
2019-09-04 21:43:15 DEBUG (MainThread) [homeassistant.components.zha.core.channels] [0xc5e1:ZDO](ZB-SmartPlug-1.0.0): channel: 'async_configure' stage succeeded
2019-09-04 21:43:15 DEBUG (MainThread) [homeassistant.components.zha.core.channels] [0xc5e1:85:0x0000]: bound 'basic' cluster: Status.SUCCESS
2019-09-04 21:43:15 DEBUG (MainThread) [homeassistant.components.zha.core.channels] [0xc5e1:85:0x0000]: finished channel configuration
2019-09-04 21:43:15 DEBUG (MainThread) [homeassistant.components.zha.core.channels] [0xc5e1:85:0x0702]: bound 'smartenergy_metering' cluster: Status.SUCCESS
2019-09-04 21:43:15 DEBUG (MainThread) [homeassistant.components.zha.core.channels] [0xc5e1:85:0x0006]: bound 'on_off' cluster: Status.SUCCESS
2019-09-04 21:43:15 DEBUG (MainThread) [homeassistant.components.zha.core.channels] [0xc5e1:85:0x0000]: initializing channel: from_cache: False
2019-09-04 21:43:15 DEBUG (MainThread) [homeassistant.components.zha.core.channels] [0xc5e1:85:0x0000]: channel: 'async_configure' stage succeeded
2019-09-04 21:43:15 DEBUG (MainThread) [homeassistant.components.zha.core.channels] [0xc5e1:85:0x0702]: reporting 'instantaneous_demand' attr on 'smartenergy_metering' cluster: 30/900/1: Result: '[[<ConfigureReportingResponseRecord status=Status.SUCCESS>]]'
2019-09-04 21:43:15 DEBUG (MainThread) [homeassistant.components.zha.core.channels] [0xc5e1:85:0x0006]: reporting 'on_off' attr on 'on_off' cluster: 0/900/1: Result: '[[<ConfigureReportingResponseRecord status=Status.SUCCESS>]]'
2019-09-04 21:43:15 DEBUG (MainThread) [homeassistant.components.zha.core.channels] [0xc5e1:85:0x0702]: finished channel configuration
2019-09-04 21:43:15 DEBUG (MainThread) [homeassistant.components.zha.core.channels] [0xc5e1:85:0x0702]: channel: 'async_configure' stage succeeded
2019-09-04 21:43:16 DEBUG (MainThread) [homeassistant.components.zha.core.channels] [0xc5e1:85:0x0006]: finished channel configuration
2019-09-04 21:43:16 DEBUG (MainThread) [homeassistant.components.zha.core.channels] [0xc5e1:85:0x0006]: channel: 'async_configure' stage succeeded
2019-09-04 21:43:16 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xc5e1](ZB-SmartPlug-1.0.0): completed configuration
2019-09-04 21:43:16 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xc5e1](ZB-SmartPlug-1.0.0): stored in registry: ZhaDeviceEntry(name='EDP-WITHUS ZB-SmartPlug-1.0.0', ieee='<REDACTED>', last_seen=1567629795.7848678)
```

So it seems that the configuration for both the on/off and instantaneous demand is successful, but the state is not being updated anyway.
By the way, in Zigbee2mqtt the same device worked and reported both states.

dmulcahey, 2019-09-05:
that all looks good... could be a few things at play: the config params are different... 
we are using `'instantaneous_demand' attr on 'smartenergy_metering' cluster: 30/900/1`
z2m is using: `'seMetering', 'instantaneousDemand', 10, 60, 1`

the lower and upper time bounds are much higher in ZHA. you could try changing them here:
https://github.com/home-assistant/home-assistant/blob/6acfede512dcb2c0f41b5d591a3a2794e4299192/homeassistant/components/zha/core/channels/smartenergy.py#L80

you can see how to format the report dict here:
https://github.com/home-assistant/home-assistant/blob/6acfede512dcb2c0f41b5d591a3a2794e4299192/homeassistant/components/zha/core/const.py#L130

Try making them match z2m and reconfigure the device in the ZHA config panel to apply the changes after restarting HA. Lets start here and see where this leads us...

abmantis, 2019-09-16:
Fixed in https://github.com/zigpy/zigpy/pull/222

closed 2019-09-16 by abmantis