zigpy/zha-device-handlers #145: support Xiaomi WXKG11LM (lumi.remote.b1acn01)

ErnstEeldert, 2019-09-04:
I have the Xiaomi WXKG11LM (lumi.remote.b1acn01) single button remote from Xiaomi. I would like to add support for this, so I checked with the contributing guidelines. Here's the info shown after pairing the device:

`home-assistant   | 2019-09-01 16:18:45 INFO (MainThread) [zigpy.endpoint] [0x4d6e:1] Discovered endpoint information: <SimpleDescriptor endpoint=1 profile=260 device_type=259 device_version=1 input_clusters=[0, 18, 3] output_clusters=[0]>`

From what I see in other implementations, it looks to be closer to the remote.b186acn01 device than to the WXKG11LM sensor_switch.aq2 variety.

So I guess my question is, how would I go about mapping the various in and output clusters/attributes, and how would I setup my dev environment in such a way it's easy to debug?

Thanks!

dmulcahey, 2019-09-05:
Do you have a link to where you purchased this? There seem to be several versions of this switch. I want to make sure I answer this correctly.

ErnstEeldert, 2019-09-05:
This is the one: https://www.aliexpress.com/item/32920557799.html?spm=a2g0s.9042311.0.0.31c04c4dZhyOTi

Apparently, there are 3 different versions of this particular button (info from [deconz docs](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Supported-Devices#supported-wireless-switches)), I have the 2018 version.

ErnstEeldert, 2019-09-17:
Any guidance on steps to get this supported would be very helpful!

jayjaytyper, 2019-09-20:
I have the same issue with this device i can pair is and it shows an unknow state, and any of the button press 1,2 or long dose not give an event in HA but it dose in the pair log.

[image]

[image]

If you need any more logs or information to help add support for this please let me know i will be happy to help.

Cheers

ErnstEeldert, 2019-09-25:
I give it another shot using the dev branch today, and while it pairs, no zha_events show up, and the ZHA device shows up as unavailable:

`home-assistant   | 2019-09-25 09:13:23 INFO (MainThread) [zigpy_deconz.zigbee.application] New device joined: 0xace0, 00:15:8d:00:01:ef:56:d1
home-assistant   | 2019-09-25 09:13:23 INFO (MainThread) [zigpy.application] Device 0xace0 (00:15:8d:00:01:ef:56:d1) joined the network
home-assistant   | 2019-09-25 09:13:23 INFO (MainThread) [zigpy.device] [0xace0] Requesting 'Node Descriptor'
home-assistant   | Config directory: /config
home-assistant   | Tries remaining: 2
home-assistant   | Tries remaining: 1
home-assistant   | 2019-09-25 09:13:56 WARNING (MainThread) [zigpy_deconz.zigbee.application] Error while sending frame: 0xe9
home-assistant   | 2019-09-25 09:13:56 WARNING (MainThread) [zigpy.device] [0xace0] Requesting Node Descriptor failed: [0xace0:0:0x0002] failed transmission request: 233
home-assistant   | 2019-09-25 09:13:56 INFO (MainThread) [zigpy.device] [0xace0] Discovering endpoints
home-assistant   | Tries remaining: 3
home-assistant   | 2019-09-25 09:13:59 WARNING (MainThread) [zigpy_deconz.zigbee.application] Error while sending frame: 0xe9
home-assistant   | Tries remaining: 2
home-assistant   | 2019-09-25 09:14:02 INFO (MainThread) [zigpy.device] [0xace0] Discovered endpoints: [1]
home-assistant   | 2019-09-25 09:14:02 INFO (MainThread) [zigpy.endpoint] [0xace0:1] Discovering endpoint information
home-assistant   | Tries remaining: 3
home-assistant   | 2019-09-25 09:14:03 ERROR (MainThread) [homeassistant.core] Error doing job: Exception in callback SerialTransport._read_ready()
home-assistant   | Traceback (most recent call last):
home-assistant   |   File "/usr/local/lib/python3.7/asyncio/events.py", line 88, in _run
home-assistant   |     self._context.run(self._callback, *self._args)
home-assistant   |   File "/usr/local/lib/python3.7/site-packages/serial_asyncio/__init__.py", line 106, in _read_ready
home-assistant   |     self._protocol.data_received(data)
home-assistant   |   File "/usr/local/lib/python3.7/site-packages/zigpy_deconz/uart.py", line 63, in data_received
home-assistant   |     self._api.data_received(frame)
home-assistant   |   File "/usr/local/lib/python3.7/site-packages/zigpy_deconz/api.py", line 194, in data_received
home-assistant   |     getattr(self, '_handle_%s' % (command.name, ))(data)
home-assistant   |   File "/usr/local/lib/python3.7/site-packages/zigpy_deconz/api.py", line 255, in _handle_aps_data_indication
home-assistant   |     data[16])   # rssi
home-assistant   |   File "/usr/local/lib/python3.7/site-packages/zigpy_deconz/zigbee/application.py", line 193, in handle_rx
home-assistant   |     tsn, command_id, is_reply, args = self.deserialize(device, src_ep, cluster_id, data)
home-assistant   |   File "/usr/local/lib/python3.7/site-packages/zigpy/application.py", line 92, in deserialize
home-assistant   |     return sender.deserialize(endpoint_id, cluster_id, data)
home-assistant   |   File "/usr/local/lib/python3.7/site-packages/zigpy/device.py", line 157, in deserialize
home-assistant   |     return self.endpoints[endpoint_id].deserialize(cluster_id, data)
home-assistant   |   File "/usr/local/lib/python3.7/site-packages/zigpy/endpoint.py", line 170, in deserialize
home-assistant   |     raise KeyError("No cluster ID 0x%04x on %s" % (cluster_id, self.unique_id))
home-assistant   | KeyError: 'No cluster ID 0x0000 on (00:15:8d:00:01:ef:56:d1, 1)'
home-assistant   | 2019-09-25 09:14:04 INFO (MainThread) [zigpy.endpoint] [0xace0:1] Discovered endpoint information: <SimpleDescriptor endpoint=1 profile=260 device_type=259 device_version=1 input_clusters=[0, 18, 3] output_clusters=[0]>
home-assistant   | 2019-09-25 09:14:05 INFO (MainThread) [zigpy.device] [0xace0] Requesting 'Node Descriptor'
home-assistant   | Tries remaining: 2
home-assistant   | 2019-09-25 09:14:07 INFO (MainThread) [zigpy.device] [0xace0] Node Descriptor: <Optional byte1=2 byte2=64 mac_capability_flags=128 manufacturer_code=4151 maximum_buffer_size=127 maximum_incoming_transfer_size=100 server_mask=0 maximum_outgoing_transfer_size=100 descriptor_capability_field=0>
home-assistant   | 2019-09-25 09:14:07 WARNING (MainThread) [homeassistant.components.zha.core.discovery] there are no channels in the discovery info: {'unique_id': '00:15:8d:00:01:ef:56:d1-1', 'zha_device': <homeassistant.components.zha.core.device.ZHADevice object at 0xae1c4b50>, 'channels': [], 'component': 'switch'}
home-assistant   | 2019-09-25 09:14:07 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.zha entity: sensor.lumi_lumi_remote_b1acn01_01ef56d1_1_18
home-assistant   | 2019-09-25 09:14:09 WARNING (MainThread) [zigpy_deconz.zigbee.application] Error while sending frame: 0xe9`

[image]

dmulcahey, 2019-09-25:
Then it isn’t paired

dmulcahey, 2019-09-25:
Just like your other ticket, you aren't actually getting these to fully pair. Are you following the best practices to pair Xiaomi devices: https://community.smartthings.com/t/original-aqara-xiaomi-zigbee-sensors-contact-temp-motion-button-outlet-leak-etc/113253

Also, I have updated both tickets. When posting logs please place triple back ticks before and after the block. Single back ticks are meant for single lines.

ErnstEeldert, 2019-09-25:
Apologies for the messy logs. I changed my pairing method a bit, pressing the 'reset' button a little less, and that seems to work for the b186 switch.

For this little button however, I still can't get it to work. I managed to get the pairing to complete (I think), as the logs say "[0xe544](lumi.remote.b1acn01): completed initialization". I also see the line with the SimpleDescriptor being discovered.

What is interesting though is that no quirk is shown in the screen, so to me it looks like the quirk detection is somehow not working as expected. Here's the pairing full log:

```
Device 0xe544 (00:15:8d:00:01:ef:56:d1) joined the network
[0xe544:zdo] ZDO request 0x0013: [0xe544, 00:15:8d:00:01:ef:56:d1, 128]
[0xe544] Requesting 'Node Descriptor'
[0xe544] Node Descriptor: <Optional byte1=2 byte2=64 mac_capability_flags=128 manufacturer_code=4151 maximum_buffer_size=127 maximum_incoming_transfer_size=100 server_mask=0 maximum_outgoing_transfer_size=100 descriptor_capability_field=0>
Error calling listener.node_descriptor_updated: FOREIGN KEY constraint failed
[0xe544] Discovering endpoints
[0xe544] Discovered endpoints: [1]
[0xe544:1] Discovering endpoint information
[0xe544:1] Discovered endpoint information: <SimpleDescriptor endpoint=1 profile=260 device_type=259 device_version=1 input_clusters=[0, 18, 3] output_clusters=[0]>
[0xe544:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=99 command_id=Command.Read_Attributes_rsp>
[0xe544:1] Manufacturer: LUMI
[0xe544:1] Model: lumi.remote.b1acn01
Checking quirks for LUMI lumi.remote.b1acn01 (00:15:8d:00:01:ef:56:d1)
Considering <class 'zigpy.quirks.smartthings.SmartthingsMultiPurposeSensor'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
Fail because endpoint list mismatch: {11, 13} {1}
Considering <class 'zhaquirks.lutron.lzl4bwhl01remote.LutronLZL4BWHL01Remote2'>
Fail because profile_id mismatch on at least one endpoint
Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.osram.a19twhite.A19TunableWhite'>
Fail because endpoint list mismatch: {3} {1}
Considering <class 'zhaquirks.philips.rwl021.PhilipsRWL021'>
Fail because endpoint list mismatch: {1, 2} {1}
Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.xbee.xbee_io.XbeeSensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.xiaomi.mija.smoke.MijiaHoneywellSmokeDetectorSensor'>
Fail because device_type mismatch on at least one endpoint
device - 0xe544:00:15:8d:00:01:ef:56:d1 entering async_device_initialized - is_new_join: True
device - 0xe544:00:15:8d:00:01:ef:56:d1 has joined the ZHA zigbee network
[0xe544](lumi.remote.b1acn01): started configuration
[0xe544:ZDO](lumi.remote.b1acn01): channel: 'async_configure' stage succeeded
[0xe544:1:0x0000]: finished channel configuration
[0xe544:1:0x0012]: finished channel configuration
[0xe544:1:0x0012]: channel: 'async_configure' stage succeeded
[0xe544:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=100 command_id=Command.Read_Attributes_rsp>
[0xe544:1:0x0000]: initializing channel: from_cache: False
[0xe544:1:0x0000]: channel: 'async_configure' stage succeeded
[0xe544](lumi.remote.b1acn01): completed configuration
[0xe544](lumi.remote.b1acn01): stored in registry: ZhaDeviceEntry(name='LUMI lumi.remote.b1acn01', ieee='00:15:8d:00:01:ef:56:d1', last_seen=1569437545.276871)
there are no channels in the discovery info: {'unique_id': '00:15:8d:00:01:ef:56:d1-1', 'zha_device': <homeassistant.components.zha.core.device.ZHADevice object at 0xadb7d6d0>, 'channels': [], 'component': 'switch'}
[0xe544](lumi.remote.b1acn01): started initialization
[0xe544:ZDO](lumi.remote.b1acn01): entry loaded from storage: ZhaDeviceEntry(name='LUMI lumi.remote.b1acn01', ieee='00:15:8d:00:01:ef:56:d1', last_seen=1569437545.276871)
[0xe544:ZDO](lumi.remote.b1acn01): channel: 'async_initialize' stage succeeded
[0xe544:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=True is_reply=True disable_default_response=True> manufacturer=4447 tsn=0 command_id=Command.Report_Attributes>
[0xe544:1:0x0000] ZCL request 0x000a: [[<Attribute attrid=5 value=<TypeValue type=CharacterString, value=lumi.remote.b1acn01>>, <Attribute attrid=65281 value=<TypeValue type=CharacterString, value=!�(!�!l>>]]
[0xe544:1:0x0000] Attribute report received: model=lumi.remote.b1acn01, 65281=!�(!�!l
[0xe544:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=101 command_id=Command.Read_Attributes_rsp>
[0xe544:1:0x0000]: initializing channel: from_cache: False
[0xe544:1:0x0000]: channel: 'async_initialize' stage succeeded
[... cut ...]
[0xe544:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=True is_reply=True disable_default_response=True> manufacturer=4447 tsn=1 command_id=Command.Report_Attributes>
[0xe544:1:0x0000] ZCL request 0x000a: [[<Attribute attrid=5 value=<TypeValue type=CharacterString, value=lumi.remote.b1acn01>>, <Attribute attrid=65281 value=<TypeValue type=CharacterString, value=!�(!�!l>>]]
[0xe544:1:0x0000] Attribute report received: model=lumi.remote.b1acn01, 65281=!�(!�!l
```

dmulcahey, 2019-09-25:
I just added support for this to dev. Are you sure you’re using the newest version? It’s not in a released HA version yet.

ErnstEeldert, 2019-09-25:
Hmm. So I installed the dev branch into my homeassistant docker container, like this:

```
$ docker container exec home-assistant pip3 install git+https://github.com/dmulcahey/zha-device-handlers@dev
Collecting git+https://github.com/dmulcahey/zha-device-handlers@dev
  Cloning https://github.com/dmulcahey/zha-device-handlers (to revision dev) to /tmp/pip-req-build-mvkvtytj
  Running command git clone -q https://github.com/dmulcahey/zha-device-handlers /tmp/pip-req-build-mvkvtytj
Requirement already satisfied: zigpy-homeassistant>=0.8.0 in /usr/local/lib/python3.7/site-packages (from zha-quirks==0.0.25) (0.8.0)
Requirement already satisfied: aiohttp in /usr/local/lib/python3.7/site-packages (from zigpy-homeassistant>=0.8.0->zha-quirks==0.0.25) (3.6.1)
Requirement already satisfied: pycryptodome in /usr/local/lib/python3.7/site-packages (from zigpy-homeassistant>=0.8.0->zha-quirks==0.0.25) (3.9.0)
Requirement already satisfied: crccheck in /usr/local/lib/python3.7/site-packages (from zigpy-homeassistant>=0.8.0->zha-quirks==0.0.25) (0.6)
Requirement already satisfied: multidict<5.0,>=4.5 in /usr/local/lib/python3.7/site-packages (from aiohttp->zigpy-homeassistant>=0.8.0->zha-quirks==0.0.25) (4.5.2)
Requirement already satisfied: chardet<4.0,>=2.0 in /usr/local/lib/python3.7/site-packages (from aiohttp->zigpy-homeassistant>=0.8.0->zha-quirks==0.0.25) (3.0.4)
Requirement already satisfied: yarl<2.0,>=1.0 in /usr/local/lib/python3.7/site-packages (from aiohttp->zigpy-homeassistant>=0.8.0->zha-quirks==0.0.25) (1.3.0)
Requirement already satisfied: async-timeout<4.0,>=3.0 in /usr/local/lib/python3.7/site-packages (from aiohttp->zigpy-homeassistant>=0.8.0->zha-quirks==0.0.25) (3.0.1)
Requirement already satisfied: attrs>=17.3.0 in /usr/local/lib/python3.7/site-packages (from aiohttp->zigpy-homeassistant>=0.8.0->zha-quirks==0.0.25) (19.1.0)
Requirement already satisfied: idna>=2.0 in /usr/local/lib/python3.7/site-packages (from yarl<2.0,>=1.0->aiohttp->zigpy-homeassistant>=0.8.0->zha-quirks==0.0.25) (2.8)
Building wheels for collected packages: zha-quirks
  Building wheel for zha-quirks (setup.py): started
  Building wheel for zha-quirks (setup.py): finished with status 'done'
  Created wheel for zha-quirks: filename=zha_quirks-0.0.25-cp37-none-any.whl size=85771 sha256=003c74fffdad059398403768e68b05ae9a86cb0c46a280177036c1fa163a362e
  Stored in directory: /tmp/pip-ephem-wheel-cache-3feaq9fq/wheels/91/9a/24/fa1b1f6cf2c7a32c755e7439c3badbf6fb8545f2b5cab11f45
Successfully built zha-quirks
Installing collected packages: zha-quirks
  Found existing installation: zha-quirks 0.0.23
    Uninstalling zha-quirks-0.0.23:
      Successfully uninstalled zha-quirks-0.0.23
Successfully installed zha-quirks-0.0.25
```

However, it turns out that this needs HA to be restarted to picked up. Which in turn somehow reverts zha-quirks to 0.0.23. Any ideas on how to prevent this from happening?

dmulcahey, 2019-09-25:
You can update ha’s requirements file to specify the version you want.

ErnstEeldert, 2019-09-25:
So I managed to get it to install 0.0.25 from pypi, as I'm not sure how to do installs straight from git. 
To do so, I modified:

- /usr/src/homeassistant/homeassistant/components/zha/manifest.json 
- /usr/src/homeassistant/requirements_all.txt

I can see the updated code being installed in the site-packages folder. However, still no dice with the pairing. I noticed from the Simple Descriptor that my button has input_clusters=[0, 18, 3] output_clusters=[0], but the sensor_switch_aq3.py has input_clusters=[0, 18, 6, 1] output_clusters=[0]

dmulcahey, 2019-09-25:
Good catch... gonna have to add another handler for this. I’ll try to get to it for .100

ErnstEeldert, 2019-09-26:
I can take a stab at implementing the handler for it? I think it looks more similar to the b186acn01 switch, so I can take that as a guideline.

dmulcahey, 2019-09-26:
Please try #164

ErnstEeldert, 2019-09-26:
Thanks for this! Turns out there's a small copy/paste issue with the code, the device type is incorrect. See https://github.com/dmulcahey/zha-device-handlers/pull/164/files#r328808179.

With that fixed, I managed to pair it with the quirk applied. Button events work as expected now. Battery status is still unknown, but I believe it can take up to a few hours before the switch reports this. So will let you know tomorrow.

dmulcahey, 2019-09-26:
better now??

closed 2019-09-26 by dmulcahey

ErnstEeldert, 2019-09-28:
Tested OK!

PR #164 (2019-09-26): Add alternate version of WXKG11LM
Fixes #145