zigpy/zha-device-handlers #146: OSRAM 4x-Lightify Switch command timeout

paulbdavis, 2019-09-07:
I have the [OSRAM 4x-Lightify Switch](https://www.amazon.com/gp/product/B01M3OK6J7) and I have gotten it connected via ZHA but it is not allowing me to query it's attributes nor perform bindings

[image]

When I try either, I get the following in the HA logs

```
zigpy.exceptions.DeliveryError: [0xf072:1:0x0006]: message send failure: EmberStatus.DELIVERY_FAILED
2019-09-07 13:27:16 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection.139623448770064] Error handling message: Unknown error
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/decorators.py", line 17, in _handle_async_response
    await func(hass, connection, msg)
  File "/usr/src/homeassistant/homeassistant/components/zha/api.py", line 355, in websocket_read_zigbee_cluster_attributes
    [attribute], allow_cache=False, only_cache=False, manufacturer=manufacturer
  File "/usr/local/lib/python3.7/site-packages/zigpy/zcl/__init__.py", line 210, in read_attributes
    result = await self.read_attributes_raw(to_read, manufacturer=manufacturer)
  File "/usr/local/lib/python3.7/site-packages/zigpy/zcl/__init__.py", line 178, in read_attributes_raw
    v = await self.request(True, 0x00, schema, attributes, manufacturer=manufacturer)
  File "/usr/local/lib/python3.7/site-packages/zigpy/device.py", line 147, in request
    expect_reply=expect_reply,
  File "/usr/local/lib/python3.7/site-packages/bellows/zigbee/application.py", line 368, in request
    res = await asyncio.wait_for(req.send, timeout=APS_ACK_TIMEOUT)
  File "/usr/local/lib/python3.7/asyncio/tasks.py", line 442, in wait_for
    return fut.result()
zigpy.exceptions.DeliveryError: [0xf072:1:0x0000]: message send failure: EmberStatus.DELIVERY_FAILED
```

```
2019-09-07 13:28:16 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection.139623448770064] Error handling message: Unknown error
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/decorators.py", line 17, in _handle_async_response
    await func(hass, connection, msg)
  File "/usr/src/homeassistant/homeassistant/components/zha/api.py", line 410, in websocket_bind_devices
    await async_binding_operation(zha_gateway, source_ieee, target_ieee, BIND_REQUEST)
  File "/usr/src/homeassistant/homeassistant/components/zha/api.py", line 473, in async_binding_operation
    await asyncio.gather(*bind_tasks)
  File "/usr/local/lib/python3.7/site-packages/zigpy/device.py", line 147, in request
    expect_reply=expect_reply,
  File "/usr/local/lib/python3.7/site-packages/bellows/zigbee/application.py", line 368, in request
    res = await asyncio.wait_for(req.send, timeout=APS_ACK_TIMEOUT)
  File "/usr/local/lib/python3.7/asyncio/tasks.py", line 442, in wait_for
    return fut.result()
zigpy.exceptions.DeliveryError: [0xf072:0:0x0021]: message send failure: EmberStatus.DELIVERY_FAILED
```

MithrasPan, 2019-09-10:
I'd like to see these working as well... I'll be happy to give any info that I can.

https://community.home-assistant.io/t/zha-zigbee-tested-devices-please-add-your-device-results/17718/1214?u=mithraspan

dmulcahey, 2019-09-10:
you have to wake the remote up to perform either action. battery powered devices sleep often to preserve battery life. press a button on the remote to wake it up while performing the actions. 

in the next version or 2 there will be support for 8 events from this remote.

MithrasPan, 2019-09-11:
Well... I mean... I was kinda spamming the console with events. ...

Saw the commit, look forward to seeing what effect it has once it trickles down to my install. thanks for the time and effort :)

paulbdavis, 2019-09-16:
Looks like #149 added some stuff for this. How can I try out the master branch of this repo with the current version of HA? (running in docker, but I'm comfortable getting in there and changing things)

Adminiuga, 2019-09-16:
you could try installing pip packages as zip file from github, taking same approach as in https://github.com/home-assistant/home-assistant/issues/26281 Make sure to have snapshots, as it is quite easy to wreak havoc.

dmulcahey, 2020-03-06:
this device is working in the latest HA. if you are still having issues please open a new issue.

closed 2020-03-06 by dmulcahey