zigpy/zha-device-handlers #170: SmartThings moisturev4 sensor wrong device class

presslab-us, 2019-09-28:
The SmartThings moisturev4 sensor has the wrong device class.  It shows up as a generic on/off device, and not as a moisture sensor.

This is because the `zone_type` attribute returns `0` instead of `0x2A` as it should.

It is already detected with the `zhaquirks.smartthings.iasv4.SmartThingsIASV4` quirk, which is shared with the "motionv4" device.  It isn't that hard to replace that `zone_type` attribute.  But it seems like the "motionv4" and "moisturev4" should probably be split into two different files/classes.

Adminiuga, 2019-09-28:
Alternatively you could customize device class through config panel -> customization -> pick device, select device class and override it.

dmulcahey, 2019-09-29:
I'd rather not duplicate the entire quirk and add a custom cluster for something that is easily overridden in HA. As @Adminiuga stated you can override the device class in HA.

presslab-us, 2019-09-29:
Interesting, not the response I expected, but I understand.

dmulcahey, 2019-09-29:
I may revisit this shortly. It just isn’t high up on the priority list since there is a work around.

Adminiuga, 2019-09-29:
I was thinking about this at some point and while yeah, we could create a quirks whick takes the device type, but:
1. If more devices with wrong device class pop up, how do you stay up to date? 
2. In case of an open/close sensor HA has device classes for doors, garage doors etc while Zigbee doesn't (I could be wrong) so overriding from HA seemed to be a better option to me

presslab-us, 2019-09-29:
I understand what both you are saying.  I expected that I would implement this new quirk, but I wanted to create an issue to discuss it first.  I have seen (and experienced) writing a bunch of code only for it to be quickly rejected at the pull request.

To me it seems like it would make it easier for people to migrate to Home Assistant (especially from SmartThings, using these sensors) if things worked out of the box.  Also it just seemed like the "correct" way to fix the issue, short of changing the firmware in the sensor.

I think making the default "correct" (in this case moisture sensor is an obvious choice) and of course still allowing for the user to change the class for some ambiguous device seems like it would be the best of all.

Adminiuga, 2019-09-29:
PRs are always welcome :)

dmulcahey, 2019-09-29:
If you want to implement it, go for it 😁

PR #172 (2019-10-01): Modify SmartThings moisturev4 handler to identify as a moisture sensor
This required splitting the existing SmartThings IAS handler into two.

Closes #170
closed 2019-10-01 by dmulcahey