zigpy/zha-device-handlers #173: add quadruple clicked  to WXKG11LM

Nemesis24, 2019-10-07:
[image]

dmulcahey, 2020-03-06:
this is a software trick on other platforms. The device does not actually emit this event.

Nemesis24, 2020-04-23:
```
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "00:15:8d:00:02:13:0d:2d",
        "unique_id": "00:15:8d:00:02:13:0d:2d:1:0x0006",
        "endpoint_id": 1,
        "cluster_id": 6,
        "command": "attribute_updated",
        "args": {
            "attribute_id": 32768,
            "attribute_name": "Unknown",
            "value": 4
        }
```

Nemesis24, 2020-04-23:
I don't but mine yes !!

Nemesis24, 2020-04-23:
WXKG11LM

closed 2020-09-22 by Nemesis24