zigpy/zha-device-handlers #176: Sylvania/Osram Lightify Dimmer Switch battery always at 100%

rpatterson, 2019-10-15:
I have several Sylvania/Osram Lightify Dimmer Switches that I moved over from a SmartThings hub v2 to Home Assistant Docker v0.99.3 with a HUSBZB-1 USB stick.  Under ST, I could see battery percentages but under HA all their batteries show 100% all the time.  I've tried removing and repairing some of them but they still show the same 100% battery level after re-pairing.  It's recognized as using the `zhaquirks.centralite.3130.CentraLite3130` quirk.

Here's a log snipped from a pairing session if it helps:

```
2019-10-15 00:23:11 DEBUG (MainThread) [homeassistant.components.zha.api] Reconfiguring node with ieee_address: 84:18:26:00:00:e9:05:a4
2019-10-15 00:23:11 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xfc69](LIGHTIFY Dimming Switch): started configuration
2019-10-15 00:23:11 DEBUG (MainThread) [homeassistant.components.zha.core.channels] [0xfc69:ZDO](LIGHTIFY Dimming Switch): channel: 'async_configure' stage succeeded
2019-10-15 00:23:11 DEBUG (MainThread) [bellows.zigbee.application] Extending timeout for 84:18:26:00:00:e9:05:a4/0xfc69
2019-10-15 00:23:11 DEBUG (MainThread) [bellows.ezsp] Send command setExtendedTimeout: (84:18:26:00:00:e9:05:a4, True)
2019-10-15 00:23:11 DEBUG (MainThread) [bellows.zigbee.application] Extending timeout for 84:18:26:00:00:e9:05:a4/0xfc69
2019-10-15 00:23:11 DEBUG (MainThread) [bellows.ezsp] Send command setExtendedTimeout: (84:18:26:00:00:e9:05:a4, True)
2019-10-15 00:23:11 DEBUG (MainThread) [bellows.zigbee.application] Extending timeout for 84:18:26:00:00:e9:05:a4/0xfc69
2019-10-15 00:23:11 DEBUG (MainThread) [bellows.ezsp] Send command setExtendedTimeout: (84:18:26:00:00:e9:05:a4, True)
2019-10-15 00:23:11 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'772d21d6f02ffcb259b252a1abfbde7e'
2019-10-15 00:23:11 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'702da1d687d67e'
2019-10-15 00:23:11 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8070787e'
2019-10-15 00:23:11 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'003221d6f02ffcb259b252a1ab74697e'
2019-10-15 00:23:11 DEBUG (MainThread) [bellows.ezsp] Application frame 126 (setExtendedTimeout) received: b''
2019-10-15 00:23:11 DEBUG (MainThread) [bellows.ezsp] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0xfc69, <EmberApsFrame profileId=0 clusterId=33 sourceEndpoint=0 destinationEndpoint=0 options=320 groupId=0 sequence=112>, 112, b"p\xa4\x05\xe9\x00\x00&\x18\x84\x01\x00\x00\x03\xb5'^\x0b\x00o\r\x00\x01")
2019-10-15 00:23:11 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'0132a1d6dc457e'
2019-10-15 00:23:11 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8160597e'
2019-10-15 00:23:11 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'7d313321d6f02ffcb259b252a1ab24cb7e'
2019-10-15 00:23:11 DEBUG (MainThread) [bellows.ezsp] Application frame 126 (setExtendedTimeout) received: b''
2019-10-15 00:23:11 DEBUG (MainThread) [bellows.ezsp] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0xfc69, <EmberApsFrame profileId=0 clusterId=33 sourceEndpoint=0 destinationEndpoint=0 options=320 groupId=0 sequence=113>, 113, b"q\xa4\x05\xe9\x00\x00&\x18\x84\x01\x01\x00\x03\xb5'^\x0b\x00o\r\x00\x01")
2019-10-15 00:23:11 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'1233a1d66b0e7e'
2019-10-15 00:23:11 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'82503a7e'
2019-10-15 00:23:11 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'2230219c5443e9b259b54a25aa1593499c3e57bd9d6a6262fdc64591787f3fa7e878f93184ffa8d6d5d3d4977e'
2019-10-15 00:23:11 DEBUG (MainThread) [bellows.ezsp] Application frame 126 (setExtendedTimeout) received: b''
2019-10-15 00:23:11 DEBUG (MainThread) [bellows.ezsp] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0xfc69, <EmberApsFrame profileId=0 clusterId=33 sourceEndpoint=0 destinationEndpoint=0 options=320 groupId=0 sequence=114>, 114, b"r\xa4\x05\xe9\x00\x00&\x18\x84\x01\x06\x00\x03\xb5'^\x0b\x00o\r\x00\x01")
2019-10-15 00:23:11 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'2330a19c54fbf4837e'
2019-10-15 00:23:12 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'83401b7e'
2019-10-15 00:23:12 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'3331219c5443e9b259b54a25aa1593499c3f56bd9c6a6262fdc64591787f3ea7e878f93184ffa8d6d5d37d3a027e'
2019-10-15 00:23:12 DEBUG (MainThread) [bellows.ezsp] Application frame 52 (sendUnicast) received: b'00d1'
2019-10-15 00:23:12 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'3431a19c54f8bc747e'
2019-10-15 00:23:12 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8430fc7e'
2019-10-15 00:23:12 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'4436219c5443e9b259b54a25aa1593499c3c55bd9f6a6262fdc64591787f39a7e878f93184ffa8d6d5d32e177e'
2019-10-15 00:23:12 DEBUG (MainThread) [bellows.ezsp] Application frame 52 (sendUnicast) received: b'00d2'
2019-10-15 00:23:12 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'4536a19c54f9cbbd7e'
2019-10-15 00:23:12 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8520dd7e'
2019-10-15 00:23:12 DEBUG (MainThread) [bellows.ezsp] Application frame 52 (sendUnicast) received: b'00d3'
2019-10-15 00:23:15 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'5536b1ed542a1593d9944a65ab55926229e14e57123165fbfda0e47e'
2019-10-15 00:23:15 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8610be7e'
2019-10-15 00:23:15 DEBUG (MainThread) [bellows.ezsp] Application frame 69 (incomingMessageHandler) received: b'00000021800000400100002bb5af69fcffff027000'
2019-10-15 00:23:15 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, <EmberApsFrame profileId=0 clusterId=32801 sourceEndpoint=0 destinationEndpoint=0 options=320 groupId=0 sequence=43>, 181, -81, 0xfc69, 255, 255, b'p\x00']
2019-10-15 00:23:15 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'6536b1975443e9b259b54a25aa1593499c9f57abed94477e'
2019-10-15 00:23:15 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'87009f7e'
2019-10-15 00:23:15 DEBUG (MainThread) [bellows.ezsp] Application frame 63 (messageSentHandler) received: b'0069fc00002100000040010000d1700000'
2019-10-15 00:23:15 DEBUG (MainThread) [bellows.zigbee.application] Received messageSentHandler frame with [<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 64617, <EmberApsFrame profileId=0 clusterId=33 sourceEndpoint=0 destinationEndpoint=0 options=320 groupId=0 sequence=209>, 112, <EmberStatus.SUCCESS: 0>, b'']
2019-10-15 00:23:15 DEBUG (MainThread) [homeassistant.components.zha.core.channels] [0x52b3:1:0x0000]: bound 'basic' cluster: Status.SUCCESS
2019-10-15 00:23:15 DEBUG (MainThread) [homeassistant.components.zha.core.channels] [0x52b3:1:0x0000]: finished channel configuration
2019-10-15 00:23:15 DEBUG (MainThread) [bellows.zigbee.application] Extending timeout for 84:18:26:00:00:e9:05:a4/0xfc69
2019-10-15 00:23:15 DEBUG (MainThread) [bellows.ezsp] Send command setExtendedTimeout: (84:18:26:00:00:e9:05:a4, True)
2019-10-15 00:23:15 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'573721d6f02ffcb259b252a1abfca47e'
2019-10-15 00:23:15 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'7637a1d624ed7e'
2019-10-15 00:23:15 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8070787e'
2019-10-15 00:23:15 DEBUG (MainThread) [bellows.ezsp] Application frame 126 (setExtendedTimeout) received: b''
2019-10-15 00:23:15 DEBUG (MainThread) [bellows.ezsp] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0xfc69, <EmberApsFrame profileId=260 clusterId=0 sourceEndpoint=1 destinationEndpoint=1 options=320 groupId=0 sequence=115>, 115, b'\x00s\x00\x07\x00')
2019-10-15 00:23:15 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'6034219c5443e9b658944a24ab1593499c3d54aeedbd678cfde66c7e'
2019-10-15 00:23:15 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'0637b1ed542a1593d9944a65ab5592653efe4e57123165fafd04077e'
2019-10-15 00:23:15 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8160597e'
2019-10-15 00:23:15 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'1734a19c54fe040d7e'
2019-10-15 00:23:15 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'82503a7e'
2019-10-15 00:23:15 DEBUG (MainThread) [bellows.ezsp] Application frame 69 (incomingMessageHandler) received: b'00000021800000400100002ca2b069fcffff027100'
[... cut ...]
2019-10-15 00:23:18 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'543f219c5416aab658904124ab1593499c3950aeedb96780f8eef87e'
2019-10-15 00:23:18 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'463fa19c54f21c4a7e'
2019-10-15 00:23:18 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8520dd7e'
2019-10-15 00:23:18 DEBUG (MainThread) [bellows.ezsp] Application frame 52 (sendUnicast) received: b'00d8'
```

Note the `status=Status.UNSUPPORTED_ATTRIBUTE` after `battery_percentage_remaining`, might that be related?  I also have this same issue with a Aqara/Xaomi door sensor.

Adminiuga, 2019-10-15:
Those devices do not support `battery_percentage_remaining` attribute so the battery level is estimated based on the battery voltage. What battery voltage your device is reporting?

rpatterson, 2019-10-15:
> What battery voltage your device is reporting?

@Adminiuga All I see anywhere in the UI is or configuration screens is:

```
LR Wall Switch Battery
10 hours ago
100 %
```
and:

```
battery quantity
1
```
Also, note the line in the log snipped in the description:

```
2019-10-15 00:23:16 DEBUG (MainThread) [homeassistant.components.zha.core.channels] [0x52b3:1:0x0001]: reporting 'battery_voltage' attr on 'power' cluster: 3600/10800/1: Result: '[[<ConfigureReportingResponseRecord status=Status.SUCCESS>]]'
```

Where do I find that?

rpatterson, 2019-10-17:
I just chanced across `Clusters` in the ZHA Configuration UI.  If I:

1. select the device
2. select `PowerConfigurationCluster (Endpoint id: 1, Id: 0x0001, Type: in)` under `Clusters`
3. select `battery_voltage (id: 0x0020)` under `Attributes of the selected cluster`
4. click `Get Zigbee Attribute`

I see `26` under `Value`.  I assume that's what you mean? FWIW here are some other seemingly related values:

- `battery_volt_min_thres (id: 0x0036)` = 21
- `battery_volt_thres1 (id: 0x0037)` = 22
- `battery_volt_thres2 (id: 0x0038)` = 23
- `battery_volt_thres3 (id: 0x0039)` = 24

rpatterson, 2019-10-17:
I just:

1. removed the device in the ZHA Configuration UI
2. reset it
3. added it back to SmartThings
4. it shows `Battery 55%` in SmartThings
5. deleted it from SmartThings
6. reset it
7. added it again in the ZHA Configuration UI
8. HA shows `Battery 100%`

rpatterson, 2019-10-17:
Guessing that those `battery_volt_*thres*` attributes are what is used to estimate the percentage, I tried setting `battery_volt_thres3` to 27 but if I get the attribute after setting it, it's value is still 24.  Ditto for setting `battery_voltage` to 23.

rpatterson, 2019-10-17:
I also checked the `battery_voltage` of all 5 of my Lightify Dimmer Switches and most are 26 and a couple are 27.

rpatterson, 2019-10-17:
I found the difference between [the SmartThings logic](https://github.com/smartthingscommunity/SmartThingsPublic/blob/master/devicetypes/smartthings/zigbee-button.src/zigbee-button.groovy#L99-L117) and [the HA logic](https://github.com/dmulcahey/zha-device-handlers/blob/dev/zhaquirks/centralite/__init__.py#L16-L35) that accounts for ST thinking the battery is at 55% and HA thinking the battery is at 100%.  Essentially they're using different scales.

So the question is, which scale/logic is "correct".  Before I switched to HA, the most frequently used of my switches got low enough on battery that it stopped working; I sure wish I had the `battery_voltage` from that battery before that happened.  But I do know that my switches have been in use for ~1.5 years and that I've only changed the battery on 1, so given that one got low enough to fail and that all have been in use that long, having them read as 100% seems wrong.

Adminiuga, 2019-10-17:
@dmulcahey if `battery_volt_min_thres (id: 0x0036) = 21` then we should use that as the min voltage threshold and adjust the mapping table accordingly or use the linear scale instead.

rpatterson, 2019-10-17:
FYI, I put a new (unused in package but purchased ~1 year ago) battery in and `battery_voltage` reports 31.

rpatterson, 2019-10-17:
Is there a way to use a checkout of this repo in the docker release of HA?

Adminiuga, 2019-10-17:
you could try the approach in https://github.com/zigpy/zigpy-deconz/issues/44#issuecomment-503810629

rpatterson, 2019-10-17:
I ended up using a checkout of this repo exploiting the fact that python puts `$CWD` at the front of `sys.path`:

```
$ docker-compose -f /opt/home-assistant/docker-compose.yml exec homeassistant readlink /proc/1/cwd
/config
```

So I added to my `docker-compose.yml`:

```
    volumes:
      ...
      - /opt/home-assistant/config/zha-device-handlers/zhaquirks:/config/zhaquirks
      ...
```

And then rebuilt:

```
$ docker-compose -f /opt/home-assistant/docker-compose.yml down && docker-compose -f /opt/home-assistant/docker-compose.yml up -d
```

dmulcahey, 2019-10-24:
fixed by #178

PR #208 (2019-12-22): Linear battery voltage
Initial implementation of linear battery voltages.

Fixes: #199
Potential fix: #176

Reported values:
* Existing batteries:
  *  [zhaquirks] CentraLite 3321-S Voltage [RAW]:29 [Max]:2.8 [Min]:1.5, Battery Percent: 100.0
  *  [zhaquirks] CentraLite 3325-S Voltage [RAW]:29 [Max]:2.8 [Min]:1.5, Battery Percent: 100.0
*  New batteries:
   * [zhaquirks] SmartThings tagv4, Voltage [RAW]:30 [Max]:2.8 [Min]:1.5, Battery Percent: 100.0
closed 2019-10-24 by dmulcahey