zigpy/zha-device-handlers #177: triple click don't work (WXKG11LM)

Nemesis24, 2019-10-15:
[image]

dmulcahey, 2019-10-16:
I'm confused here... in your other issue: #173 it appears fine. nothing has changed in between... did you change something locally?

Nemesis24, 2019-10-16:
I didn't change anything, but i see that the battery was added so i think there is some changes, no?

dmulcahey, 2019-10-16:
There haven’t been any releases.

Nemesis24, 2019-10-16:
Is it possible that in this commit:
https://github.com/home-assistant/home-assistant/commit/1087abd3b52f73640b189946e9cabdc0887e2410#diff-6aa1806cad190b7724f24144c9210a37 

the file:

homeassistant/components/zha/.translations/fr.json

the line: 

"remote_button_triple_press": "Bouton\"{sous-type}\" \u00e0 trois clics"

must have {subtype} and not {sous-type}

i think this is the reason ...
Thank you.

dmulcahey, 2019-10-24:
fixed by #183

closed 2019-10-24 by dmulcahey