zigpy/zha-device-handlers #180: Add support for Feibit Scene Switches, 1 & 4 button

highground88, 2019-10-20:
Hoping you can assist adding support for these Feibit devices. Thank you for the excellent work so far!

These devices are wall switches with no relays/physical outputs - they are like remotes and rely on binding to one or more devices. I've put both them on the same request as I assume they will work similarly. The device models are **FTB56+ZSN15HG1.0** & **FB56+ZSN08KJ2.3**. There is another model which has 2 scenes however I don't own one of these, but the model is **FTB56+ZSN16HG1.0** in case you would also like to add it for others.

I located these links below which I believe are for Smart Things if they may assist in any way?:
[Virtual Device Sync?](https://storage.googleapis.com/wzukusers/user-22742145/documents/c4723b09ac374810952477b6510b327e/Virtual%20Device%20Sync.docx)
[Switch Binder?](https://storage.googleapis.com/wzukusers/user-22742145/documents/cf42a270f98d4db995910de674fb2e6e/Nue%202%20Way%20Switch%20Binder.docx)

I have also managed to get these to work previously with Koenkk's assistance in zigbee2mqtt.

Please let me know if I can do anything else to assist. Thanks!

### **"1 SCENE SWITCH"**
Manufacturer: Feibit CO.,LTD  
Model: FTB56+ZSN15HG1.0
<SimpleDescriptor endpoint=1 profile=260 device_type=4 device_version=0 input_clusters=[0, 3] output_clusters=[3, 4, 5, 6, 1]>

Output once bound to another switch:
    _"event_type": "zha_event",
    "data": {
        "unique_id": "00:12:4b:00:1b:d3:bc:ce:1:0x0006",
        "device_ieee": "00:12:4b:00:1b:d3:bc:ce",
        "endpoint_id": 1,
        "cluster_id": 6,
        "command": "on",
        "args": []_

**JOINING OUTPUT**
```
[0x0000:zdo] ZDO request 0x0036: [60, <Bool.false: 0>]
[0x6c01:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=1 command_id=Command.Report_Attributes>
[0x6c01:1:0x0001] ZCL request 0x000a: [[<Attribute attrid=53 value=<TypeValue type=uint8_t, value=0>>]]
[0x6c01:1:0x0001] Attribute report received: battery_alarm_mask=0
Device 0xa6ec (00:12:4b:00:1b:d3:bc:ce) joined the network
[0xa6ec] Requesting 'Node Descriptor'
[0xa6ec] Extending timeout for 0xa7 request
Device 0xa6ec (00:12:4b:00:1b:d3:bc:ce) joined the network
Skip initialization for existing device 00:12:4b:00:1b:d3:bc:ce
[0xa6ec] Node Descriptor: <Optional byte1=1 byte2=64 mac_capability_flags=142 manufacturer_code=4478 maximum_buffer_size=80 maximum_incoming_transfer_size=160 server_mask=0 maximum_outgoing_transfer_size=160 descriptor_capability_field=0>
[0xa6ec] Discovering endpoints
[0xa6ec] Discovered endpoints: [1]
[0xa6ec:1] Discovering endpoint information
[0xa6ec:1] Discovered endpoint information: <SimpleDescriptor endpoint=1 profile=260 device_type=4 device_version=0 input_clusters=[0, 3] output_clusters=[3, 4, 5, 6, 1]>
[0xa6ec:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=173 command_id=Command.Read_Attributes_rsp>
[0xa6ec:1] Manufacturer: Feibit CO.,LTD  
[0xa6ec:1] Model: FTB56+ZSN15HG1.0
Checking quirks for Feibit CO.,LTD   FTB56+ZSN15HG1.0 (00:12:4b:00:1b:d3:bc:ce)
Considering <class 'zigpy.quirks.smartthings.SmartthingsMultiPurposeSensor'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
Fail because endpoint list mismatch: {11, 13} {1}
Considering <class 'zhaquirks.lutron.lzl4bwhl01remote.LutronLZL4BWHL01Remote2'>
Fail because profile_id mismatch on at least one endpoint
Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.osram.a19twhite.A19TunableWhite'>
Fail because endpoint list mismatch: {3} {1}
Considering <class 'zhaquirks.philips.rwl021.PhilipsRWL021'>
Fail because endpoint list mismatch: {1, 2} {1}
Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.xbee.xbee_io.XbeeSensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.xiaomi.mija.smoke.MijiaHoneywellSmokeDetectorSensor'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
Fail because device_type mismatch on at least one endpoint
device - 0xa6ec:00:12:4b:00:1b:d3:bc:ce entering async_device_initialized - is_new_join: True
[... 180 lines cut ...]
[0x899c:2:0x0006]: channel: 'async_initialize' stage succeeded
[0x899c:1:0x0005]: initializing channel: from_cache: False
[0x899c:1:0x0005]: channel: 'async_initialize' stage succeeded
[0x899c:1:0x0006]: initializing channel: from_cache: False
[0x899c:1:0x0006]: channel: 'async_initialize' stage succeeded
[0x8c08:zdo] ZDO request 0x0006: [0xfffd, 260, [25], []]
[0x899c:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=222 command_id=Command.Read_Attributes_rsp>
[0x899c:1:0x0000]: initializing channel: from_cache: False
[0x899c:1:0x0000]: channel: 'async_initialize' stage succeeded
[0x899c](FB56+ZSN08KJ2.3): power source: Mains
[0x899c](FB56+ZSN08KJ2.3): completed initialization
[0x899c:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=0 command_id=Command.Report_Attributes>
[0x899c:1:0x0001] ZCL request 0x000a: [[<Attribute attrid=53 value=<TypeValue type=uint8_t, value=0>>]]
[0x899c:1:0x0001] Attribute report received: battery_alarm_mask=0
[0x1c16:zdo] ZDO request 0x0001: [0x0000, 0, 0]
[0x8c08:zdo] ZDO request 0x0001: [0x0000, 0, 0]
[0x1c16:1:0x0019] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=202 command_id=Command.Read_Attributes_rsp>
[0x1c16:1:0x0019] ZCL request 0x0101: [0, 4478, 0, 1, None]
[0x1c16:1:0x0019] OTA query_next_image handler for 'FeiBit FNB56-ZSW03LX2.0': field_control=0, manufacture_id=4478, image_type=0, current_file_version=1, hardware_version=None
[0x1c16:1:0x0019] No OTA image is available
[0x8c08:11:0x0019] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=156 command_id=Command.Read_Attributes_rsp>
[0x8c08:11:0x0019] ZCL request 0x0101: [0, 4478, 0, 1, None]
[0x8c08:11:0x0019] OTA query_next_image handler for 'FeiBit FNB56-ZSW01LX2.0': field_control=0, manufacture_id=4478, image_type=0, current_file_version=1, hardware_version=None
[0x8c08:11:0x0019] No OTA image is available
```

closed 2021-08-20 by github-actions[bot]

tapd, 2022-01-07:
Hello, I have one of these 4 Scene switches (FB56+ZSN08KJ2.3) that I've been using for years. I recently changed my ZigBee co-ordinator to the Sonoff ZigBee 3 which recognises the FeiBit devices. I'd like to be able to develop a zha custom quirk for this scene switch but I'm not just not sure where to begin even after reading through the documentation.

Can anyone assist with this please?

MattWestb, 2022-01-07:
Its looks like the 1 scene switch is sending light commands (On/Off) and shall being easy getting automation from.
The 4 scene i have not looking at but its looks having cluster for light steering and shall being OK getting automations tringging from it.

You can using the tuya TS044F quirk that is one dimmer switch as ground and changing the cluster in the signature so ZHA is matching and loading the quirk. Then the quirk is loading you can working on the automation.

Use local quirk for doing the test and restarting HA between making the changes so its reloading the code OK.

tapd, 2022-01-08:
> 
I have one of those Tuya switches using the ts004f quirk, so I'll give this a go and see how far I get. Thanks for pointing me in the right direction.