zigpy/zha-device-handlers #186: Yale Real Living SYDL220 (Australian version of YRL220): Battery drains in days

thimic, 2019-10-30:
Hi,

I have the Yale Real Living SYDL220 lock, connected to a Raspbee hat using the ZHA component in Home Assistant. The Raspberry Pi running Home Assistant is located about 1m away from the door lock, so signal strength should not be an issue.

When not using the Lock’s Zigbee module (ZBM1), the battery lasts for months. When connecting the lock to my Home Assistant setup through the Lovelace UI, the battery drains within a few days. I am new to Zigbee devices and don’t quite know where to start with regard to debugging this. My guess is that the lock never goes to sleep, though that’s really just a hunch. There is a separate Zigbee setup in the apartment with Philips Hue lights and a Hue Bridge, though I have no idea if this might cause issues.

Additionally, the lock shows up in Home Assistant with two battery sensors, none of which are working. And the lock does not report its state in a manner that Home Assistant recognises.

I think I have found a cluster attribute that reflects the lock’s state, though none of the long list of battery attributes seem to produce a value.

Any help on this appreciated. I’m happy to provide any information that might be needed to figure this out. The battery drain especially is a bit painful.

Thanks,
Michael

[image]
[image]
[image]
[image]

dmulcahey, 2019-11-01:
This issue should be in home assistant. The quirk for this doesn't affect this. Can you move this issue to the home assistant repo and do the following: remove the device from the ZHA config UI, wait 5 minutes and then re-add it. Take the entire contents of the device adding panel's console and attach it to the ticket.

thimic, 2019-11-01:
Thanks @dmulcahey. I'll do that today and close this issue ones done.

thimic, 2019-11-02:
This is now done. See [Home Assistant issue](https://github.com/home-assistant/home-assistant/issues/28462).

closed 2019-11-02 by thimic