zigpy/zha-device-handlers #187: Ikea E1743 On/Off Switch Support

mannkind, 2019-11-01:
This is a feature/feasibility request to support the Ikea E1743 On/Off Switch that comes [separate](https://www.ikea.com/us/en/p/tradfri-wireless-dimmer-white-10408598/) or [bundled with a bulb](https://www.ikea.com/us/en/p/tradfri-dimmer-kit-90435932/).

I'd be happy to help with debugging/logs/even attempting to write a handler, but I'm not entirely sure where to start. I'm also willing to send someone one of the E1743's if that'd be helpulf.

For what it's worth, ZHA does does pair and reports battery level.

[image]

Additional information that _appears_ to be important from looking at `fivebtnremote.py` (for example).
```
Manufacturer: IKEA of Sweden
Model: IKEA of Sweden TRADFRI on/off switch
<SimpleDescriptor endpoint=1 profile=260 device_type=2080 device_version=1 input_clusters=[0, 1, 3, 9, 32, 4096, 64636] output_clusters=[3, 4, 6, 8, 25, 258, 4096]>
```

Thanks!

dmulcahey, 2019-11-01:
interesting... this may just work ootb and it may not... have you gone into the event tool in HA and subscribed to zha_event to see if you get events for button pushes?

mannkind, 2019-11-01:
That’s what I was hoping: zha_event. But, In the event tool I don’t see zha_event as an option. I tried subscribing anyway, but nothing seems to come through when pushing (long or short) the button.

harkoprinssen, 2019-11-04:
you need to subscribe to the zha_event, it's a manual action, not an option. Find the event tab and scroll down a bit :)

closed 2019-12-01 by dmulcahey

mannkind, 2019-12-02:
I upgraded zha_quirks to the dev branch. I’m now able to see zha_event events from the buttons in HA.

Thank you!

adrum, 2019-12-08:
I reported this in Discord, but thought I should post it here as well.  I picked up a couple of these in this kit [here](https://www.ikea.com/us/en/p/tradfri-control-outlet-kit-70364803/). The ones I received used the ZLL profile instead of ZHA:

Manufacturer: IKEA of Sweden
Model: IKEA of Sweden TRADFRI on/off switch

```<SimpleDescriptor endpoint=1 profile=49246 device_type=2080 device_version=248 input_clusters=[0, 1, 3, 9, 258, 4096, 64636] output_clusters=[3, 4, 6, 8, 25, 258, 4096]>```

I actually tried making my own quirk. It picked up properly, but it didn't seem to do much since none of the expected events were displayed. 

`twobtnremotezll.py`
```
"""Device handler for IKEA of Sweden TRADFRI remote control (ZLL)."""
from zigpy.profiles import zll
from zigpy.quirks import CustomCluster, CustomDevice
from zigpy.zcl.clusters.closures import WindowCovering
from zigpy.zcl.clusters.general import (
    Alarms,
    Basic,
    Groups,
    Identify,
    LevelControl,
    OnOff,
    Ota,
    PollControl,
    PowerConfiguration,
)
from zigpy.zcl.clusters.lightlink import LightLink

from . import IKEA, LightLinkCluster
from .. import DoublingPowerConfigurationCluster
from ..const import (
    ARGS,
    CLUSTER_ID,
    COMMAND,
    COMMAND_MOVE,
    COMMAND_MOVE_ON_OFF,
    COMMAND_OFF,
    COMMAND_ON,
    DEVICE_TYPE,
    DIM_DOWN,
    DIM_UP,
    ENDPOINT_ID,
    ENDPOINTS,
    INPUT_CLUSTERS,
    LONG_PRESS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
    SHORT_PRESS,
    TURN_OFF,
    TURN_ON,
)

IKEA_CLUSTER_ID = 0xFC7C  # decimal = 64636

class IkeaTradfriRemote2BtnZLL(CustomDevice):
    """Custom device representing IKEA of Sweden TRADFRI remote control."""

    signature = {
        # <SimpleDescriptor endpoint=1 profile=49246 device_type=2080
        # device_version=248
        # input_clusters=[0, 1, 3, 9, 258, 4096, 64636]
        # output_clusters=[3, 4, 6, 8, 25, 258, 4096]>

        MODELS_INFO: [(IKEA, "TRADFRI on/off switch")],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zll.PROFILE_ID,
                DEVICE_TYPE: zll.DeviceType.CONTROLLER,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    Identify.cluster_id,
                    Alarms.cluster_id,
                    WindowCovering.cluster_id,
                    LightLink.cluster_id,
                    IKEA_CLUSTER_ID,
                ],
                OUTPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Groups.cluster_id,
                    OnOff.cluster_id,
                    LevelControl.cluster_id,
                    Ota.cluster_id,
                    WindowCovering.cluster_id,
                    LightLink.cluster_id,
                ],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zll.PROFILE_ID,
                DEVICE_TYPE: zll.DeviceType.CONTROLLER,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    DoublingPowerConfigurationCluster,
                    Identify.cluster_id,
                    Alarms.cluster_id,
                    WindowCovering.cluster_id,
                    LightLinkCluster,
                    IKEA_CLUSTER_ID,
                ],
                OUTPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Groups.cluster_id,
                    OnOff.cluster_id,
                    LevelControl.cluster_id,
                    Ota.cluster_id,
                    WindowCovering.cluster_id,
                    LightLink.cluster_id,
                ],
            }
        }
    }

    device_automation_triggers = {
        (SHORT_PRESS, TURN_ON): {COMMAND: COMMAND_ON, CLUSTER_ID: 6, ENDPOINT_ID: 1},
        (LONG_PRESS, DIM_UP): {
            COMMAND: COMMAND_MOVE_ON_OFF,
            CLUSTER_ID: 8,
            ENDPOINT_ID: 1,
            ARGS: [0, 83],
        },
        (SHORT_PRESS, TURN_OFF): {COMMAND: COMMAND_OFF, CLUSTER_ID: 6, ENDPOINT_ID: 1},
        (LONG_PRESS, DIM_DOWN): {
            COMMAND: COMMAND_MOVE,
            CLUSTER_ID: 8,
            ENDPOINT_ID: 1,
            ARGS: [1, 83],
        },
    }
```

adrum, 2019-12-14:
@dmulcahey Can you provide any direction on this? I feel like I was close.

adrum, 2019-12-26:
@dmulcahey any update on this?

PR #203 (2019-12-01): Add support for the IKEA Tradfri on off switch
Fixed #187