zigpy/zha-device-handlers #190: Support new Ikea Tradfri motion sensor E1745

h3ndrik, 2019-11-13:
There is a new version of the Tradfri motion sensor avaiable:
https://www.ikea.com/de/de/p/tradfri-funk-bewegungsmelder-weiss-70429913/

I believe the quirks should be somewhat like this, but i can't get it to work:

        MANUFATURER_SPECIFIC_CLUSTER_ID = 0xfc7c # decimal = 64636

        # <SimpleDescriptor endpoint=1 profile=260 device_type=2128
        # device_version=1
        # input_clusters=[0, 1, 3, 9, 32, 4096, 64636]
        # output_clusters=[3, 4, 6, 8, 25, 4096]>
        MODELS_INFO: [(IKEA, "TRADFRI motion sensor")],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SENSOR,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    Identify.cluster_id,
                    Alarms.cluster_id,
                    PollControl.cluster_id,
                    LightLink.cluster_id,
                    MANUFATURER_SPECIFIC_CLUSTER_ID,
                ],
                OUTPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Groups.cluster_id,
                    OnOff.cluster_id,
                    LevelControl.cluster_id,
                    Ota.cluster_id,
                    LightLink.cluster_id,
                ],
            }

dmulcahey, 2019-11-28:
Can you try the lightlink cluster in #203 and let me know if it works?

h3ndrik, 2020-01-04:
thank you very much. that does work. i prepared a pull request with the changes (#232)

closed 2020-01-13 by h3ndrik