zigpy/zha-device-handlers #191: Supporting discontinued IKEA wireless dimmer

waweber, 2019-11-16:
I'm looking at adding support for the older, now discontinued wireless dimmers from IKEA. They're the round style ones that you can only rotate left or right.

I looked at how adding quirks works based on other examples, but I could use some guidance.

Example code for reference:

```python
class IkeaTradfriWirelessDimmer(CustomDevice):
    """Custom device representing the rotating wireless dimmers from IKEA."""

    signature = {
        # <SimpleDescriptor endpoint=1 profile=260 device_type=2064
        # device_version=2
        # input_clusters=[0, 1, 3, 9, 2821, 4096]
        # output_clusters=[3, 4, 6, 8, 25, 4096]>
        MODELS_INFO: [(IKEA, "TRADFRI wireless dimmer")],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zll.PROFILE_ID,
                DEVICE_TYPE: zll.DeviceType.COLOR_SCENE_CONTROLLER,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    Identify.cluster_id,
                    Alarms.cluster_id,
                    DIAGNOSTICS_CLUSTER_ID,
                    LightLink.cluster_id,
                ],
                OUTPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Groups.cluster_id,
                    OnOff.cluster_id,
                    LevelControl.cluster_id,
                    Ota.cluster_id,
                    LightLink.cluster_id,
                ],
            }
        },
    }

    # no changes...
    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zll.PROFILE_ID,
                DEVICE_TYPE: zll.DeviceType.COLOR_SCENE_CONTROLLER,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    Identify.cluster_id,
                    Alarms.cluster_id,
                    DIAGNOSTICS_CLUSTER_ID,
                    LightLink.cluster_id,
                ],
                OUTPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Groups.cluster_id,
                    OnOff.cluster_id,
                    LevelControl.cluster_id,
                    Ota.cluster_id,
                    LightLink.cluster_id,
                ],
            }
        }
    }

    device_automation_triggers = {
        (DEVICE_ROTATED, LEFT): {
            COMMAND: COMMAND_MOVE,
            CLUSTER_ID: LevelControl.cluster_id,
            ARGS: [1, int],
        },
        (DEVICE_ROTATED, RIGHT): {
            COMMAND: COMMAND_MOVE,
            CLUSTER_ID: LevelControl.cluster_id,
            ARGS: [0, int],
        },
    }
```

I've also added the "DEVICE_ROTATED" string to the constants. Overall, I will need to add several other commands (including STOP).

Several questions:

1. Do the action strings in HA need to be defined somewhere first? Zigbee2mqtt handles this device and supports a "rotate left" and also "rotate left quickly" action (based on the second int in the ARGS). I can't seem to find where these actions are defined.
2. These appear to be Voluptuous schemas, so putting the `int` type there should make it match "any integer" in the second argument, correct? (this represents the rate at which the device is being rotated)
3. Is there a way to make multiple zigbee commands result in the same action? The device emits "MOVE" and also "MOVE_WITH_ON_OFF" commands, and maybe those should both just be turned into "rotated left/right" actions. Can I put some sort of OR schema in here or something?
4. Zigbee2mqtt handles this device in a far more complicated way: whenever a MOVE or STOP command is received, it uses the rotation rate, value, and timestamp from the last move command, and interpolates how much the value would have changed in that time. It then emits an action containing a `"brightness": value` key. Thus, the CustomDevice instance would need to maintain an internal state, and also be able to intercept/process/rewrite several commands in a way that `device_automation_triggers` doesn't seem to be able to do. Is this something that can be done with this library?

dmulcahey, 2019-11-19:
This should already work. it fires ZHA_EVENT in HA.

waweber, 2019-11-19:
Yes, it does and I suppose I could make automations based on zha_event. I'm trying to reproduce the level of support zigbee2mqtt has, where it emits specific device actions and keeps track of the interpolated level

dmulcahey, 2020-12-21:
I believe the light stepping and the blueprint support recently added to HA is all you need for this. There is a blueprint on the forum for the symfonisk remote that should be easy enough to modify. Closing this as it is not a quirks concern.

closed 2020-12-21 by dmulcahey