zigpy/zha-device-handlers #198: Samsung Smart Button seems to get added however does not work when listening for zha_events

ethanopp, 2019-11-22:
[0x0000:zdo] ZDO request 0x0036: [60, <Bool.false: 0>]
Device 0x09e7 (28:6d:97:00:01:04:d7:c4) joined the network
[0x09e7] Requesting 'Node Descriptor'
[0x09e7] Extending timeout for 0x63 request
[0xb98c] Delivery error for seq # 0x5f, on endpoint id 1 cluster 0x0500: message send failure
[0xb98c:1:0x0500]: channel: 'async_configure' stage succeeded
[0xb98c](button): completed configuration
[0xb98c](button): stored in registry: ZhaDeviceEntry(name='Samjin button', ieee='28:6d:97:00:01:04:d7:c4', last_seen=1574436140.791139)
Device 0x4669 (28:6d:97:00:01:04:d7:c4) joined the network
Device 28:6d:97:00:01:04:d7:c4 changed id (0x09e7 => 0x4669)
Canceling old initialize call
[0x4669] Requesting Node Descriptor failed: 
[0x4669] Discovering endpoints
[0x4669] Extending timeout for 0x65 request
[0x4669] Requesting 'Node Descriptor'
[0x4669] Extending timeout for 0x67 request
Device 0x4669 (28:6d:97:00:01:04:d7:c4) joined the network
Skip initialization for existing device 28:6d:97:00:01:04:d7:c4
[0x4669:zdo] ZDO request 0x0013: [0x4669, 28:6d:97:00:01:04:d7:c4, 128]
[0x4669] Discovered endpoints: [1]
[0x4669:1] Discovering endpoint information
[0x4669] Extending timeout for 0x69 request
[0x4669] Node Descriptor: <Optional byte1=2 byte2=64 mac_capability_flags=128 manufacturer_code=4673 maximum_buffer_size=82 maximum_incoming_transfer_size=82 server_mask=11264 maximum_outgoing_transfer_size=82 descriptor_capability_field=0>
[0x4669] Discovering endpoints
[0x4669] Extending timeout for 0x6b request
[0x4669:1] Discovered endpoint information: <SimpleDescriptor endpoint=1 profile=260 device_type=1026 device_version=0 input_clusters=[0, 1, 3, 32, 1026, 1280, 2821] output_clusters=[3, 25]>
[0x4669] Extending timeout for 0x6d request
[0x4669] Discovered endpoints: [1]
[0x4669:1] Discovering endpoint information
[0x4669] Extending timeout for 0x6f request
Ignoring message (b'186d01040000420653616d6a696e0500004206627574746f6e') on cluster 0: unknown endpoint or cluster id: 'No cluster ID 0x0000 on (28:6d:97:00:01:04:d7:c4, 1)'
[0x4669:1] Discovered endpoint information: <SimpleDescriptor endpoint=1 profile=260 device_type=1026 device_version=0 input_clusters=[0, 1, 3, 32, 1026, 1280, 2821] output_clusters=[3, 25]>
[0x4669] Extending timeout for 0x71 request
[0x4669:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=113 command_id=Command.Read_Attributes_rsp>
[0x4669:1] Manufacturer: Samjin
[0x4669:1] Model: button
Checking quirks for Samjin button (28:6d:97:00:01:04:d7:c4)
Considering <class 'zhaquirks.samjin.button.SamjinButton'>
Found custom device replacement for 28:6d:97:00:01:04:d7:c4: <class 'zhaquirks.samjin.button.SamjinButton'>
device - 0x4669:28:6d:97:00:01:04:d7:c4 entering async_device_initialized - is_new_join: True
device - 0x4669:28:6d:97:00:01:04:d7:c4 has joined the ZHA zigbee network
[0x4669](button): started configuration
[0x4669:ZDO](button): channel: 'async_configure' stage succeeded
[0x4669] Extending timeout for 0x73 request
[0x4669] Extending timeout for 0x75 request
[0x4669] Extending timeout for 0x77 request
[0x4669:1:0x0000]: bound 'basic' cluster: Status.SUCCESS
[0x4669:1:0x0000]: finished channel configuration
[0x4669] Extending timeout for 0x79 request
[0x4669:1:0x0001]: bound 'power' cluster: Status.SUCCESS
[0x4669] Extending timeout for 0x7b request
[0x4669:1:0x0402]: bound 'temperature' cluster: Status.SUCCESS
[0x4669] Extending timeout for 0x7d request
[0x4669:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=121 command_id=Command.Read_Attributes_rsp>
[0x4669:1:0x0000]: initializing channel: from_cache: False
[0x4669:1:0x0000]: channel: 'async_configure' stage succeeded
[0x4669:1:0x0500]: started IASZoneChannel configuration
[0x4669] Extending timeout for 0x7f request
[0x4669:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=123 command_id=Command.Configure_Reporting_rsp>
[0x4669:1:0x0001]: reporting 'battery_voltage' attr on 'power' cluster: 3600/10800/1: Result: '[[<ConfigureReportingResponseRecord status=Status.SUCCESS>]]'
[0x4669:1:0x0402] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=125 command_id=Command.Configure_Reporting_rsp>
[0x4669:1:0x0402]: reporting 'measured_value' attr on 'temperature' cluster: 30/900/50: Result: '[[<ConfigureReportingResponseRecord status=Status.SUCCESS>]]'
[0x4669:1:0x0500]: bound 'ias_zone' cluster: Status.SUCCESS
[0x4669] Extending timeout for 0x81 request
[0x4669:1:0x0500] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=129 command_id=Command.Write_Attributes_rsp>
[0x4669:1:0x0402]: finished channel configuration
[0x4669:1:0x0402]: channel: 'async_configure' stage succeeded
[0x4669:1:0x0500]: wrote cie_addr: 00:0d:6f:00:0b:5d:c0:09 to 'ias_zone' cluster: [<WriteAttributesStatusRecord status=Status.SUCCESS>]
[0x4669:1:0x0500]: finished IASZoneChannel configuration
[0x4669] Extending timeout for 0x83 request
[... 29 lines cut ...]
[0x4669:1:0x0500]: initializing channel: from_cache: False
[0x4669:1:0x0500]: channel: 'async_initialize' stage succeeded
[0x4669:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=147 command_id=Command.Read_Attributes_rsp>
[0x4669] Extending timeout for 0x95 request
[0x4669:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=149 command_id=Command.Read_Attributes_rsp>
[0x4669:1:0x0001]: initializing channel: from_cache: False
[0x4669:1:0x0001]: channel: 'async_initialize' stage succeeded
[0x4669](button): power source: Battery or Unknown
[0x4669](button): completed initialization
[0x4669:zdo] ZDO request 0x0036: [180, <Bool.true: 1>]
[0xe2b1:1:0x0500] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=12 command_id=0>
[0xe2b1:1:0x0500] ZCL request 0x0000: [1, 0, 0, 0]
[0xe2b1:1:0x0500]: Updated alarm state: 1
[0xe2b1:1:0x0500] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=13 command_id=0>
[0xe2b1:1:0x0500] ZCL request 0x0000: [1, 0, 0, 0]
[0xe2b1:1:0x0500]: Updated alarm state: 1
[0x4669] Extending timeout for 0x97 request
[0x4669:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=151 command_id=Command.Read_Attributes_rsp>
[0x4669:1:0x0000] ZCL request 0x0001: [[<ReadAttributeRecord attrid=4 status=Status.SUCCESS value=Samjin>]]
[0x4669] Extending timeout for 0x99 request
[0x4669:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=153 command_id=Command.Read_Attributes_rsp>
[0x4669:1:0x0000] ZCL request 0x0001: [[<ReadAttributeRecord attrid=5 status=Status.SUCCESS value=button>]]
[0x4669:1:0x0402] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=5 command_id=Command.Report_Attributes>
[0x4669:1:0x0402] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=int16s, value=2609>>]]
[0x4669:1:0x0402] Attribute report received: measured_value=2609

Adminiuga, 2019-11-22:
Do you get anything from `zigpy` or `deconz` in the log when you press the button?

ethanopp, 2019-11-22:
Oddly enough after restarting the docker container, the events seem to be triggering... Still getting some errors in the log though:
```
2019-11-22 10:43:38 WARNING (MainThread) [zigpy.zdo] [0xa26c:zdo] Unsupported ZDO request:0x8001
2019-11-22 10:43:38 WARNING (MainThread) [zigpy.zdo] [0xa26c:zdo] Unsupported ZDO request:0x8001
2019-11-22 10:43:38 WARNING (MainThread) [zigpy.zdo] [0xa26c:zdo] Unsupported ZDO request:0x8001
2019-11-22 10:44:19 WARNING (MainThread) [zigpy.zdo] [0xa26c:zdo] Unsupported ZDO request:0x8001
2019-11-22 10:53:28 WARNING (MainThread) [zigpy.zdo] [0xa26c:zdo] Unsupported ZDO request:0x8001
2019-11-22 11:01:13 WARNING (MainThread) [zigpy.zdo] [0xa26c:zdo] Unsupported ZDO request:0x8001
2019-11-22 11:02:44 WARNING (MainThread) [zigpy.zdo] [0xa26c:zdo] Unsupported ZDO request:0x8001
2019-11-22 11:07:58 WARNING (MainThread) [zigpy.zdo] [0xa26c:zdo] Unsupported ZDO request:0x8001
2019-11-22 11:08:40 WARNING (MainThread) [zigpy.zdo] [0xa26c:zdo] Unknown ZDO cluster 0x8038
2019-11-22 11:08:40 WARNING (MainThread) [zigpy.zdo] [0xa26c:zdo] Unsupported ZDO request:32824
2019-11-22 11:08:53 WARNING (MainThread) [zigpy.zdo] [0xa26c:zdo] Unsupported ZDO request:0x8001
2019-11-22 11:21:21 WARNING (MainThread) [zigpy.zdo] [0xa26c:zdo] Unsupported ZDO request:0x8001
2019-11-22 11:24:38 WARNING (MainThread) [zigpy.zdo] [0xa26c:zdo] Unknown ZDO cluster 0x8038
2019-11-22 11:24:38 WARNING (MainThread) [zigpy.zdo] [0xa26c:zdo] Unsupported ZDO request:32824
2019-11-22 11:26:30 WARNING (MainThread) [zigpy.zdo] [0xa26c:zdo] Unsupported ZDO request:0x8001
2019-11-22 11:26:30 WARNING (MainThread) [zigpy.zdo] [0xa26c:zdo] Unsupported ZDO request:0x8001
```

Adminiuga, 2019-11-22:
Those are safe to ignore. Those are debug messages in the current dev tree.

ethanopp, 2019-11-22:
Alright - well everything seems to be working for now from the container reset to ill close this for now...

Thanks

closed 2019-11-22 by ethanopp