zigpy/zha-device-handlers #199: Smartthings battery levels reporting incorrectly

ryanwinter, 2019-11-22:
Samsung devices shares it's battery range with centralite. Since this change, my devices are reporting low levels, with a bunch now reporting 0% battery, and they are still working.

https://github.com/dmulcahey/zha-device-handlers/commit/6cba0ff725db1e5d66b577ddfb283d2f7fd73313

Do the Samsung devices need their own battery logic to have them report correctly?

dmulcahey, 2019-11-22:
I’m not surprised. I’m not sold on the new ranges myself... can you monitor them so we can track the min side voltage where they stop working? That’s what I’m doing myself.

Adminiuga, 2019-11-22:
I think for devices that we know are using CR2 or CR123 battery we could use a lower voltage threshold.

ryanwinter, 2019-11-22:
The max voltage is also incorrect, I replaced batteries in a few devices recently, and now they are reporting 30%. 

I just checked and these devices are reporting 24 volts, which would have reported 90% on the old scale which seems correct.

For min voltage, I just replaced most of my devices as they had reached 1%. They were still working at that level (16 volts).

So basically I have seen devices working in the 16 - 26 volts range which seems to align well with the original range.

dmulcahey, 2019-11-22:
What kind of batteries? 100 is usually 3.0v for Lithium batteries at a minimum. Most start between 3.4 and 3.2ish. 2.4v seems really low for a new battery.

ryanwinter, 2019-11-22:
They are CR2032 so yes they are 3V. Maybe the reported voltage is not actual. I'll pull one down and measure it when I get a chance.

dmulcahey, 2019-11-22:
I have plenty of those in use and they generally start at 3.2 when new... I have 8 of the sylvania dimmers that use these

ryanwinter, 2019-11-22:
Referring to the smartthings community repo for this device, it uses the old scale:
https://github.com/SmartThingsCommunity/SmartThingsPublic/blob/master/devicetypes/smartthings/smartsense-multi-sensor.src/smartsense-multi-sensor.groovy

ryanwinter, 2019-11-22:
[Multi](https://github.com/SmartThingsCommunity/SmartThingsPublic/blob/93021d5ef014dfc497b04774ffa615c1a0b9d639/devicetypes/smartthings/smartsense-multi-sensor.src/smartsense-multi-sensor.groovy#L294), [motion ](https://github.com/SmartThingsCommunity/SmartThingsPublic/blob/93021d5ef014dfc497b04774ffa615c1a0b9d639/devicetypes/smartthings/smartsense-motion-sensor.src/smartsense-motion-sensor.groovy#L198)& [moisture](https://github.com/SmartThingsCommunity/SmartThingsPublic/blob/93021d5ef014dfc497b04774ffa615c1a0b9d639/devicetypes/smartthings/smartsense-moisture-sensor.src/smartsense-moisture-sensor.groovy#L182) all use the old logic.

roblandry, 2019-11-30:
I just noticed my smartthings tag v4 are not showing any battery, when pulling the values, they are in fact < 21. IF the new values added in 6cba0ff725db1e5d66b577ddfb283d2f7fd73313 are in fact correct for some devices, maybe we need to split out the values to retain the old logic where applicable.

roblandry, 2019-12-01:
@ryanwinter check out #208, it should resolve your issues.

PR #208 (2019-12-22): Linear battery voltage
Initial implementation of linear battery voltages.

Fixes: #199
Potential fix: #176

Reported values:
* Existing batteries:
  *  [zhaquirks] CentraLite 3321-S Voltage [RAW]:29 [Max]:2.8 [Min]:1.5, Battery Percent: 100.0
  *  [zhaquirks] CentraLite 3325-S Voltage [RAW]:29 [Max]:2.8 [Min]:1.5, Battery Percent: 100.0
*  New batteries:
   * [zhaquirks] SmartThings tagv4, Voltage [RAW]:30 [Max]:2.8 [Min]:1.5, Battery Percent: 100.0
closed 2019-12-22 by dmulcahey