zigpy/zha-device-handlers #202: Wrong IKEA Tradfri E14 WS Opal 400lm configuration

nerdosity, 2019-11-28:
Supported_features is set as 51 (RGB picker, dimmable, tunable white).
Should be 43 (dimmable, tunable white).

dmulcahey, 2019-11-28:
There isn’t a quirk for this which would mean the bulb is reporting this on its own.

nerdosity, 2019-11-28:
Well then should be very similar to this one, right?
https://github.com/dmulcahey/zha-device-handlers/blob/dev/zhaquirks/osram/a19twhite.py

dmulcahey, 2019-11-28:
Not following what you mean?

nerdosity, 2019-11-28:
Ok sorry I'm not expert, I just searched for a simple tunable white bulb, I thought it would be a good starting point for me to try. 
I would like to help, I have access to a CC2531 flashable with sniffer firmware, a zigate and tons of ZigBee devices (battery powered and mains).
Please let me know what I can provide.

dmulcahey, 2019-11-29:
Remove and re-join it with zha and paste the content of the add devices screen logs in a code block here. Then I’ll help you make a quirk.

dmulcahey, 2019-11-29:
We’ll probably have to do something like this: https://github.com/dmulcahey/zha-device-handlers/pull/182/files to override the value the bulb returns.

closed 2021-08-20 by github-actions[bot]