zigpy/zha-device-handlers #207: Add  support for eWeLink SA-003-Zigbee switch

martinlong1978, 2019-12-01:
I picked up some of these. These pair and working great on Smartthings: 

https://www.ebay.co.uk/itm/383218431663

Smartthings identifies this as a 
```
endpointId: 01
manufacturer: eWeLink
model: SA-003-Zigbee
zigbeeNodeType: ROUTE
```
It looks like Smartthings directly supports this using this device handler, which might be useful for IDs and implementation detail: https://raw.githubusercontent.com/SmartThingsCommunity/SmartThingsPublic/master/devicetypes/smartthings/zigbee-switch.src/zigbee-switch.groovy

When I search, it seems like something is found, but it fails to complete in some way. Here is my log, but please let me know if you need more. 
```
2019-12-01 10:06:31 DEBUG (MainThread) [zigpy_xbee.zigbee.application] Broadcast request seq 244
2019-12-01 10:06:31 DEBUG (MainThread) [zigpy_xbee.api] Command tx_explicit (00:00:00:00:00:00:ff:fc, <BroadcastAddress.ALL_ROUTERS_AND_COORDINATOR: 65532>, 0, 0, 54, 0, 0, 32, b'\xf4<\x00')
2019-12-01 10:06:31 DEBUG (MainThread) [zigpy_xbee.uart] Sending: b'\x11\x16\x00\x00\x00\x00\x00\x00\xff\xfc\xff\xfc\x00\x00\x006\x00\x00\x00 \xf4<\x00'
2019-12-01 10:06:31 DEBUG (MainThread) [zigpy_xbee.uart] Frame received: b'\x8b\x16\xff\xfe\x00-\x00'
2019-12-01 10:06:31 DEBUG (MainThread) [zigpy_xbee.api] Frame received: tx_status
2019-12-01 10:06:31 DEBUG (MainThread) [zigpy_xbee.api] tx_explicit to 0xfffe: TXStatus.BROADCAST_APS_TX_ATTEMPT after 0 tries. Discovery Status: DiscoveryStatus.SUCCESS, Frame #22
2019-12-01 10:06:31 DEBUG (MainThread) [zigpy_xbee.api] at command: NJ (60,)
2019-12-01 10:06:31 DEBUG (MainThread) [zigpy_xbee.api] Command at (b'NJ', b'<')
2019-12-01 10:06:31 DEBUG (MainThread) [zigpy_xbee.uart] Sending: b'\x08\x17NJ<'
2019-12-01 10:06:31 DEBUG (MainThread) [zigpy_xbee.uart] Frame received: b'\x88\x17NJ\x00'
2019-12-01 10:06:31 DEBUG (MainThread) [zigpy_xbee.api] Frame received: at_response
2019-12-01 10:06:31 DEBUG (MainThread) [zigpy_xbee.api] at command: AC ()
2019-12-01 10:06:31 DEBUG (MainThread) [zigpy_xbee.api] Command at (b'AC', b'')
2019-12-01 10:06:31 DEBUG (MainThread) [zigpy_xbee.uart] Sending: b'\x08\x18AC'
2019-12-01 10:06:31 DEBUG (MainThread) [zigpy_xbee.uart] Frame received: b'\x88\x18AC\x00'
2019-12-01 10:06:31 DEBUG (MainThread) [zigpy_xbee.api] Frame received: at_response
2019-12-01 10:06:31 DEBUG (MainThread) [zigpy_xbee.api] at command: CB (2,)
2019-12-01 10:06:31 DEBUG (MainThread) [zigpy_xbee.api] Command at (b'CB', b'\x02')
2019-12-01 10:06:31 DEBUG (MainThread) [zigpy_xbee.uart] Sending: b'\x08\x19CB\x02'
2019-12-01 10:06:31 DEBUG (MainThread) [zigpy_xbee.uart] Frame received: b'\x88\x19CB\x00'
2019-12-01 10:06:31 DEBUG (MainThread) [zigpy_xbee.api] Frame received: at_response
2019-12-01 10:06:54 DEBUG (MainThread) [zigpy_xbee.uart] Frame received: b'\x91\x00\x12K\x00\x1fR :\xb0\xb8\x00\x00\x00\x13\x00\x00\x02\x00\xb8\xb0: R\x1f\x00K\x12\x00\x8e'
2019-12-01 10:06:54 DEBUG (MainThread) [zigpy_xbee.api] Frame received: explicit_rx_indicator
2019-12-01 10:06:54 DEBUG (MainThread) [zigpy_xbee.api] _handle_explicit_rx: (00:12:4b:00:1f:52:20:3a, 0xb0b8, 0, 19, 2, b'00b8b03a20521f004b12008e')
2019-12-01 10:06:54 INFO (MainThread) [zigpy_xbee.zigbee.application] New device joined: NWK 0xb0b8, IEEE 00:12:4b:00:1f:52:20:3a
2019-12-01 10:06:54 INFO (MainThread) [zigpy.application] Device 0xb0b8 (00:12:4b:00:1f:52:20:3a) joined the network
2019-12-01 10:06:54 DEBUG (MainThread) [zigpy.zdo] [0xb0b8:zdo] ZDO request 0x0013: [0xb0b8, 00:12:4b:00:1f:52:20:3a, 142]
2019-12-01 10:06:54 INFO (MainThread) [zigpy.device] [0xb0b8] Requesting 'Node Descriptor'
2019-12-01 10:06:54 DEBUG (MainThread) [zigpy.device] [0xb0b8] Extending timeout for 0xf5 request
2019-12-01 10:06:54 DEBUG (MainThread) [zigpy_xbee.zigbee.application] Zigbee request tsn #245: b'f5b8b0'
2019-12-01 10:06:54 DEBUG (MainThread) [zigpy_xbee.api] Command tx_explicit (00:12:4b:00:1f:52:20:3a, 0xb0b8, 0, 0, 0x0002, 0, 0, 96, b'\xf5\xb8\xb0')
2019-12-01 10:06:54 DEBUG (MainThread) [zigpy_xbee.uart] Sending: b'\x11\x1a\x00\x12K\x00\x1fR :\xb0\xb8\x00\x00\x00\x02\x00\x00\x00`\xf5\xb8\xb0'
2019-12-01 10:06:54 DEBUG (MainThread) [zigpy_xbee.uart] Frame received: b'\x91\x00\x12K\x00\x1fR :\xb0\xb8\x01\xff\x00\x06\x01\x04\x00\x08\x00\n\x00\x00\x10\x00'
2019-12-01 10:06:54 DEBUG (MainThread) [zigpy_xbee.api] Frame received: explicit_rx_indicator
2019-12-01 10:06:54 DEBUG (MainThread) [zigpy_xbee.api] _handle_explicit_rx: (00:12:4b:00:1f:52:20:3a, 0xb0b8, 255, 6, 0, b'08000a00001000')
2019-12-01 10:06:54 DEBUG (MainThread) [zigpy.device] Ignoring message (b'08000a00001000') on cluster 6: unknown endpoint or cluster id: 1
2019-12-01 10:07:25 DEBUG (MainThread) [zigpy_xbee.uart] Frame received: b'\x8b\x1a\xff\xfd\x00$B'
2019-12-01 10:07:25 DEBUG (MainThread) [zigpy_xbee.api] Frame received: tx_status
2019-12-01 10:07:25 DEBUG (MainThread) [zigpy_xbee.api] tx_explicit to 0xfffd: TXStatus.ADDRESS_NOT_FOUND after 0 tries. Discovery Status: DiscoveryStatus.SUCCESS, Frame #26
2019-12-01 10:07:26 DEBUG (MainThread) [zigpy.device] [0xb0b8] Extending timeout for 0xf6 request
2019-12-01 10:07:26 DEBUG (MainThread) [zigpy_xbee.zigbee.application] Zigbee request tsn #246: b'f6b8b0'
2019-12-01 10:07:26 DEBUG (MainThread) [zigpy_xbee.api] Command tx_explicit (00:12:4b:00:1f:52:20:3a, 0xb0b8, 0, 0, 0x0002, 0, 0, 96, b'\xf6\xb8\xb0')
2019-12-01 10:07:26 DEBUG (MainThread) [zigpy_xbee.uart] Sending: b'\x11\x1b\x00\x12K\x00\x1fR :\xb0\xb8\x00\x00\x00\x02\x00\x00\x00`\xf6\xb8\xb0'
2019-12-01 10:07:58 DEBUG (MainThread) [zigpy_xbee.uart] Frame received: b'\x8b\x1b\xff\xfd\x04$B'
2019-12-01 10:07:58 DEBUG (MainThread) [zigpy_xbee.api] Frame received: tx_status
2019-12-01 10:07:58 DEBUG (MainThread) [zigpy_xbee.api] tx_explicit to 0xfffd: TXStatus.ADDRESS_NOT_FOUND after 4 tries. Discovery Status: DiscoveryStatus.SUCCESS, Frame #27
2019-12-01 10:07:58 WARNING (MainThread) [zigpy.device] [0xb0b8] Requesting Node Descriptor failed: TXStatus.ADDRESS_NOT_FOUND
2019-12-01 10:07:58 INFO (MainThread) [zigpy.device] [0xb0b8] Discovering endpoints
2019-12-01 10:07:58 DEBUG (MainThread) [zigpy.device] [0xb0b8] Extending timeout for 0xf7 request
2019-12-01 10:07:58 DEBUG (MainThread) [zigpy_xbee.zigbee.application] Zigbee request tsn #247: b'f7b8b0'
2019-12-01 10:07:58 DEBUG (MainThread) [zigpy_xbee.api] Command tx_explicit (00:12:4b:00:1f:52:20:3a, 0xb0b8, 0, 0, 0x0005, 0, 0, 96, b'\xf7\xb8\xb0')
2019-12-01 10:07:58 DEBUG (MainThread) [zigpy_xbee.uart] Sending: b'\x11\x1c\x00\x12K\x00\x1fR :\xb0\xb8\x00\x00\x00\x05\x00\x00\x00`\xf7\xb8\xb0'
2019-12-01 10:08:10 DEBUG (MainThread) [zigpy_xbee.uart] Frame received: b'\x91\x14\xb4W\xff\xfe+\xecuIJ\x01\x01\x00\x08\x01\x04\x00\x08A\n\x00\x00 \xfe'
2019-12-01 10:08:10 DEBUG (MainThread) [zigpy_xbee.api] Frame received: explicit_rx_indicator
2019-12-01 10:08:10 DEBUG (MainThread) [zigpy_xbee.api] _handle_explicit_rx: (14:b4:57:ff:fe:2b:ec:75, 0x494a, 1, 8, 0, b'08410a000020fe')
[... 6 lines cut ...]
2019-12-01 10:08:11 DEBUG (MainThread) [zigpy_xbee.api] _handle_explicit_rx: (14:b4:57:ff:fe:3c:d0:a2, 0x3952, 1, 8, 0, b'08410a000020fe')
2019-12-01 10:08:11 DEBUG (MainThread) [zigpy.zcl] [0x3952:1:0x0008] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=65 command_id=Command.Report_Attributes>
2019-12-01 10:08:11 DEBUG (MainThread) [zigpy.zcl] [0x3952:1:0x0008] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=uint8_t, value=254>>]]
2019-12-01 10:08:11 DEBUG (MainThread) [zigpy.zcl] [0x3952:1:0x0008] Attribute report received: current_level=254
2019-12-01 10:08:11 DEBUG (MainThread) [homeassistant.components.zha.core.channels] [0x3952:1:0x0008]: received attribute: 0 update with value: 254
2019-12-01 10:08:11 DEBUG (MainThread) [zigpy_xbee.uart] Frame received: b'\x91\x08k\xd7\xff\xfeR \x9d\xfa8\x01\x01\x00\x08\x01\x04\x00\x08?\n\x00\x00 \xfe'
2019-12-01 10:08:11 DEBUG (MainThread) [zigpy_xbee.api] Frame received: explicit_rx_indicator
2019-12-01 10:08:11 DEBUG (MainThread) [zigpy_xbee.api] _handle_explicit_rx: (08:6b:d7:ff:fe:52:20:9d, 0xfa38, 1, 8, 0, b'083f0a000020fe')
2019-12-01 10:08:11 DEBUG (MainThread) [zigpy.zcl] [0xfa38:1:0x0008] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=63 command_id=Command.Report_Attributes>
[... cut ...] +00:00>, new_state=<state device_tracker.40_4e_36_d3_dc_3f=home; source_type=router, latitude=52.97423600589089, longitude=-1.0983109474182131, gps_accuracy=0, scanner=LuciDeviceScanner, dev=br-lan, stale=False, mac=40:4E:36:D3:DC:3F, noarp=False, permanent=False, failed=False, family=4, proxy=False, router=False, reachable=True, probe=False, delay=False, incomplete=False, ip=10.100.128.94, hostname=None, host=10.100.128.1, friendly_name=40 4e 36 d3 dc 3f @ 2019-12-01T09:22:04.766631+00:00>>
```

Adminiuga, 2019-12-01:
Not a problem with zha-device-handlers, but apparently these  cheap devices do not support aps encryption. You could try locating `zigpy-xbee` module location and modify https://github.com/zigpy/zigpy-xbee/blob/1ef9fb9a7260823dd7f647b385207ffe28086393/zigpy_xbee/zigbee/application.py#L207 and change the value to 0x00

martinlong1978, 2019-12-01:
Thank you! That worked. Should I leave it set like that, or change it back
once paired?

Adminiuga, 2019-12-01:
Leave it. I don't think it would work with APS encryption enabled. Need to confirm whether it is enabled in EZSP and maybe disable it in xbee upstream.

closed 2021-08-20 by github-actions[bot]