zigpy/zha-device-handlers #209: Support for LDS ZBT-CCTSwitch-D0001

ad1217, 2019-12-05:
This is a 4 button remote that came from an [EcoSmart Kit](https://www.homedepot.com/p/EcoSmart-60-Watt-Equivalent-A19-SMART-LED-Light-Bulb-Tunable-White-Starter-Kit-1-Bulb-A9A19A60WESDZR1/310288335) sold by Home Depot. It pairs with ZHA and presents it's battery level, but produces nothing in the log when buttons are pressed. I have little experience with Zigbee, so I'm not sure where to go from here.

FCC ID is `2AB2Q6ARCZABZH` if that helps.

<details>
  <summary>Quirk Stub</summary>

```python3
"""Device handler for LDS ZBT-CCTSwitch-D001."""
from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
from zigpy.zcl.clusters.general import (
    Basic,
    Groups,
    PowerConfiguration,
    Identify,
    LevelControl,
    Ota,
    OnOff,
)
from zigpy.zcl.clusters.lighting import Color
from zigpy.zcl.clusters.lightlink import LightLink

from ..const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
 )

MANUFACTURER_SPECIFIC_CLUSTER_ID_1 = 0xFD01  # decimal = 64769

class LDSZBTCCTSwitch(CustomDevice):
    """Custom device representing LDS ZBT-CCTSwitch-D001."""

    signature = {
        #  <SimpleDescriptor endpoint=1 profile=260 device_type=2048
        #  device_version=1
        #  input_clusters=[0, 1, 3, 4096, 64769]
        #  output_clusters=[3, 4, 6, 8, 25, 768, 4096]>
        MODELS_INFO: [("LDS", "ZBT-CCTSwitch-D0001")],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.COLOR_CONTROLLER,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    Identify.cluster_id,
                    LightLink.cluster_id,
                    MANUFACTURER_SPECIFIC_CLUSTER_ID_1,
                ],
                OUTPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Groups.cluster_id,
                    OnOff.cluster_id,
                    LevelControl.cluster_id,
                    Ota.cluster_id,
                    Color.cluster_id,
                    LightLink.cluster_id,
                ],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
[... 318 lines cut ...]
2019-12-05 19:17:44 DEBUG (MainThread) [bellows.ezsp] Application frame 128 (incomingRouteErrorHandler) received: b'4254f1'
2019-12-05 19:17:44 DEBUG (MainThread) [bellows.ezsp] Application frame 128 (incomingRouteErrorHandler) received: b'4254f1'
2019-12-05 19:17:46 DEBUG (MainThread) [bellows.ezsp] Application frame 69 (incomingMessageHandler) received: b'000000218000004001000089ffcc54f1ffff022f00'
2019-12-05 19:17:46 DEBUG (MainThread) [bellows.ezsp] Application frame 63 (messageSentHandler) received: b'0054f10000210000004001000084320000'
2019-12-05 19:17:46 DEBUG (MainThread) [bellows.ezsp] Application frame 69 (incomingMessageHandler) received: b'00000021800000400100008affcc54f1ffff023000'
2019-12-05 19:17:46 DEBUG (MainThread) [bellows.ezsp] Application frame 63 (messageSentHandler) received: b'0054f10000210000004001000085330000'
2019-12-05 19:17:46 DEBUG (MainThread) [bellows.ezsp] Application frame 69 (incomingMessageHandler) received: b'00000021800000400100008bffcc54f1ffff023100'
2019-12-05 19:17:46 DEBUG (MainThread) [bellows.ezsp] Application frame 63 (messageSentHandler) received: b'0054f10000210000004001000086340000'
2019-12-05 19:17:46 INFO (MainThread) [homeassistant.components.zha.api] Issue bind devices: source_ieee: [cc:cc:cc:ff:fe:c6:d1:7a] target_ieee: [cc:cc:cc:ff:fe:8d:63:71]
2019-12-05 19:17:47 DEBUG (MainThread) [bellows.ezsp] Send command nop: ()
2019-12-05 19:17:48 DEBUG (MainThread) [bellows.ezsp] Application frame 5 (nop) received: b''
2019-12-05 19:17:58 DEBUG (MainThread) [bellows.ezsp] Send command nop: ()
2019-12-05 19:17:58 DEBUG (MainThread) [bellows.ezsp] Application frame 5 (nop) received: b''
2019-12-05 19:18:08 DEBUG (MainThread) [bellows.ezsp] Send command nop: ()
2019-12-05 19:18:08 DEBUG (MainThread) [bellows.ezsp] Application frame 5 (nop) received: b''
2019-12-05 19:18:18 DEBUG (MainThread) [bellows.ezsp] Send command nop: ()
2019-12-05 19:18:18 DEBUG (MainThread) [bellows.ezsp] Application frame 5 (nop) received: b''
2019-12-05 19:18:28 DEBUG (MainThread) [bellows.ezsp] Send command nop: ()
2019-12-05 19:18:28 DEBUG (MainThread) [bellows.ezsp] Application frame 5 (nop) received: b''
2019-12-05 19:18:38 DEBUG (MainThread) [bellows.ezsp] Send command nop: ()
2019-12-05 19:18:38 DEBUG (MainThread) [bellows.ezsp] Application frame 5 (nop) received: b''
2019-12-05 19:18:48 DEBUG (MainThread) [bellows.ezsp] Send command nop: ()
2019-12-05 19:18:48 DEBUG (MainThread) [bellows.ezsp] Application frame 5 (nop) received: b''
```
</details>

Adminiuga, 2019-12-06:
Pairing log looks normal. It bound all clusters it supposed to bind.
Bind attempt log show the request did not go through. Were you binding it to a light? Try again, but press a button on the remote just prior the binding attempt and one more time after you sent the bind request to keep device awake.

Also, in dev panel -> events subscribe to zha_event and press button on the remote. Do you get anything?

ad1217, 2019-12-06:
I get nothing when listening to `zha_event`.

I was binding it to a light, and pressing buttons before and after seems to have no effect. 
<details>
  <summary>Bind Attempt 2</summary>

```
2019-12-06 17:13:16 DEBUG (MainThread) [homeassistant.components.zha.api] processing binding operation for: source_ieee: [cc:cc:cc:ff:fe:c6:d1:7a] target_ieee: [cc:cc:cc:ff:fe:8d:63:71] cluster: 6
2019-12-06 17:13:16 DEBUG (MainThread) [homeassistant.components.zha.api] processing binding operation for: source_ieee: [cc:cc:cc:ff:fe:c6:d1:7a] target_ieee: [cc:cc:cc:ff:fe:8d:63:71] cluster: 8
2019-12-06 17:13:16 DEBUG (MainThread) [homeassistant.components.zha.api] processing binding operation for: source_ieee: [cc:cc:cc:ff:fe:c6:d1:7a] target_ieee: [cc:cc:cc:ff:fe:8d:63:71] cluster: 768
2019-12-06 17:13:16 DEBUG (MainThread) [zigpy.device] [0xb9b2] Extending timeout for 0x0b request
2019-12-06 17:13:16 DEBUG (MainThread) [bellows.ezsp] Send command setExtendedTimeout: (cc:cc:cc:ff:fe:c6:d1:7a, True)
2019-12-06 17:13:16 DEBUG (MainThread) [zigpy.device] [0xb9b2] Extending timeout for 0x0c request
2019-12-06 17:13:16 DEBUG (MainThread) [bellows.ezsp] Send command setExtendedTimeout: (cc:cc:cc:ff:fe:c6:d1:7a, True)
2019-12-06 17:13:16 DEBUG (MainThread) [zigpy.device] [0xb9b2] Extending timeout for 0x0d request
2019-12-06 17:13:16 DEBUG (MainThread) [bellows.ezsp] Send command setExtendedTimeout: (cc:cc:cc:ff:fe:c6:d1:7a, True)
2019-12-06 17:13:16 DEBUG (MainThread) [bellows.ezsp] Application frame 126 (setExtendedTimeout) received: b''
2019-12-06 17:13:16 DEBUG (MainThread) [bellows.ezsp] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0xb9b2, <EmberApsFrame profileId=0 clusterId=33 sourceEndpoint=0 destinationEndpoint=0 options=320 groupId=0 sequence=11>, 14, b'\x0bz\xd1\xc6\xfe\xff\xcc\xcc\xcc\x01\x06\x00\x03qc\x8d\xfe\xff\xcc\xcc\xcc\x01')
2019-12-06 17:13:16 DEBUG (MainThread) [bellows.ezsp] Application frame 126 (setExtendedTimeout) received: b''
2019-12-06 17:13:16 DEBUG (MainThread) [bellows.ezsp] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0xb9b2, <EmberApsFrame profileId=0 clusterId=33 sourceEndpoint=0 destinationEndpoint=0 options=320 groupId=0 sequence=12>, 15, b'\x0cz\xd1\xc6\xfe\xff\xcc\xcc\xcc\x01\x08\x00\x03qc\x8d\xfe\xff\xcc\xcc\xcc\x01')
2019-12-06 17:13:16 DEBUG (MainThread) [bellows.ezsp] Application frame 126 (setExtendedTimeout) received: b''
2019-12-06 17:13:16 DEBUG (MainThread) [bellows.ezsp] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0xb9b2, <EmberApsFrame profileId=0 clusterId=33 sourceEndpoint=0 destinationEndpoint=0 options=320 groupId=0 sequence=13>, 16, b'\rz\xd1\xc6\xfe\xff\xcc\xcc\xcc\x01\x00\x03\x03qc\x8d\xfe\xff\xcc\xcc\xcc\x01')
2019-12-06 17:13:16 DEBUG (MainThread) [bellows.ezsp] Application frame 52 (sendUnicast) received: b'0063'
2019-12-06 17:13:17 DEBUG (MainThread) [bellows.ezsp] Application frame 52 (sendUnicast) received: b'0064'
2019-12-06 17:13:17 DEBUG (MainThread) [bellows.ezsp] Application frame 69 (incomingMessageHandler) received: b'000401020402010001000099f2abfb36ffff0808130a0000293dff'
2019-12-06 17:13:17 DEBUG (MainThread) [zigpy.zcl] [0x36fb:2:0x0402] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=19 command_id=Command.Report_Attributes>
2019-12-06 17:13:17 DEBUG (MainThread) [zigpy.zcl] [0x36fb:2:0x0402] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=int16s, value=-195>>]]
2019-12-06 17:13:17 DEBUG (MainThread) [zigpy.zcl] [0x36fb:2:0x0402] Attribute report received: measured_value=-195
2019-12-06 17:13:17 DEBUG (MainThread) [bellows.ezsp] Application frame 52 (sendUnicast) received: b'0065'
2019-12-06 17:13:17 DEBUG (MainThread) [bellows.ezsp] Application frame 69 (incomingMessageHandler) received: b'0000002180000040010000b1ffbcb2b9ffff020c00'
2019-12-06 17:13:17 DEBUG (MainThread) [bellows.ezsp] Application frame 63 (messageSentHandler) received: b'00b2b900002100000040010000640f0000'
2019-12-06 17:13:17 DEBUG (MainThread) [bellows.ezsp] Application frame 69 (incomingMessageHandler) received: b'0000002180000040010000b2ffbcb2b9ffff020d00'
2019-12-06 17:13:17 DEBUG (MainThread) [bellows.ezsp] Application frame 63 (messageSentHandler) received: b'00b2b90000210000004001000065100000'
2019-12-06 17:13:19 DEBUG (MainThread) [bellows.ezsp] Application frame 69 (incomingMessageHandler) received: b'0000002180000040010000b0ffbbb2b9ffff020b00'
2019-12-06 17:13:20 DEBUG (MainThread) [bellows.ezsp] Send command nop: ()
2019-12-06 17:13:20 DEBUG (MainThread) [bellows.ezsp] Application frame 5 (nop) received: b''
2019-12-06 17:13:26 DEBUG (MainThread) [bellows.ezsp] Application frame 69 (incomingMessageHandler) received: b'0000002180000040010000b6ffbcb2b9ffff020b00'
2019-12-06 17:13:26 DEBUG (MainThread) [zigpy.device] [0xb9b2] Invalid state on future for 0x0b seq -- probably duplicate response
2019-12-06 17:13:26 DEBUG (MainThread) [bellows.ezsp] Application frame 63 (messageSentHandler) received: b'00b2b900002100000040010000630e0000'
2019-12-06 17:13:26 INFO (MainThread) [homeassistant.components.zha.api] Issue bind devices: source_ieee: [cc:cc:cc:ff:fe:c6:d1:7a] target_ieee: [cc:cc:cc:ff:fe:8d:63:71]
2019-12-06 17:13:30 DEBUG (MainThread) [bellows.ezsp] Send command nop: ()
[... cut ...]
2019-12-06 17:14:06 DEBUG (MainThread) [bellows.ezsp] Application frame 5 (nop) received: b''
2019-12-06 17:14:16 DEBUG (MainThread) [bellows.ezsp] Send command nop: ()
2019-12-06 17:14:16 DEBUG (MainThread) [bellows.ezsp] Application frame 5 (nop) received: b''
```
</details> 

I tried Touch-Linking a remote to some lights as per the manual, which worked, but all the remotes controlled all of the lights that had been linked. I assume that is not how that is supposed to work, since it is not as helpful.

Adminiuga, 2019-12-06:
Nah, actually this time remote did report the successful bind, although it is not clearly shown in the logs:
```
### Bind request with 0x0b TSN
2019-12-06 17:13:16 DEBUG (MainThread) [bellows.ezsp] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0xb9b2, <EmberApsFrame profileId=0 clusterId=33 sourceEndpoint=0 destinationEndpoint=0 options=320 groupId=0 sequence=11>, 14, b'\x0bz\xd1\xc6\xfe\xff\xcc\xcc\xcc\x01\x06\x00\x03qc\x8d\xfe\xff\xcc\xcc\xcc\x01')

### Bind request with 0x0c TSN
2019-12-06 17:13:16 DEBUG (MainThread) [bellows.ezsp] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0xb9b2, <EmberApsFrame profileId=0 clusterId=33 sourceEndpoint=0 destinationEndpoint=0 options=320 groupId=0 sequence=12>, 15, b'\x0cz\xd1\xc6\xfe\xff\xcc\xcc\xcc\x01\x08\x00\x03qc\x8d\xfe\xff\xcc\xcc\xcc\x01')

### Bind request with 0x0d TSN
2019-12-06 17:13:16 DEBUG (MainThread) [bellows.ezsp] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0xb9b2, <EmberApsFrame profileId=0 clusterId=33 sourceEndpoint=0 destinationEndpoint=0 options=320 groupId=0 sequence=13>, 16, b'\rz\xd1\xc6\xfe\xff\xcc\xcc\xcc\x01\x00\x03\x03qc\x8d\xfe\xff\xcc\xcc\xcc\x01')

### Reply to 0x0c TSN with success status (last two bytes)
2019-12-06 17:13:17 DEBUG (MainThread) [bellows.ezsp] Application frame 69 (incomingMessageHandler) received: b'0000002180000040010000b1ffbcb2b9ffff020c00'

### Reply to 0x0d TSN with success status (last two bytes)
2019-12-06 17:13:17 DEBUG (MainThread) [bellows.ezsp] Application frame 69 (incomingMessageHandler) received: b'0000002180000040010000b2ffbcb2b9ffff020d00'

### Reply to 0x0b TSN with success status (last two bytes)
2019-12-06 17:13:19 DEBUG (MainThread) [bellows.ezsp] Application frame 69 (incomingMessageHandler) received: b'0000002180000040010000b0ffbbb2b9ffff020b00'

```

So it bound successfully, the question why it still doesn't work. I'd expect it to be able to control the light directly. 

@dmulcahey We should collect and log results of https://github.com/home-assistant/home-assistant/blob/c9415ab75d84aedf5a1e667e40c49c3383fcbe91/homeassistant/components/zha/api.py#L560

Adminiuga, 2019-12-06:
i'll try to get one tomorrow and reproduce the issue. Per looks, it seems to be quite standard and should just work.

dmulcahey, 2019-12-06:
I have a bunch of ideas on how to clean up binding

Adminiuga, 2019-12-09:
it is sending group command on ZLL group

Adminiuga, 2019-12-09:
```
2019-12-08 19:16:17 INFO (MainThread) [zigpy.device] [0xbca5] Discovered endpoints: [1]
2019-12-08 19:16:17 INFO (MainThread) [zigpy.endpoint] [0xbca5:1] Discovering endpoint information
2019-12-08 19:16:18 INFO (MainThread) [zigpy.endpoint] [0xbca5:1] Discovered endpoint information: <SimpleDescriptor endpoint=1 profile=260 device_type=2048 device_version=1 input_clusters=[0, 1, 3, 4096, 64769] output_clusters=[3, 4, 6, 8, 25, 768, 4096]>
2019-12-08 19:16:19 DEBUG (MainThread) [zigpy.endpoint] [0xbca5:1] Manufacturer: LDS
2019-12-08 19:16:19 DEBUG (MainThread) [zigpy.endpoint] [0xbca5:1] Model: ZBT-CCTSwitch-D0001
```

Adminiuga, 2019-12-09:
On basic cluster it has attribute id 0x000B which reads `www.xxxx.com` 🤷‍♂

dmulcahey, 2019-12-09:
> On basic cluster it has attribute id 0x000B which reads `www.xxxx.com` 🤷‍♂

Wtf??? LOL

closed 2019-12-10 by dmulcahey

Adminiuga, 2019-12-11:
Apparently this device doesn't do direct binds, even though it replies with success to bind request operations.
 But soon, like really soon (next year) we're going to have Device Group management, then you could add the lights to the same group as this remote and control them at once directly.

dmulcahey, 2019-12-11:
My goal is to have the UI PR’s opened by the end of the weekend :)

PR #211 (2019-12-10): Add LDS ZBT-CCTSwitch-D0001 device.
Add a quirk for LDS ZBT-CCTSwitch-D0001 device.
Fixes #209 

@dmulcahey is there anything I'm missing in `device_automation_triggers` dict?