zigpy/zha-device-handlers #210: Ikea Remote E1810 not matching quirks?

andyzickler, 2019-12-10:
I'm trying to setup a new Ikea 5 button remote E1810 in ha but am not seeing any `zha_event`s. The logs say `Fail because profile_id mismatch on at least one endpoint` My endpoint information is `<SimpleDescriptor endpoint=1 profile=260 device_type=2080 device_version=1 input_clusters=[0, 1, 3, 32, 4096, 64636] output_clusters=[3, 4, 6, 8, 25, 4096]>`

I'm using the HUSBZB-1 and I was able to pair it with the ikea gateway and verify that it's on the latest firmware revision 2.3.014

Finally here's the full pairing log:

```
[0x0000:zdo] ZDO request 0x0036: [60, <Bool.false: 0>]
[0x5544:zdo] ZDO request 0x0006: [0xfffd, 49246, [25], []]
[0x0000:zdo] ZDO request 0x0036: [60, <Bool.false: 0>]
Device 0xd800 (ec:1b:bd:ff:fe:2a:c8:eb) joined the network
[0xd800] Requesting 'Node Descriptor'
[0xd800] Extending timeout for 0x0a request
Device 0xe02c (ec:1b:bd:ff:fe:2a:c8:eb) joined the network
Device ec:1b:bd:ff:fe:2a:c8:eb changed id (0xd800 => 0xe02c)
Canceling old initialize call
[0xe02c] Requesting Node Descriptor failed: 
[0xe02c] Discovering endpoints
[0xe02c] Extending timeout for 0x0c request
[0xe02c] Requesting 'Node Descriptor'
[0xe02c] Extending timeout for 0x0e request
Device 0xe02c (ec:1b:bd:ff:fe:2a:c8:eb) joined the network
Skip initialization for existing device ec:1b:bd:ff:fe:2a:c8:eb
[0xe02c:zdo] ZDO request 0x0013: [0xe02c, ec:1b:bd:ff:fe:2a:c8:eb, 128]
[0xe02c] Discovered endpoints: [1]
[0xe02c:1] Discovering endpoint information
[0xe02c] Extending timeout for 0x10 request
[0xe02c] Node Descriptor: <Optional byte1=2 byte2=64 mac_capability_flags=128 manufacturer_code=4476 maximum_buffer_size=82 maximum_incoming_transfer_size=82 server_mask=11264 maximum_outgoing_transfer_size=82 descriptor_capability_field=0>
[0xe02c] Discovering endpoints
[0xe02c] Extending timeout for 0x12 request
[0xe02c:1] Discovered endpoint information: <SimpleDescriptor endpoint=1 profile=260 device_type=2080 device_version=1 input_clusters=[0, 1, 3, 32, 4096, 64636] output_clusters=[3, 4, 6, 8, 25, 4096]>
[0xe02c] Extending timeout for 0x14 request
[0xe02c] Discovered endpoints: [1]
[0xe02c:1] Discovering endpoint information
[0xe02c] Extending timeout for 0x16 request
Ignoring message (b'181401040000420e494b4541206f662053776564656e0500004216545241444652492072656d6f746520636f6e74726f6c') on cluster 0: unknown endpoint or cluster id: 'No cluster ID 0x0000 on (ec:1b:bd:ff:fe:2a:c8:eb, 1)'
[0xe02c:1] Discovered endpoint information: <SimpleDescriptor endpoint=1 profile=260 device_type=2080 device_version=1 input_clusters=[0, 1, 3, 32, 4096, 64636] output_clusters=[3, 4, 6, 8, 25, 4096]>
[0xe02c] Extending timeout for 0x18 request
[0xe02c:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=24 command_id=Command.Read_Attributes_rsp>
[0xe02c:1] Manufacturer: IKEA of Sweden
[0xe02c:1] Model: TRADFRI remote control
Checking quirks for IKEA of Sweden TRADFRI remote control (ec:1b:bd:ff:fe:2a:c8:eb)
Considering <class 'zhaquirks.ikea.fivebtnremote.IkeaTradfriRemote'>
Fail because profile_id mismatch on at least one endpoint
Considering <class 'zigpy.quirks.ikea.TradfriPlug'>
Fail because endpoint list mismatch: {1, 2, 242} {1}
Considering <class 'zigpy.quirks.smartthings.SmartthingsMultiPurposeSensor'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
Fail because endpoint list mismatch: {11, 13} {1}
Considering <class 'zhaquirks.lutron.lzl4bwhl01remote.LutronLZL4BWHL01Remote2'>
Fail because profile_id mismatch on at least one endpoint
Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.osram.a19twhite.A19TunableWhite'>
Fail because endpoint list mismatch: {3} {1}
Considering <class 'zhaquirks.philips.rwl021.PhilipsRWL021'>
Fail because endpoint list mismatch: {1, 2} {1}
Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.xbee.xbee_io.XbeeSensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.xiaomi.mija.smoke.MijiaHoneywellSmokeDetectorSensor'>
Fail because device_type mismatch on at least one endpoint
device - 0xe02c:ec:1b:bd:ff:fe:2a:c8:eb entering async_device_initialized - is_new_join: True
device - 0xe02c:ec:1b:bd:ff:fe:2a:c8:eb has joined the ZHA zigbee network
[... 39 lines cut ...]
[0xe02c:1:0x1000]: channel: 'async_initialize' stage succeeded
[0xe02c:1:0x0006]: initializing channel: from_cache: False
[0xe02c:1:0x0006]: channel: 'async_initialize' stage succeeded
[0xe02c:1:0x0008]: initializing channel: from_cache: False
[0xe02c:1:0x0008]: channel: 'async_initialize' stage succeeded
[0xe02c:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=42 command_id=Command.Read_Attributes_rsp>
[0xe02c:1:0x0000]: initializing channel: from_cache: False
[0xe02c:1:0x0000]: channel: 'async_initialize' stage succeeded
[0xe02c:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=44 command_id=Command.Read_Attributes_rsp>
[0xe02c] Extending timeout for 0x2e request
[0xe02c:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=46 command_id=Command.Read_Attributes_rsp>
[0xe02c] Extending timeout for 0x30 request
[0xe02c:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=48 command_id=Command.Read_Attributes_rsp>
[0xe02c] Extending timeout for 0x32 request
[0xe02c:zdo] ZDO request 0x0036: [180, <Bool.true: 1>]
[0xe02c:1:0x0003] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=2 command_id=1>
[0xe02c:1:0x0003] ZCL request 0x0001: []
[0xe02c:1:0x0003] No handler for cluster command 1
[0xe02c:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=50 command_id=Command.Read_Attributes_rsp>
[0xe02c:1:0x0001]: initializing channel: from_cache: False
[0xe02c:1:0x0001]: channel: 'async_initialize' stage succeeded
[0xe02c](TRADFRI remote control): power source: Battery or Unknown
[0xe02c](TRADFRI remote control): completed initialization
[0xe02c] Extending timeout for 0x34 request
```

dmulcahey, 2019-12-10:
I’ll add another quirk to support this alternative version. They seem to be releasing 2 versions of each device. 1 using the zha profile and one using the zll profile.

closed 2019-12-20 by dmulcahey

causefx, 2020-02-11:
I'm also unable to use this ikea remote...

Bellows shows the correct endpoints:
```
Device:
  NWK: 0x9462
  IEEE: ec:1b:bd:ff:fe:8f:ea:7f
  Endpoints:
    1: profile=0x104, device_type=DeviceType.NON_COLOR_CONTROLLER
      Input Clusters:
        Basic (0)
        Power Configuration (1)
        Identify (3)
        Poll Control (32)
        LightLink (4096)
        Manufacturer Specific (64636)
      Output Clusters:
        Identify (3)
        Groups (4)
        On/Off (6)
        Level control (8)
        Ota (25)
        LightLink (4096)
```
I am getting zero zha_events for this remote.

Only thing i can see is that there is a "Unsupported ZDO request:0x8021" on the setup logs:
```
[0x0000:zdo] ZDO request 0x0036: [60, <Bool.false: 0>]
[0x8ce4:1:0x0402] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=26 command_id=Command.Report_Attributes>
[0x8ce4:1:0x0402] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=int16s, value=1328>>]]
[0x8ce4:1:0x0402] Attribute report received: measured_value=1328
Device 0x683b (ec:1b:bd:ff:fe:8f:ea:7f) joined the network
[0x683b] Requesting 'Node Descriptor'
[0x683b] Extending timeout for 0x2f request
Device 0x9462 (ec:1b:bd:ff:fe:8f:ea:7f) joined the network
Device ec:1b:bd:ff:fe:8f:ea:7f changed id (0x683b => 0x9462)
Canceling old initialize call
[0x9462] Requesting Node Descriptor failed: 
[0x9462] Discovering endpoints
[0x9462] Extending timeout for 0x31 request
[0x9462] Requesting 'Node Descriptor'
[0x9462] Extending timeout for 0x33 request
Device 0x9462 (ec:1b:bd:ff:fe:8f:ea:7f) joined the network
Skip initialization for existing device ec:1b:bd:ff:fe:8f:ea:7f
[0x9462:zdo] ZDO request 0x0013: [0x9462, ec:1b:bd:ff:fe:8f:ea:7f, 128]
[0x9462] Discovered endpoints: [1]
[0x9462:1] Discovering endpoint information
[0x9462] Extending timeout for 0x35 request
[0x9462] Node Descriptor: <Optional byte1=2 byte2=64 mac_capability_flags=128 manufacturer_code=4476 maximum_buffer_size=82 maximum_incoming_transfer_size=82 server_mask=11264 maximum_outgoing_transfer_size=82 descriptor_capability_field=0>
[0x9462] Discovering endpoints
[0x9462] Extending timeout for 0x37 request
[0x9462:1] Discovered endpoint information: <SimpleDescriptor endpoint=1 profile=260 device_type=2080 device_version=1 input_clusters=[0, 1, 3, 32, 4096, 64636] output_clusters=[3, 4, 6, 8, 25, 4096]>
[0x9462] Extending timeout for 0x39 request
[0x9462] Discovered endpoints: [1]
[0x9462:1] Discovering endpoint information
[0x9462] Extending timeout for 0x3b request
Ignoring message (b'183901040000420e494b4541206f662053776564656e0500004216545241444652492072656d6f746520636f6e74726f6c') on cluster 0: unknown endpoint or cluster id: 'No cluster ID 0x0000 on (ec:1b:bd:ff:fe:8f:ea:7f, 1)'
[0x9462:1] Discovered endpoint information: <SimpleDescriptor endpoint=1 profile=260 device_type=2080 device_version=1 input_clusters=[0, 1, 3, 32, 4096, 64636] output_clusters=[3, 4, 6, 8, 25, 4096]>
[0x9462] Extending timeout for 0x3d request
[0x9462:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=61 command_id=Command.Read_Attributes_rsp>
[0x9462:1] Manufacturer: IKEA of Sweden
[0x9462:1] Model: TRADFRI remote control
Checking quirks for IKEA of Sweden TRADFRI remote control (ec:1b:bd:ff:fe:8f:ea:7f)
Considering <class 'zhaquirks.ikea.fivebtnremote.IkeaTradfriRemote'>
Fail because profile_id mismatch on at least one endpoint
Considering <class 'zhaquirks.ikea.fivebtnremotezha.IkeaTradfriRemote'>
Found custom device replacement for ec:1b:bd:ff:fe:8f:ea:7f: <class 'zhaquirks.ikea.fivebtnremotezha.IkeaTradfriRemote'>
device - 0x9462:ec:1b:bd:ff:fe:8f:ea:7f entering async_device_initialized - is_new_join: True
device - 0x9462:ec:1b:bd:ff:fe:8f:ea:7f has joined the ZHA zigbee network
[... 59 lines cut ...]
[0x9462:1:0x0000]: channel: 'async_initialize' stage succeeded
[0x9462:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=85 command_id=Command.Read_Attributes_rsp>
[0x9462] Extending timeout for 0x57 request
[0x9462:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=87 command_id=Command.Read_Attributes_rsp>
[0x9462] Extending timeout for 0x59 request
[0x9462:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=89 command_id=Command.Read_Attributes_rsp>
[0x9462] Extending timeout for 0x5b request
[0x9462:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=91 command_id=Command.Read_Attributes_rsp>
[0x9462:1:0x0001]: initializing channel: from_cache: False
[0x9462:1:0x0001]: channel: 'async_initialize' stage succeeded
[0x9462](TRADFRI remote control): power source: Battery or Unknown
[0x9462](TRADFRI remote control): completed initialization
[0x9462] Extending timeout for 0x5d request
[... cut ...]
[0x6ea7:1:0x0402] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=int16s, value=1780>>]]
[0x6ea7:1:0x0402] Attribute report received: measured_value=1780
[0x8ce4:1:0x0402] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=27 command_id=Command.Report_Attributes>
[0x8ce4:1:0x0402] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=int16s, value=1340>>]]
[0x8ce4:1:0x0402] Attribute report received: measured_value=1340
```

Adminiuga, 2020-02-11:
@causefx Please open a new issue. In your case  it does match the signature, just a different problem.

@dmulcahey this looks problematic
```
[0x9462:1:0x1000] ZCL deserialize: <ZCLHeader frame_control=<FrameControlframe_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=67 command_id=65>
[0x9462:1:0x1000]: channel: 'async_configure' stage failed ex: list index out of range
```

@causefx if you decide to open new issue, please provide details about your radio and HA versions. +  Enable debug logging for the radio lib, as I'd like to see the raw incoming message

causefx, 2020-02-11:
Sorry about that - i'll open a new one now.

dmulcahey, 2020-02-11:
@Adminiuga yeah I have to dig more into this

PR #216 (2019-12-20): 210 add tradfri remote zha quarks
This should fix #210

I've tested this with my remote and am seeing events for all the buttons!