zigpy/zha-device-handlers #214: Add support for Xiaomi Mijia Honeywell gas sensor

smockle, 2019-12-12:
Shows up as `binary_sensor.lumi_lumi_sensor_natgas` when added in Home Assistant, but fills the log with `there are no channels in the discovery info` warnings and `Message send failure` errors:

```
INFO (MainThread) [zigpy.application] Device 0xc3xx (00:15:8d:00:xx:xx:xx:xx) joined the network
INFO (MainThread) [zigpy.device] [0xc3xx] Requesting 'Node Descriptor'
INFO (MainThread) [bellows.zigbee.application] ZDO Device announce: 0xc3xx, 00:15:8d:00:xx:xx:xx:xx
INFO (MainThread) [zigpy.application] Device 0xc3xx (00:15:8d:00:xx:xx:xx:xx) joined the network
INFO (MainThread) [zigpy.device] [0xc3xx] Node Descriptor: <Optional byte1=1 byte2=64 mac_capability_flags=142 manufacturer_code=4151 maximum_buffer_size=127 maximum_incoming_transfer_size=100 server_mask=0 maximum_outgoing_transfer_size=100 descriptor_capability_field=0>
INFO (MainThread) [zigpy.device] [0xc3xx] Discovering endpoints
INFO (MainThread) [zigpy.device] [0xc3xx] Discovered endpoints: [1]
INFO (MainThread) [zigpy.endpoint] [0xc3xx:1] Discovering endpoint information
INFO (MainThread) [zigpy.endpoint] [0xc3xx:1] Discovered endpoint information: <SimpleDescriptor endpoint=1 profile=260 device_type=257 device_version=2 input_clusters=[0, 4, 3, 1, 2, 10, 1280] output_clusters=[25, 10]>
WARNING (MainThread) [homeassistant.components.zha.core.discovery] there are no channels in the discovery info: {'unique_id': '00:15:8d:00:xx:xx:xx:xx-1', 'zha_device': <homeassistant.components.zha.core.device.ZHADevice object at 0x124cfd0xx>, 'channels': [], 'component': 'light'}
ERROR (MainThread) [homeassistant.core] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/Users/clay/.virtualenv/homeassistant/lib/python3.7/site-packages/zigpy/device.py", line 184, in request
    self.nwk, dst_ep, cluster
zigpy.exceptions.DeliveryError: [0x4cxx:1:0x000a]: Message send failure
```

* [Product Manual](https://files.xiaomi-mi.com/files/New-manuals/Honeywell-gas-detector-EN.pdf)
* [Related Comment in Home Assistant Community Thread](https://community.home-assistant.io/t/zha-zigbee-tested-devices-please-add-your-device-results/17718/358)

Adminiuga, 2019-12-12:
With xiaomi, when pairing, you need to keep pressing the button once every 2s to keep device awake.

smockle, 2019-12-13:
@Adminiuga Thanks for the reply! I’ve done that for battery-powered Aqara devices, but I didn’t know it was necessary for mains-powered devices like the Mijia Honeywell gas sensor. Thanks for the suggestion!

Unfortunately, after removing the device from Home Assistant and re-added it (while pressing the button every two seconds), I still see the following two errors in the Home Assistant > Developer Tools > Logs:

[image]

dmaciaszek, 2020-01-24:
In my case, sensor works only with binary sensor but I have the same log error `Failed to parse message...data is to short` as you.

closed 2021-08-20 by github-actions[bot]